tock/tock #1482: Tracking: tock-register-interface 0.5.0 (explicit vis in macros)

andre-richter, 2019-11-24:
As a follow-up to https://github.com/tock/tock/pull/1461#pullrequestreview-317256634, this issue can be used to track the progress along the way to release `0.5.0` of `tock-register-interface`.

TBD:
- [ ] Add visibility keyword (pub) parsing to all macros.
- [ ] Update existing codebase to use it.

CC @ppannuto @gendx 

Once done, I can update and release https://github.com/rust-embedded/register-rs `0.5.0` so that we have both in sync again. Right now, we are in a diverged state.

andre-richter, 2019-12-03:
Hi @bradjc @ppannuto @gendx,

may I ask about your internal priority of addressing those issues?

With `register-rs`, which depends on the tock register version on crates.io, I am currently blocked from two angles:

1. The latest version on crates.io has the unit test generation using the `#[test]` attribute macro which breaks testing with `custom_test_frameworks`, for which I opened #1489.
2. Even if #1489 would be merged using one or the other currently discussed approach, the 0.5.0 release of tock registers seems to be on hold until the pub keyword parsing is added to all macros as stated in #1461. 

I am asking because if it won't be possible to have 0.5.0 anytime soon, I am considering packing a temporary copy of tock-registers for `register-rs` for the time being that unblocks the testing framework, so that consumers of `register-rs` (my teaching OS project for the rust-embedded WG, oreboot) won't hit those issues.

Please let me know what you think.

Thanks a lot,
Andre

andre-richter, 2019-12-03:
Thanks for the fast response :+1: 
Of course I'll wait another week. I just wanted to get a feel if we are talking weeks or months.

andre-richter, 2019-12-13:
oreboot is planning to use register_structs!() soon: https://github.com/oreboot/oreboot/pull/206

Depending on if 0.5.0 lands before or not, pending pub keywords will have to be adjusted later.

closed 2020-01-10 by bradjc

ppannuto, 2020-01-10:
@andre-richter and we're live! https://crates.io/crates/tock-registers/0.5.0

Sorry this ran into the holidays and took a little longer than ideal

andre-richter, 2020-01-10:
No worries. I'm happy, thanks! :)

Will incorporate it over the weekend and see if everything works as expected on my side.

commit a35e4f53 (2020-01-10): tock-registers: release v0.5

Changes since v0.4.1:

 - #1510
   - Register visibility granularity: don't automatically make everything
      `pub`, rather give creation macro callers visbility control.

 - #1489
   - Make `register_structs!` unit test generation opt-out, so that
     `custom-test-frameworks` environments can disable them.

 - #1481
   - Add `#[derive(Copy, Clone)]` to InMemoryRegister.

 - #1428
   - Implement `mask()` for `FieldValue<u16>` which seems to have been
     skipped at some point.
   - Implement `read()` for `FieldValue` so that individual fields
     can be extra [... cut]
PR #1537 (2020-01-10): tock-registers: release v0.5
### Pull Request Overview

Bump version to release v0.5 of the library.

Closes #1482.

### Testing Strategy

This pull request was tested by compiling.

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make formatall`.