tock/tock #1500: Finish implementing nRF52 ADC driver

bradjc, 2019-12-09:
The ADC driver for the nRF52 chip only supports single samples. The continuous sampling function needs to be implemented:

https://github.com/tock/tock/blob/77f2a4915d29e4b3b9849970bb51ce9e71d851aa/chips/nrf52/src/adc.rs#L333-L335

Lockna, 2020-09-10:
Is anyone doing this issue yet?

bradjc, 2020-09-10:
Not that I'm aware of!

arjundeopujari, 2020-09-11:
@bradjc I can work on this issue if you want.

arjundeopujari, 2020-09-11:
@Lockna If you are already working on this issue, that is fine.  I just dont want two members to work on the same issue in parallel.

Lockna, 2020-09-11:
@arjundeopujari No, I'm not working on it. You can do that :)

arjundeopujari, 2020-09-11:
@Lockna @bradjc Cool. Ill try my luck at this problem then. Thanks.

arjundeopujari, 2020-09-14:
@Lockna @bradjc Looking at the most recent pr's, it seems another member is hard at work and committed to establishing the entire ADC interface including this issue.  I'll leave it to him/her then.

Kemalele, 2023-11-22:
Does anybody is doing it? I'd like to try to resolve this issue if not

bradjc, 2023-11-22:
#3453 happened between that previous comment and now, but, no, no one is working on the last missing APIs for the nrf52.

Kemalele, 2023-11-23:
> #3453 happened between that previous comment and now, but, no, no one is working on the last missing APIs for the nrf52.

ok, i'll try my luck then