tock/tock #1501: [RFC] trait for servicing interrupts

gendx, 2019-12-10:
## Observations

While working on chips from the nRF5x family, one thing that came up is that some chips support more interrupts than others. Currently, the code to service pending interrupts is in the `nrf52` crate.

https://github.com/tock/tock/blob/b22a314a8c55e8d8d7925d5e7eff5bbac0a0c07f/chips/nrf52/src/chip.rs#L52-L62

There is some common code and then a `match` over the interrupt number, that calls the suitable `handle_interrupt` function for the suitable driver.

With this architecture, an interrupt only defined on say `nrf52840` would have to be added to `nrf52`, so the driver would also need to be in `nrf52` instead of `nrf52840`. This isn't separating well various chips of the same family.

Alternatively, `nrf52840` could redefine the whole `chips.rs`, which would be a lot of duplicated code for maybe a few unique drivers.

## Draft Proposal

To make servicing of interrupts more flexible, an `InterruptServiceTrait` could be define as follows:

```rust
trait InterruptServiceTrait {
    unsafe fn service_interrupt(&self, interrupt: u32) -> bool;
}
```

- The function is `unsafe`, similarly to the `match` inside the current `service_pending_interrupts`.
- It returns a boolean, whether this interrupt number was serviced or not.

This allows composable interrupt services, for example `nrf52` defines a set of interrupt handlers, and on top of that `nrf52840` only defines its specific interrupts, returning `false` otherwise to indicate that the parent `nrf52` should handle the interrupt.

The chip's constructor, currently here:

https://github.com/tock/tock/blob/b22a314a8c55e8d8d7925d5e7eff5bbac0a0c07f/chips/nrf52/src/chip.rs#L22-L23

would be updated to take a `interrupt_service: &'static dyn InterruptServiceTrait` as parameter to service the interrupt. Then, each "end-user" chip such as `nrf52840` or `nrf52833` can pass a suitable `InterruptServiceTrait` to the generic `NRF52` chip constructor.

It seems that such a trait wouldn't be specific to Nordic chips but could be useful for all chips, as a trait in the kernel.

Any thoughts?

## Misc

Another trait that could be useful to clarify things would be an `InterruptHandler` trait that defines the `handle_interrupt` method. For now, drivers have this method by convention, but it isn't formalized as a trait. It could also be useful to put the "enable/disable all interrupts" pattern in this trait.

bradjc, 2019-12-10:
I like this idea, as I agree that copying chip.rs would lead to a lot of duplicated code. I also think this could be prototyped inside of the nRF52 crates, so it would be a relatively localized change, and if it proves useful then we could potentially expand its use.

gendx, 2020-03-05:
@phil-levis, since you mentioned in https://github.com/tock/tock/pull/1618#issuecomment-591726845 that:

> But it is important that Component remain a trait. If we just rely on functions named `create`, if we ever want to layer software we'll need to parameterize by types. Everything standard in the kernel should be a trait.

Should the `InterruptService` trait now implemented on nRF5x also be promoted to a trait in the kernel? It's been working well for Nordic boards, and it's quite standard among chips.

Likewise, I'd propose to have an `InterruptHandler` trait.

```rust
trait InterruptHandler {
    fn handle_interrupt(&self);
}
```

This would better formalize what has been happening in practice, and open more possibilities for "layering software".

WDYT?

phil-levis, 2020-03-05:
> Likewise, I'd propose to have an `InterruptHandler` trait.

This generally seems like a good idea to me. If we want anything to be composable or interchangeable we want to use a trait.

gendx, 2020-03-05:
Ok. I'll let this sit a bit more so that other opinions can chime in before we decide on implementing that or not.

bradjc, 2020-03-18:
InterruptService: I remain unconvinced that we want to pass specific peripherals through the layers to have one interrupt match point, rather than keep the interrupt match point in the same place where the peripheral is defined. However, in practice, it seems like this is going to be rare, and it won't be visible for most chips.

InterruptHandler: Agree. I think it is clear that the handle_interrupt pattern works and it should definitely be in a trait. I would vote for a name more like `PeripheralDriver` that matches what the code is that should implement it, like what we do with `Platform` or `Chip`.

hudson-ayers, 2020-08-13:
I am fleshing this out some more in #2069 (and extending it).

hudson-ayers, 2021-01-02:
I believe this has been addressed by #2069 and the related PRs. Please re-open if you feel otherwise, or I missed anything!

closed 2021-01-02 by hudson-ayers