tock/tock #1515: Invalid counter specification on nRF52(840)

gendx, 2019-12-19:
According to the specification, the real-time counter only has 24 bits on nRF52840 ([source](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf52840%2Frtc.html&cp=4_0_0_5_21)). I could confirm this empirically on nRF52840-DK with [this example program](https://github.com/tock/libtock-rs/pull/120).

However, the current definition in the nrf5x chip mis-defines it as a 32-bit register.

https://github.com/tock/tock/blob/8039ff50976b2a1866cbd72f7fdef3982c0d8326/chips/nrf5x/src/rtc.rs#L43-L44

The `max_tics` is also incorrect and should be `(1 << 24) - 1` instead.

https://github.com/tock/tock/blob/8039ff50976b2a1866cbd72f7fdef3982c0d8326/chips/nrf5x/src/rtc.rs#L141-L143

However, I haven't seen any chip defining 24-bit alarms so far. This would be a first use case of https://github.com/tock/tock/pull/1413.

This means that in the current state, all bets are off regarding timers after 512 seconds (the counter being 32 kHz), as the overflows will be incorrectly computed.

This issue also relates to https://github.com/tock/tock/issues/1513.

niklasad1, 2019-12-19:
I agree that `fn max_tics` is wrong but I don't see any problem in storing the value in an `u32` though because `COUNTER` is supposed to be stored at offset `0x504` and `PRESCALER` is supposed to be stored at `0x508`. I think that is still correct right?

However, we could:

1. introduce different types for all possible bit widths,  `31bits` to `1 bit` with additional padding in the memory map
2.  use `u32` then document that this is `u32` but only x number of bits should be used for the `functionality y`

gendx, 2019-12-19:
In terms of registers, the `CC` register type stores 24 bits so this should be used (or renamed `Counter`) for the `counter` register at the very least (of course under the hood it will store it in a `u32` but only 24 bits would be available via the register bitfield).

https://github.com/tock/tock/blob/8039ff50976b2a1866cbd72f7fdef3982c0d8326/chips/nrf5x/src/rtc.rs#L80-L82

> However, we could:
> 
> 1. introduce different types for all possible bit widths,  `31bits` to `1 bit` with additional padding in the memory map
> 2. use `u32` then document that this is `u32` but only x number of bits should be used for the `functionality y`

The features provided by `register_bitfields` should be enough, I don't think we need other custom types, at least in the register definition.

niklasad1, 2019-12-19:
> The features provided by register_bitfields should be enough, I don't think we need other custom types, at least in the register definition.

Agree

rajivr, 2019-12-20:
> However, I haven't seen any chip defining 24-bit alarms so far. This would be a first use case of #1413.

We are using 24-bit alarm timers in our out-of-tree chip crate. In our use-case, we are not using virtual timers, but only `hil::time::Time` and `hil::time::Alarm` traits and building up from there.

gendx, 2019-12-20:
> We are using 24-bit alarm timers in our out-of-tree chip crate. In our use-case, we are not using virtual timers, but only `hil::time::Time` and `hil::time::Alarm` traits and building up from there.

Well, a lot of components and capsules rely on the virtual timers, so incorrect virtual timers are affecting all of those: https://github.com/tock/tock/search?q=VirtualMuxAlarm

commit 5f33e500 (2019-12-20): Merge #1516

1516: Fix nrf5x's counter register to use 24 bits instead of 32 (fix #1515). r=niklasad1 a=gendx

### Pull Request Overview

This pull request fixes the register definition of nrf5x's chips `counter` register in the RTC. According to the specification, this counter only contains 24 bits of value, and not 32.

This pull request only fixes the register definition, but the virtual alarm still need to be refactored to support 24-bit timestamps (see https://github.com/tock/tock/pull/1413). However, this pull request should keep the current behavior in the meantime, i.e. shouldn't break [... cut]
closed 2019-12-20 by bors[bot]