tock/tock #1532: [RFC] Finer grained memory protection on Cortex-M3 MPUs

gendx, 2020-01-09:
## Motivation

Although the Cortex-M3 supports 8 regions for memory protection, Tock currently uses only two regions (per app): a ReadWrite region for RAM and a ReadOnly region for code in flash. The ReadWrite RAM region is further divided into 8 sub-regions, the first N of which are made accessible to the app and the last 8 - N are reserved for grants for this app.

Due to the way the regions work on Cortex-M3 (a region must be a power-of-two size, and with a starting address aligned on this same power-of-two), this leads to sub-optimal memory allocation for processes.

For example, the nRF52840-DK board has 256KB SRAM starting at `0x20000000`, the beginning of which is reserved for the kernel's stack and static data structures. With the current allocation strategy, the most memory a single app can get is 112KB.

```
+ 0x20000000 +
|  kernel    | 128 KB
+ 0x20020000 +
|  app rw    | 112 KB
+ 0x2003C000 +
|  app grant |  16 KB
+ 0x20040000 +
```

However, the kernel doesn't generally use 128KB; the grant space doesn't generally use 16KB either.

## Proposal

Instead of using only one memory region for an app's RAM space, more regions would allow to split the space in a finer grained manner.

Taking as example the 256KB of SRAM available on nRF52840-DK, this whole space could be a whole region split into 8 sub-regions of 32KB each. Then each of these 32KB could itself be split into 8 sub-regions of 4KB each. And if needed, the 4KB could themselves be split into 8 sub-regions of 512 bytes each.

By combining regions of various sizes, we could allocate memory to apps with a granularity of 4KB (with 3 regions per app) or 512 bytes (with 5 regions per app), as follows.
- Create a region covering the whole space. Split it into 8 sub-regions, enabling only the sub-regions that are fully in the app's space. There should be (at most) 2 sub-regions that are partly in the app's space: at the start and at the end of the app's space. These "boundary" 32KB sub-regions are left disabled.
- For each of the two boundary regions obtained above, create a new region of 32KB, split it into 4KB sub-regions, and enable the sub-regions that are fully in the app's space.
- There can again be sub-regions that are partially in the app's space, that can be split in a finer-grained manner with smaller sub-sub-regions.
- At the end, we constrain the app's rw space to be contiguous and aligned on 4KB "pages" (if we use 3 regions) or 512 bytes "pages" (if we use 5 regions).

With this strategy, an app on nRF52840-DK could potentially use as much as 62 "pages" of 4KB (counting one page for grants, and assuming that the kernel only needs 4KB), i.e. 248KB, more than twice as much as today.

The 512 byte granularity is likely less useful, but could be if one plans to run many (small) apps at the same time on a given board (given that each app requires at least 2 pages, one for rw memory and one for grants - so at most 31 apps could run with 4KB pages).

### Example

Consider the following memory layout constraints:
- We use 4KB "pages".
- The kernel requires 4KB < N <= 8KB, i.e. 2 pages.
- The app needs 64KB < N <= 68KB, i.e. 17 pages.
- The app's grant needs N <= 4KB, i.e. 1 page.

The ideal layout would be the following:

```
+ 0x20000000 +
|  kernel    |  8 KB
+ 0x20002000 +
|  app rw    | 68 KB
+ 0x20013000 +
|  app grant |  4 KB
+ 0x20014000 +
```

where the next app could start its `rw` region at `0x20014000`.

The app would be protected by the following 3 regions.

```
region #0
MPU parameters:
- start address = 0x20000000
- size = 0x40000 (256 KB)
- disabled pattern = 0b11111101

+ 0x20000000 +
|  disabled  | 32 KB
+ 0x20008000 +
|     rw     | 32 KB
+ 0x20010000 +
|  disabled  | 32 KB
+ 0x20018000 +
|  disabled  | 32 KB
+ 0x20020000 +
|  disabled  | 32 KB
+ 0x20028000 +
|  disabled  | 32 KB
+ 0x20030000 +
|  disabled  | 32 KB
+ 0x20038000 +
|  disabled  | 32 KB
+ 0x20040000 +
```
```
region #1
MPU parameters:
- start address = 0x20000000
- size = 0x8000 (32 KB)
- disabled pattern = 0b00000011

+ 0x20000000 +
|  disabled  | 4 KB
+ 0x20001000 +
|  disabled  | 4 KB
+ 0x20002000 +
|     rw     | 4 KB
+ 0x20003000 +
|     rw     | 4 KB
+ 0x20004000 +
|     rw     | 4 KB
+ 0x20005000 +
|     rw     | 4 KB
+ 0x20006000 +
|     rw     | 4 KB
+ 0x20007000 +
|     rw     | 4 KB
+ 0x20008000 +
```
```
region #2
MPU parameters:
- start address = 0x20010000
- size = 0x8000 (32 KB)
- disabled pattern = 0b11111000

+ 0x20010000 +
|     rw     | 4 KB
+ 0x20011000 +
|     rw     | 4 KB
+ 0x20012000 +
|     rw     | 4 KB
+ 0x20013000 +
|  disabled  | 4 KB
+ 0x20014000 +
|  disabled  | 4 KB
+ 0x20015000 +
|  disabled  | 4 KB
+ 0x20016000 +
|  disabled  | 4 KB
+ 0x20017000 +
|  disabled  | 4 KB
+ 0x20018000 +
```

## Extensions

For a given board, the RAM size would be rounded up to the next power of two, and this space would be divided into 8\*8 = 64 "pages" (in case of 3 regions per app), or 8\*8\*8 = 512 "pages" (in case of 5 regions per app). Each page can then be (1) reserved for the kernel, (2) a rw application page or (3) a grant application page.

As far as I understand, only the rw application pages need to be protected by the MPU's regions, and this is done one app at a time.

### Flash

The same method could be applied to the flash space, keeping in mind that there are only 8 regions available in total for SRAM+flash.

### RISC-V

I haven't checked how such a thing would work on the RISC-V boards, but the current kernel's MPU HIL should be compatible with this RFC's strategy on Cortex-M3, so this can be implemented independently of RISC-V.

bradjc, 2020-01-13:
I haven't worked with the cortex MPU, but I do know that we also use a region (regions?) for IPC.

ppannuto, 2020-01-13:
We currently allocate the 5 remaining regions to IPC channels, but that's
more an artifact of there being 5 remaining regions than a design choice to
have 5 IPC channels.

That is the tradeoff, however: Tock's IPC mechanism is shared memory
between applications, which is done by allowing another application access
to a region in your process's address space -- the MPU needs to be able to
open those windows when the companion application is running.

IPC has not seen a lot of use outside of dedicated examples, so that
interface is something that could be up for discussion, but I believe Amit
put non-trivial thought into its initial design.

gendx, 2020-02-07:
I didn't have the IPC use case in mind, thanks for the precision!

I don't see this as a high priority, but it could be interesting as an alternate memory allocation strategy (similarly to how we could support alternate scheduling strategies).

bradjc, 2021-02-22:
If we change IPC to a message passing implementation then we could pursue something like this.

bradjc, 2021-04-26:
I think we should absolutely do this if we don't need to reserve MPU regions for IPC.

closed 2021-04-26 by bradjc