tock/tock #1538: Tracking: Keep architecture-specific code in arch

ppannuto, 2020-01-10:
#1393 revealed that there is some code in the `chips` folder that really belongs in the `arch` directory. On a quick skim, most of this seems to be a symptom of new architecture bringup and arches where we have just one chip currently using it so the would-be duplication hadn't yet been spotted.

Regardless of cause, we should investigate each of the places where we had to use `cfgs` to control architecture outside of the architecture crate, and either move the code, or document why arch-specific code is appropriate there:

 - [ ] chips/arty_e21/src/chip.rs
 - [ ] chips/cc26x2/src/crt1.rs
 - [ ] chips/cc26x2/src/prcm.rs
 - [ ] chips/nrf52/src/ble_radio.rs
 - [ ] chips/nrf52/src/clock.rs
 - [ ] chips/nrf52/src/crt1.rs
 - [ ] chips/nrf52/src/nvmc.rs
 - [ ] chips/nrf52/src/ppi.rs
 - [ ] chips/nrf5x/src/gpio.rs
 - [ ] chips/sam4l/src/flashcalw.rs
 - [ ] chips/sam4l/src/lib.rs
 - [ ] chips/stm32f4xx/src/lib.rs
 - [ ] libraries/riscv-csr/src/csr.rs

bradjc, 2020-02-19:
I'm not sure how actionable this issue is. I think in the remaining cases the arch flag is needed to permit assembly code, and the assembly might be very specific to a certain chip and not general enough for all chips on that arch.

bradjc, 2020-06-19:
There are low level blocks outside of arch, like setting up the vector table, and other configuration, but it doesn't make sense to move that to /arch.

closed 2020-06-19 by bradjc