tock/tock #1542: Trustzone-M

obraunsdorf, 2020-01-13:
Hi there!
Is there any ongoing effort in making tock use the TrustZone-M or porting it completely to be executed in the Secure World?
If not, would you consider it a desirable goal for tock-os?

Thanks you!

bradjc, 2020-01-13:
I'm not aware of any efforts. Looking into how it would work and what use cases it makes sense for would definitely be of interest.

obraunsdorf, 2020-01-13:
Yes, I am struggling with the use cases a little bit. At least for running it completely as secure world OS: since tock's main purpose seems to be getting multiple applications to run concurrently and securely separated, I am not sure if this would be a feature someone would want for his/her code _within_ the secure world? 
What do you think?

bradjc, 2020-01-13:
Can trustzone hide parts of an application from the kernel? I could imagine an application wanting to run on someone else's hardware, but still be hidden from the kernel (or the owner of the hardware).

rajivr, 2020-01-14:
> Can trustzone hide parts of an application from the kernel?

This depends on the context in which the application is running. A secure world application and secure world kernel is hidden (and protected) from non-secure world application and non-secure world kernel.

> I could imagine an application wanting to run on someone else's hardware, but still be hidden from the kernel (or the owner of the hardware).

Depending on the application requirement, we might not need TrustZone for this. We can also use something like XOM (https://community.arm.com/developer/ip-products/processors/b/processors-ip-blog/posts/what-is-execute-only-memory-xom)