tock/tock #1544: use of exclusive_range_pattern is unnecessary and error prone

bcantrill, 2020-01-16:
In going through the Rust unstable features being used, we noticed use of `exclusive_range_pattern`.  This is only being used in a handful of places -- and at least at first blush, it feels like it's being used incorrectly.  For example, at https://github.com/tock/tock/blob/master/chips/e310x/src/chip.rs#L51-L58:

```rust
            while let Some(interrupt) = plic::next_pending() {
                match interrupt {
                    interrupts::UART0 => uart::UART0.handle_interrupt(),
                    index @ interrupts::GPIO0..interrupts::GPIO31 => {
                        gpio::PORT[index as usize].handle_interrupt()
                    }
                    _ => debug!("Pidx {}", interrupt),
                }
```
It appears that GPIO31 in this case can indeed be a valid GPIO interrupt -- but the exclusive range is in this case eliding the handling of it.  (If this code is correct, it seems that a better way to phrase it would be with an inclusive pattern (`..=`) to `GPIO30` and then call out the fallthrough of `GPIO31`.)  In the other places that enable `exclusive_range_pattern`, it is not in fact used. (This was in part noticed by the OpenTitan port, which copied use `exclusive_range_pattern` and then introduced a bug in UART handling as a result -- so it is indeed error-prone.)

It seems like use of `exclusive_range_pattern` should be removed; happy to open a pull request with this removal if and as needed.

bradjc, 2020-01-16:
Nice catch. We'd be happy to merge a PR to fix & remove this.

bcantrill, 2020-01-16:
Awesome -- will fix; thanks!

bradjc, 2020-03-18:
#1647

closed 2020-04-09 by bradjc