tock/tock #1545: `static mut` data considered harmful (as opposed to `static_init`)

gendx, 2020-01-16:
## Problem description

Despite the existing [`static_init` macro](https://github.com/tock/tock/blob/master/kernel/src/common/utils.rs) to initialize static data, there are still many uses of "raw" `static mut` buffers or other data. `git grep "^static mut"` and `git grep "^pub static mut"` yield many results across the code base.

However, `static mut` is fundamentally unsafe, because it's relatively easy to alias static data from anywhere in the program. There are indeed evidence of why it is fundamentally problematic in the Rust language - and it may even be deprecated in the future:
- "Why is static mut bad?" https://users.rust-lang.org/t/why-is-static-mut-bad/32888/6
- "Consider deprecation of UB-happy `static mut`" https://github.com/rust-lang/rust/issues/53639

In particular, given the pervasive definition of `unsafe` functions in Tock, it's quite easy to forget that accessing `static mut` data is unsafe and why it is unsafe. All one needs to do to alias a `static mut BUF` is to write `&mut BUF` in an unsafe block of code (and a large chunk of Tock's code is wrapped in `unsafe` functions). This was one of my concerns in https://github.com/tock/tock/issues/1447.

An example is the debug-writer component, which eagerly creates mutable references to the kernel's `INTERNAL_BUF` and `OUTPUT_BUF` static mut objects.

https://github.com/tock/tock/blob/b25a1f80e6db330988b6c8ce039b80351cc7402f/boards/components/src/debug_writer.rs#L41-L56

On the other hand the `static_init` macro "hides" the underlying static buffer (it's private to the macro's block of code), and returns a single `&'static mut` reference instead. The advantage of `&'static mut` over plain `static mut` is that it requires much more effort to alias it. Although it's still possible to alias in unsafe code, one needs to write something like `&mut *(buf as *mut T)` to alias the reference, and the `as *mut T` part is much more visible as a red flag that unsafe aliasing is happening.

Here is another toy example showing how much effort is required to alias data in both cases (when the code is not globally wrapped in an `unsafe` block).

```rust
static mut FOO: [u8; 128] = [0; 128];
static mut BAR: [u8; 128] = [0; 128];

fn main() {
    foo();
    bar(unsafe { &mut BAR });
}

fn foo() {
    let x = unsafe { &mut FOO };
    let y = unsafe { &mut FOO };
    // Undefined behavior because x and y both alias FOO.
    concurrent(x, y);
}

fn bar(buf: &'static mut [u8]) {
    // error[E0503]: cannot use `*buf` because it was mutably borrowed
    //let alias = &mut *buf;
    
    // To alias the static data, one needs to cast to a raw *mut pointer and
    // then reborrow it.
    let alias = unsafe { &mut *(buf as *mut [u8]) };
    
    // Undefined behavior because buf and alias both alias BAR.
    concurrent(buf, alias);
}

// Rust's aliasing rules assume that `left` and `right` don't alias.
fn concurrent(left: &mut [u8], right: &mut [u8]) {
    left[0] = 0;
    right[1] = 1;
    left[2] = 2;
    right[3] = 3;
    println!("{:?}", left);
}
```

## Proposal

- Audit all uses of plain `static mut` data in Tock, and replace them with `&'static mut` (or `&'a mut`) references that are initialized with the `static_mut` macro in boards.
- Understand if there are cases where `static_init` isn't enough, and why.

jrvanwhy, 2020-01-16:
If the Tock kernel does not run code in interrupt context, and will remain single-threaded, then we could use normal (non-`mut`) `static` declarations as well. I think a `TockStatic` type that wraps an arbitrary type and implements `Sync` (but does not allow code to retrieve a mutable reference) is safe given Tock's threading model.

If I'm right about the safety, that's another alternative we can pursue.

gendx, 2020-01-17:
> If the Tock kernel does not run code in interrupt context, and will remain single-threaded, then we could use normal (non-`mut`) `static` declarations as well. I think a `TockStatic` type that wraps [...]

I don't quite follow what you mean. How do you plan to mutate an object if it's not `mut` or one cannot retrieve mutable references? The goal of the `static_init` macro is precisely to obtain an (exclusive) mutable reference to a static object. For example, the [kernel's `RingBuffer`](https://github.com/tock/tock/blob/master/kernel/src/common/ring_buffer.rs) takes a mutable slice to back the buffer (and it's rightfully a mutable reference, as the slice is mutated), so one needs to give it a mutable reference.

I'm also not sure what `Sync` has to do with this problem.

Besides, what scope of the code is actually guaranteed not to run in interrupt context? Are you referring to the `kernel/` folder? Most of the raw `static mut` objects are not there:

```
$ git grep "^static mut" | cut -d'/' -f1 | sort | uniq -c
     61 boards
      9 chips
      1 doc
      2 kernel
$ git grep "^pub static mut" | cut -d'/' -f1 | sort | uniq -c
      4 arch
     29 boards
     38 capsules
    106 chips
      3 doc
      2 kernel
```

rajivr, 2020-01-17:
> Besides, what scope of the code is actually guaranteed not to run in interrupt context?

In Tock, almost all of the kernel code is guaranteed _not_ to run in interrupt context (handler mode). During normal execution, only code that we run in interrupt context is `generic_isr()`, `systick_handler()` and `svc_handler()`.  

`generic_isr()` disables the interrupt at NVIC. We then switch to privileged thread mode (kernel context) and do most of work in kernel context. For `svc_handler()` we modify `SYSCALL_FIRED` from interrupt context. In case of `systick_handler()`, we modify `SYSTICK_EXPIRED` from interrupt context.

phil-levis, 2020-01-17:
Yes -- Tock does most interrupt processing in the bottom half, basically for Rust's type safety. Architecturally you're allowed to directly write interrupt handlers if performance (latency) requires it, but that code is as `unsafe` as it gets.

gendx, 2020-01-17:
Thanks for the explanation on how interrupts are handled! Much clearer indeed.

hudson-ayers, 2020-11-10:
A note on my refactoring to reduced use of `static mut`: https://github.com/tock/tock/issues/1662#issuecomment-724530463

hudson-ayers, 2022-10-17:
Following the component cleanup, most uses of `static mut` have been removed from the codebase. Using imix as an example xx uses remain:

- `kernel/`: 4 (`DEBUG_GPIOS`/`DEBUG_QUEUE`/`DEBUG_WRITER`/`DYNAMIC_DEFERRED_CALL`)
- `arch/`: 3 (`SYSCALL_FIRED`/`APP_HARD_FAULT`/`SCB_REGISTERS`). 
- `chips/sam4l`: 1 (an atomic usize used in the dma driver)
- `boards/imix`: 4 (`WRITER`/`PROCESSES`/`CHIP`/`PROCESS_PRINTER`)

bradjc, 2022-10-17:
```
❯ cd capsules
❯ git grep 'pub static mut'
src/app_flash_driver.rs://! pub static mut APP_FLASH_BUFFER: [u8; 512] = [0; 512];
src/ble_advertising_driver.rs:pub static mut BUF: [u8; PACKET_LENGTH] = [0; PACKET_LENGTH];
src/i2c_master.rs:pub static mut BUF: [u8; 64] = [0; 64];
src/i2c_master_slave_driver.rs:pub static mut BUFFER1: [u8; 256] = [0; 256];
src/i2c_master_slave_driver.rs:pub static mut BUFFER2: [u8; 256] = [0; 256];
src/i2c_master_slave_driver.rs:pub static mut BUFFER3: [u8; 256] = [0; 256];
src/max17205.rs:pub static mut BUFFER: [u8; 8] = [0; 8];
src/mcp230xx.rs:pub static mut BUFFER: [u8; 7] = [0; 7];
src/nonvolatile_to_pages.rs://! pub static mut PAGEBUFFER: sam4l::flashcalw::Sam4lPage = sam4l::flashcalw::Sam4lPage::new();
src/pca9544a.rs:pub static mut BUFFER: [u8; 5] = [0; 5];
src/sdcard.rs:pub static mut TXBUFFER: [u8; 515] = [0; 515];
src/sdcard.rs:pub static mut RXBUFFER: [u8; 515] = [0; 515];
src/sdcard.rs:pub static mut KERNEL_BUFFER: [u8; 512] = [0; 512];
src/tsl2561.rs:pub static mut BUFFER: [u8; 4] = [0; 4];
```

hudson-ayers, 2022-10-17:
oh damnit, I copied my regex from the last time I did this and forgot it had a carveout to ignore buffers (as opposed to other static objects)

hudson-ayers, 2023-03-01:
Eventually we should seek to replace most of these remaining uses with a static reference to a `SyncCell` wrapped type (https://github.com/tock/tock/pull/3382#discussion_r1117311647)

commit 81225e8d (2023-06-09): Merge #3464 #3473

3464: capsules: remove lingering pub static mut r=ppannuto a=bradjc

### Pull Request Overview

This includes some comments as well as a few remaining static mut buffers used by boards.

Fixes #1545.

### Testing Strategy

travis

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.

3473: siphash24: add test r=ppannuto a=bradjc

### Pull Request Overview

This pull request adds a simple test for the SipHash24 software hash capsule.

### Testing Strategy

R [... cut]
commit c8ab95ac (2023-06-09): Merge #3464

3464: capsules: remove lingering pub static mut r=bradjc a=bradjc

### Pull Request Overview

This includes some comments as well as a few remaining static mut buffers used by boards.

Fixes #1545.

### Testing Strategy

travis

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
PR #3464 (2023-06-09): capsules: remove lingering pub static mut
### Pull Request Overview

This includes some comments as well as a few remaining static mut buffers used by boards.

Fixes #1545.

### Testing Strategy

travis

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
closed 2023-06-09 by bors[bot]