tock/tock #1549: Lack of panic output when there are no processes.

phil-levis-google, 2020-01-17:
If there is a panic, the kernel does not print out processor state unless there are processes:

https://github.com/tock/tock/blob/1e7e3f14a440a756383087d318614eed7868dca0/kernel/src/debug.rs#L60

prints CPU state by calling `panic_process_info', which in turn prints its information through Process:

```    if !procs.is_empty() {
        procs[0].as_ref().map(|process| {
            process.fault_fmt(writer);
        });
    }

    // print data about each process
    let _ = writer.write_fmt(format_args!("\r\n---| App Status |---\r\n"));
    for idx in 0..procs.len() {
        procs[idx].as_ref().map(|process| {
            process.process_detail_fmt(writer);
        });
    }
```

Getting debug output for a kernel panic should not require having a user process (e.g., if your entire system is in the kernel). It seems like this approach was used mostly so debug doesn't need a reference to `chip`. E.g., `Process::fault_fmt` just uses the chip reference:

```    unsafe fn fault_fmt(&self, writer: &mut dyn Write) {
        self.chip.userspace_kernel_boundary().fault_fmt(writer);
    }```

Would it make more sense for debug to have a direct reference to a UserspaceKernelBoundary? Better yet, it seems like this debug output functionality should be in a separate trait.

ppannuto, 2020-01-17:
I think a lot of the debug architecture has grown organically as we've
added things that feel useful. I agree that now that we have some idea of
what kind of debug information is useful and how to generate it a pass that
re-architects the implementation to be a bit more principled would be a
good idea.

bradjc, 2020-01-17:
That `fault_fmt` is named what it is and is inside of the `UserspaceKernelBoundary` trait is an artifact of how panic printing was done before `UserspaceKernelBoundary` existed, and including `fault_fmt` inside of the new trait was convenient at the time given how significant the rest of the changes were. The implementation is also very architecture dependent, so it kind of made sense to include it there at the time.

Looking at it now, however, having a new trait for arch-specific debug makes a lot more sense.

For cortex-m, the fault handlers in the lib.rs files would need to be updated since they only store the needed state when the fault occurs in user mode.

bradjc, 2020-02-19:
Fixed in #1590

closed 2020-02-19 by bradjc