tock/tock #1552: "Safe" wait primitive?

ppannuto, 2020-01-18:
While reviewing #1551, I noticed this:

```rust
        // start comparator
        regs.events_ready.set(0);
        regs.tasks_start.set(1);
        // wait for comparator to be ready
        // delay is on order of 3 microseconds so spin wait is OK
        while regs.events_ready.get() == 0 {}
    }
```

We do stuff like this in a few places, but as we start busy-looping on more complex peripherals, I wonder if we perhaps introduce the possibility for the kernel to hang in a way that's challenging to debug. Should we have a kernel primitive of some form that safely spins?

```rust
fn wait_for_hardware<F: Frequency> (fn: [closure type], max_µs: u32) -> bool {
     // Compute max number of tries given `max_us` and core speed
     let max_tries = ...
     while (max_tries) {
          if (fn()) return true;
     }
     false
}

        // wait for comparator to be ready
        if ! wait_for_hardware(|| {regs.events_ready.get() == 0}, 3) {
              return InitializationError; //?
              panic!("useful error message"); //?
        }
```

Or is this the kind of situation where we'd never really be able to do anything useful in the case of failure?

lgnashold, 2020-01-19:
With reference to #1551, it could be useful if the analog comparator is broken for some reason but the rest of the chip is okay. It would definitely be easier to debug an error in initializing the comparator rather than just getting stuck spinning.

phil-levis, 2020-01-19:
I think the standard approach is have a watchdog. Its handler could print the stack frame.

Or we could have lots of kernel debug messages, @bradjc ;)

ppannuto, 2020-01-20:
Sure, I guess the question is whether this is something better than a standard watchdog that we can do now. Especially in the context of multi-app boards, one might imagine a board with a dead sensor having one application fail (`ENODEVICE` perhaps from powering on the device failing) while the rest carries on.

Perhaps it's a bit fanciful -- or prematurely complex for the current reality of Tock -- and not useful / needed in practice. However, I guess one hope could be that a 'robust / reliable / etc' OS would aspire to never watchdog.

(Aside: This isn't meant to block #1551 in any way; I pulled this into a separate issue deliberately. The question here is more about this underlying philosophical question of how to handle dead hardware as an OS, to the extent that we can anyway)

bradjc, 2020-01-20:
I think this is an interesting push in a very Tock direction. There are others along this line:
- `map()` calls where the option is unexpectedly `None`
- functions/operations that cause an unexpected `panic!()`

I'm also intrigued to see how many capsules (broad definition) in Tock could fail and still have at least one process keep running.

phil-levis-google, 2020-01-31:
My experience is that hanging bugs are the easiest to find. Well, as long as you can attach a debugger. :)

I think this starts getting back to Brad's argument that we should be very careful in when we allow the kernel to panic (especially when it comes to things like unwrap()). @TPackard is getting very close with durable kernel log, which some software failures can go to (like my RF233 buffer management). I'd also argue that off-chip peripherals should generally not cause panics. But on-chip failures can and should.

I know @bcantrill has some strong opinions on fault models based on his years in Solaris and cloud services.

ppannuto, 2020-01-31:
I think the “if debugger” point is exactly the motivation for this —
imagine a deployed system where instead of a node mysteriously dropping
out, it sends a diagnostic “Err: ADC failure”!

I’m not advocating for panic as the response by any means — indeed I think
I fought you the hardest on the rf233 panic’s :). The kernel log sounds
interesting, excited to see that

phil-levis-google, 2020-01-31:
> I think the “if debugger” point is exactly the motivation for this — imagine a deployed system where instead of a node mysteriously dropping out, it sends a diagnostic “Err: ADC failure”!

Yes, of course. That's why I put a :). You can't attach a debugger to most devices in the field!

hudson-ayers, 2020-05-28:
I am pretty strongly in favor of this -- I don't even think we need to commit either way to how users of the interface handle a failure. But IMO a panic is better than a hang, and an error code is better than either. 

Given Tock's claims about dependability, I think it is very important that a failure in a single peripheral not be guaranteed to cause all applications, including those that may not need that peripheral at all, to fail.