tock/tock #1557: Broken GPIOs on nRF52840-DK

gendx, 2020-01-22:
I flashed the latest Tock kernel on a nRF52840-DK board, along with the `button_leds` example app (https://github.com/tock/libtock-rs/blob/c84685a99ea55cdb36823c234d4d1fc00dbc3672/examples/button_leds.rs).

The app is supposed to toggle a LED whenever the corresponding button is pressed, but with the latest Tock kernel it's not working as intended anymore. Most button presses seem just ignored by Tock (i.e. the LED is not toggled).

gendx, 2020-01-22:
**Update:** The problem is that setting the `PullUp` floating state was removed in #1541 - and that the component doesn't support setting that. CC @hudson-ayers @bradjc.

- https://github.com/tock/tock/pull/1541/files#diff-29f6b060c97fc794cb5569c7c1a95f45L233-L235
- https://github.com/tock/tock/pull/1541/files#diff-444e1e15bd50663750639738ff8576a1L287-L289

Currently working on a fix.

gendx, 2020-01-22:
Also I think that we should be doing a better job at testing (and documenting more precisely which tests have been run) before merging these kind of breaking changes to boards. The relatively simple `button_leds` test on the migrated boards in https://github.com/tock/tock/pull/1541 should have caught it.

hudson-ayers, 2020-01-22:
Thanks for catching this, sorry that I introduced the bug in the first place. I checked functionality on Hail and Imix and got lazy on testing..didn't realize that the nrf52 boards had to set the floating state for the buttons while the sam4l boards do not.

closed 2020-01-29 by bradjc