tock/tock #1559: Finished Processes

sv2bb, 2020-01-22:
Is there a way for processes to mark that they are finished? If so, how?
Why is there no finished state for a process?

hudson-ayers, 2020-01-23:
Hey Samyukta,
1) I am currently working on implementing a scheduler trait to make it easy to implement different scheduling algorithms in the kernel. This seems like a great use case that I would like my trait to eventually support, so I'll try to keep it in mind as I go! If you have any details on the format of the graph that you plan to give to the Kernel to order processes I would love to hear them.

2) I don't think there is currently a mechanism for processes to mark that they are finished. Looking at process state + callbacks seems cleaner than adding another syscall. Right now I don't think we track the number of callbacks enabled for a given process, but I think that could be done pretty easily by adding a `callback_count` field to the `Process` struct and incrementing it everytime a `Syscall::SUBSCRIBE`  is handled in sched.rs with a non-null pointer, and decrementing it when it is called with a null pointer. You could even add a `process::State::Finished` state that you would transition to whenever a `Syscall::YIELD` is handled and `callback_count == 0`.

sv2bb, 2020-01-23:
Thanks for your input, Hudson! Yes, I'll update you with the graph stuff when I have something reasonable. Regarding the callback checks, that's about what Brad and I discussed so I'll see where I get with that.

bradjc, 2020-01-27:
I think the idea that the kernel could detect a process is finished automatically is elegant, but I'm not sure it's really feasible for two reasons:

1. The kernel would have to check that there are no callbacks for the process now, and that there will be no callbacks for the process in the future. Keeping a count of subscribe and unsubscribe calls would work, but only if a process does not subscribe to the same driver+subdriver twice, or unsubscribe twice.
2. The app itself would somehow have to keep track of what it has subscribed to, and then be able to de-register all of its callbacks. This would likely be tricky since callbacks can be registered inside of libraries.

Enabling a process to simply state that it is finished (using exit) seems like a more straightforward approach. Alternatively, I could see a combination where if there are easy cases to detect when an app has ended then the kernel could update the process state to reflect that.

bradjc, 2020-01-27:
I am wondering, however, if there is no "ended" state in Tock currently because the use case hasn't really come up or if there is a design reason for not including it.

bradjc, 2020-02-18:
It might be useful to make a concept like this a bit more general. An app may wish to `exit`, meaning it is done, and no longer needs to run. However, apps may also want to be able to crash with the expectation that the kernel will restart them. Perhaps the app got into some weird state the developer didn't know how to handle, and thinks that a restart might help.

hudson-ayers, 2020-02-18:
> I am wondering, however, if there is no "ended" state in Tock currently because the use case hasn't really come up or if there is a design reason for not including it.

I did mention this idea a couple times on the Tock call and no-one mentioned any design reason for not including one. Though I can't remember if that was the call that @alevy missed.

> It might be useful to make a concept like this a bit more general. An app may wish to exit, meaning it is done, and no longer needs to run. However, apps may also want to be able to crash with the [...]

In theory an app with its fault policy set to restart could call `return 1/0;`. Though I agree that if we add a syscall for exit it might as well take an argument that specifies an option for "fault me and restart".

bradjc, 2021-02-12:
Support for this has been merged into the 2.0 branch.

closed 2021-02-12 by bradjc