tock/tock #1563: Has anyone tried to make Tock work on a Kendryte K210 based board (RISC-V)?

mindjuice, 2020-01-27:
I've been experimenting with the K210 using some Rust code examples:

`https://github.com/laanwj/k210-sdk-stuff/tree/master`

The particular board I'm using right now is the Maix Dock:

`https://www.seeedstudio.com/Sipeed-M1-dock-suit-M1-dock-2-4-inch-LCD-OV2640-K210-Dev-Board-1st-RV64-AI-board-for-Edge-Computing.html`

More info here:

`https://kendryte.com/downloads/`

Wondering if anyone has played with getting this to work with Tock, or if it's planned (and if it's even suitable -- not sure the K210 has the required hardware support, as it seems Tock might require an MMU, and I can't see any documentation saying the K210 has one.

bradjc, 2020-01-27:
No, no one has tried getting Tock on that chip (that I am aware of). It certainly looks like an interesting platform, but Tock doesn't currently have 64 bit RISC-V support, and we haven't been targeting multi-core platforms so far.

I think it raises an interesting question: Is Tock a reasonable OS for such a platform? We've been targeting much more constrained platforms, and Tock may not be a good fit for something like the K210. On the other hand, the K210 does not seem to support Linux, so what type of OS _is_ appropriate?

alistair23, 2020-01-27:
The K210 is a 64-bit platform with M, S and U mode support. The only key feature that it is lacking for a full Linux boot is a working MMU and a larger amount of DRAM. There are forks of no-MMU Linux that run on it that are slowly being upstreamed.

I think it ends up being a little overpowered for Tock. Considering that Tock has no RV64 support it would be a chunk of work to support the K210 (although maybe it's a good investment).

mindjuice, 2020-01-28:
Thanks for the info.  The chip developer provides freeRTOS support, and I’ve been debating using that or going bare metal, but I’d prefer an all-Rust solution.

This K210 chip is quite powerful and I expect it might gain popularity.  TBH it’s overkill for our application, but I understand the price point is very good, and I can clock it down to use less power (drawing about 210mA right now).

alevy, 2020-01-29:
Feel free to re-open if there are follow-ons, but I'm marking this closed for now.

closed 2020-01-29 by alevy