tock/tock #1569: Process Console output is not quite right

bradjc, 2020-01-28:
If you run the `list` command with the process console right now (on Hail, at least) you get output that looks like:

```
list PID    Name                Quanta  Syscalls  Dropped Callbacks  Restarts    State
  00	blink                11950        14                  0         0  Running

```

which is...close, but not quite right. I found myself asking, what happened to the newline after list?

Well, the answer to that question is: it moved to _after_ the list output, hence the extra new line there. So why did that happen?

Well, it's a byproduct of #1522, which changes TX transmission from happening right away to happening in the next kernel loop. The process console is telling the virtual UART to send the new line, and the virtual uart queues that, but before the virtual uart can send it the process console calls `debug!()`, which also asks the virtual uart to queue some output. When the kernel loops back around, the virtual uart looks at what it has to send, finds debug's output first, sends that, and then later sends the process console output.

I'm not really sure how to fix this. Thoughts?

phil-levis, 2020-01-30:
It shouldn’t be mixing debug! and its own UART client; there’s no assurance of ordering. The super-correct approach is to maintain an internal buffer to queue its output. Then it can control ordering.

Alternatively, the simpler solution is it could wait until the write to \r\n after a command completes (you get the transmitted_buffer callback) before executing the command. 

Phil

> If you run the list command with the process console right now (on Hail, at least) you get output that looks like:
>
> list PID    Name                Quanta  Syscalls  Dropped Callbacks  Restarts [...]

———————
Philip Levis (he/him)
Associate Professor, Computer Science and Electrical Engineering
Faculty Director, lab64 Maker Space
Stanford University
http://csl.stanford.edu/~pal

bradjc, 2020-01-30:
One option is to wait for the hypothetical console_mux work to develop which would remove the use of `debug!()` from the process console capsule.

phil-levis, 2020-01-30:
I think the console mux is going to be a long discussion — I’m moderately against it. Consoles shouldn’t be muxed. They are, well, consoles. Being able to interact with just a simple serial port is important.

Phil

> 
> One option is to wait for the hypothetical console_mux work to develop which would remove the use of debug!() from the process console capsule.

———————
Philip Levis (he/him)
Associate Professor, Computer Science and Electrical Engineering
Faculty Director, lab64 Maker Space
Stanford University
http://csl.stanford.edu/~pal

PR #1790 (2020-04-30): capsule: process console: fix printout ordering
This avoids a potential issue with a virtual uart for the process console. Since the console echos characters to the console itself, but uses `debug!()` for a majority of its printouts, the last echoed newline can get stuck in the virtual uart queue behind the debug!() output, causing slightly malformed output in the terminal. This makes sure that the newline has been echoed before calling any debug!() commands.

This fixes #1569.

### Testing Strategy

This pull request was tested by running the process console on hail and verifying the output l [... cut]
closed 2020-04-30 by bradjc