tock/tock #1572: Doc: Transparency and policies for goverance: core-wg

ppannuto, 2020-01-31:
Call notes (quite raw), as starting point for formal documentation:

What is charter of the core-wg?

 - kernel & capsules clearly owned by core
 --- but maybe some capsules, e.g. IPv6 networking group, owned by federated organization
 --- we sort of do this already for more experimental capsules
 --- could do this better on crate boundaries (capsules is a bit of a giant mess presently)
 - boards should be owned by the person motivated to maintain it
 --- Though some should be our "model platform" that are owned/endorsed by core
 - chips are like boards, but may fall to core to maintain once implemented
 - arch are similar to boards, but even more likely to fall to core to maintain once implemented
 - n.b. "what these crates (arch, chips, etc) have to do is still dictated by the kernel crate"
 --- delegating implementation detail ownership, but not interface ownership
 - external libraries
 --- "should probably be owned by core"
 --- "but what about library for one chip"?
 --- "if used by the kernel, should be maintained by core" ?
 --- "want to avoid library used by one chip though.."
 --- motivates core owning libraries, perhaps

Secondary discussion:
 - We should also consider a more explicit mission statement for Tock itself

Action Item: @phil-levis will propose sample charter

phil-levis, 2020-02-15:
I've shared a core charter with the group for comments, once it's been through a few passes I'll submit a PR.

bradjc, 2020-03-18:
#1700

ppannuto, 2020-04-09:
Closed by #1700

closed 2020-04-09 by ppannuto