tock/tock #1607: Tock 2.0

phil-levis, 2020-02-15:
Based on some discussions in -dev:

syscall return values: https://groups.google.com/d/msg/tock-dev/jicoPcapHNU/ynBCaV7LAQAJ
alarm hil: https://groups.google.com/d/msg/tock-dev/CWecIZq7jqg/AvLtMBedEwAJ

as well as discussions on moving to Result within the kernel:

https://groups.google.com/d/msg/tock-dev/G2uyQmkQEUQ/ROdQZUtmDQAJ

as well as discussions at Tock World 4 on the need for a read-only allow:

#1274 

as well as discussion on a new syscall for a process to exit, it seems like it might be time to for Tock 2.0. The goal of this issue is to discuss and reach consensus on the changes we will make. These should be constrained to changes that are not backwards compatible (i.e., require a 2.0). Let's not open up a general discussion of features we'd like, etc.

So it seems like the 4 starting points are:
 - Changing syscall ABI to have 64-bit (two word) return values,
 - Changing kernel to have a unified Result return type and having a clear translation to syscall ABI (e.g., for libtock-rs),
 - Adding a read-only allow syscall,
 - Adding a process exit syscall.

@alevy @bradjc @pfmooney @ppannuto @jrvanwhy @hudson-ayers @jon-flatley @brghena @niklasad1

gendx, 2020-02-17:
I'd add that a 3-word return ABI could be useful as well:
- 1 word for the status code.
- 2 words for up to 64 bits of useful result.

Rationale:
- This would allow syscalls to return 64-bit timestamps for example (besides the status code). See: https://groups.google.com/d/msg/tock-dev/CWecIZq7jqg/RrWySQLbBgAJ
- Syscalls already have 4 input registers, so returning in 3 registers shouldn't increase pressure on registers (from the userspace's point of view). At least from what I observed in practice with `objdump` (for libtock-rs) the compiler already assumes that all of the 4 inputs are overwritten by the kernel.
- Syscalls that only need to return 32 bits can simply ignore the third word (both in terms of writing a result in the kernel and reading it in userspace).

bradjc, 2020-02-17:
I also want to think about how tockloader and TABs can help detect if someone tries to run a 1.0 binary on a 2.0 kernel.

rajivr, 2020-02-18:
Once Tock 2.0 system-call design is finalized, it would be nice if we can specify the register level interface in `doc/Syscalls.md`, something similar to how it is documented here.

http://man7.org/linux/man-pages/man2/syscall.2.html
http://xenbits.xen.org/docs/unstable/hypercall/arm/include,public,arch-arm.h.html#incontents_arm_abi

This can be helpful for users who would want to use TockOS system call infrastructure, but not necessarily its semantics.

I can write the documentation for ARM architecture.

gendx, 2020-02-18:
> I also want to think about how tockloader and TABs can help detect if someone tries to run a 1.0 binary on a 2.0 kernel.

The current TBF header has a [version](https://github.com/tock/tock/blob/master/kernel/src/tbfheader.rs#L16) (although rather a version of the TBF format than a version of the kernel). We could add a new `kernel_version` field as well to prevent loading of incompatible applications at runtime.

It would then be the responsibility of the application compiler to populate this field in the TAB.

Regarding version detection at tockloader level, there could be some well-known address in flash where the Tock kernel version is written - tockloader would read that and compare it to the version in the TAB to accept or reject loading accordingly.

All of that would work for versions of Tock >= 2, given that Tock 1 doesn't have this kernel versioning of apps. If we bump the TBF header version from 2 to 3 at the same time, this would also prevent loading 1.x apps on Tock 2.x and 2.x apps on Tock 1.x.

---

Another question to add is what level of versioning we want to achieve? ABI compatibility i.e. coarse-grained detection of 1.x vs. 2.x? Or some finer-grained compatibility taking into account the list of drivers available on 2.0 vs. 2.1 for example (although in principle this can be handled at runtime with the `ENODEVICE` return code)?

Should we give apps the freedom to choose? For example:
- Application A requires Tock >= 2.1 and doesn't want to be loaded on Tock 2.0 because this older kernel doesn't support an essential driver.
- Application B works on Tock 2.1, but accepts to be loaded on Tock 2.0 as well by gracefully handling missing drivers.

phil-levis, 2020-02-18:
> Another question to add is what level of versioning we want to achieve? ABI compatibility i.e. coarse-grained detection of 1.x vs. 2.x? Or some finer-grained compatibility taking into account the [...]

Tock considers this to be an ENODEVICE problem. Particular system call drivers aren't intended to be fundamentally part of a release; a board's `main.rs` is part of the board, not kernel definition. For example, a driver could exist in the tree in 2.1 but not have been included in a particular board build.

gendx, 2020-02-19:
> Tock considers this to be an ENODEVICE problem. Particular system call drivers aren't intended to be fundamentally part of a release; a board's `main.rs` is part of the board, not kernel definition. [...]

Thanks!

What if at some point in the future we decide to change the driver numbers to re-organize them? Are driver numbers reserved forever or could there be breaking releases of Tock allowing that? Although I don't expect needing it in the near future, there can be value in future-proofing Tock if the need arises at some point.

I can imagine that if applications are tied to a specific major version of Tock, then bumping up the major version of Tock would allow to re-number the drivers if necessary.

bradjc, 2020-02-19:
> I can imagine that if applications are tied to a specific major version of Tock, then bumping up the major version of Tock would allow to re-number the drivers if necessary.

Yes I believe that a major version number change would permit any arbitrary changes to the syscall interface, including renumbering drivers (or the numbers used inside of the drivers as well).

phil-levis, 2020-02-19:
What if at some point in the future we decide to change the driver numbers to re-organize them? Are driver numbers reserved forever or could there be breaking releases of Tock allowing that? Although I don't expect needing it in the near future, there can be value in future-proofing Tock if the need arises at some point.Good point. I think that would constitute a major version number as well, as it would change the interface between userspace and the kernel.We tried to lay out the numbers to be reasonably future proof to avoid something like this, but it could always happen due to unforeseen circumstances.Phil

gendx, 2020-02-20:
So it looks like adding a kernel version (at least major version) in the TBF header (thereby creating a new TBF format v3) would allow to address @bradjc's concern about running a binary on an incompatible kernel:

> I also want to think about how tockloader and TABs can help detect if someone tries to run a 1.0 binary on a 2.0 kernel.

Aside that, do you want to keep all the discussions on this issue, or to spawn more specific RFC issues for planning each breaking change of Tock 2.0?

bradjc, 2020-02-20:
> Aside that, do you want to keep all the discussions on this issue, or to spawn more specific RFC issues for planning each breaking change of Tock 2.0?

I say split it, and keep this issue as a way to keep track of the different RFCs/tracking issues.

Woyten, 2020-02-21:
In the same way that libtock-rs can benefit from a read_only_allow command, it might make sense to add a subscribe_once command which triggers a subscription at most once.

bradjc, 2020-02-21:
Adding a reference to #1288

phil-levis, 2020-02-24:
> In the same way that libtock-rs can benefit from a read_only_allow command, it might make sense to add a subscribe_once command which triggers a subscription at most once.

Is there a compelling use case for this? The use case for read_only_allow is sharing const data with the kernel (e.g., writing out a const array). Without read_only_allow, you have to copy the buffer into RAM and so have a RAM buffer at least as large as your largest const buffer.

subscribe_once, in the cases that it's needed, seems like it could be handled by a userspace. I.e., when the callback invokes, you just unsubscribe. Following the fixes to callback semantics from @gendx (#1376 ), this should then stop all pending callbacks.

gendx, 2020-02-24:
> Is there a compelling use case for this? The use case for read_only_allow is sharing const data with the kernel (e.g., writing out a const array). Without read_only_allow, you have to copy the [...]

I don't think "callback once" can be reliably simulated by "callback + unsubscribe". If the underlying event triggers at a too high frequency, the app doesn't have the time to handle the callback and unsubscribe before the callback queue for this app is full. At this point, callbacks originating from other drivers are dropped due to the queue being full.

On the other hand, with the kernel being aware of a "send me a callback once" intent, the kernel can limit itself to putting one callback in the queue, thereby not affecting the availability of other drivers.

I think that makes a difference.

I'm also not sure how many drivers are reliable w.r.t. dropped callbacks. If you take the [RNG driver](https://github.com/tock/tock/blob/master/capsules/src/rng.rs), a callback is triggered to signal userspace when the buffer has been filled with random data. However, userspace implementations will typically block until the callback has been fired ([libtock-c](https://github.com/tock/libtock-c/blob/master/libtock/rng.c#L58), [libtock-rs](https://github.com/tock/libtock-rs/blob/master/src/rng.rs#L37)). So if the callback is dropped (see [process.rs](https://github.com/tock/tock/blob/master/kernel/src/process.rs#L544-L545) and [callback.rs](https://github.com/tock/tock/blob/master/kernel/src/callback.rs#L94-L96)), the application will typically get stuck in an infinite loop. And I'm not aware of any driver actually taking action when `Callback::schedule` fails.

Having "once-only callbacks" as a concept in the kernel could at least give guarantees of availability in that regard for applications that need these guarantees.

alistair23, 2020-03-07:
As discussed today I would like to raise including a yield-no-wait syscall. This could either be done with:
 1. A new syscall that is yield-no-wait
 2. Extend yield to take a command argument which could specify no_wait (preferred)

Thoughts?

bradjc, 2020-03-09:
I would like to raise including a yield-no-wait syscall. This could either be done with:
> 
>     1. A new syscall that is yield-no-wait
> 
>     2. Extend yield to take a command argument which could specify no_wait (preferred)

`yield` with options makes sense to me.

gendx, 2020-03-10:
> As discussed today I would like to raise including a yield-no-wait syscall.

I've also thought about it, and this definitely makes sense!
+1 to adding a parameter to the current `yield` syscall.

jrvanwhy, 2020-03-17:
I'd like to add another topic to the Tock 2.0 discussion: re-think the application list/TBF header design with the threat model in mind.

The current threat model draft (in #1632) asks the application loader to validate several of the TBF header fields. I've been contemplating whether security-relevant headers should be separate from other TBF headers. In particular, the use of an intrusive linked list as the application list seems odd.

labbott, 2020-03-18:
> I'd like to add another topic to the Tock 2.0 discussion: re-think the application list/TBF header design with the threat model in mind.
>
> The current threat model draft (in #1632) asks the [...]

I like this idea. I've been looking at extending TBF to better support non-relocatable applications and getting rid of the implicit list would make some things easier.

bradjc, 2020-03-18:
> The current threat model draft (in #1632) asks the application loader to validate several of the TBF header fields. I've been contemplating whether security-relevant headers should be separate from [...]

I'm not entirely sure this is a 2.0 discussion per se, depending on what you have in mind for changes. I don't _think_ we stabilized a notion of how the kernel finds apps, or that an app in flash at address X for 1.0 would also work at address X in flash at 1.5 (we just say that the same app .tbf file that worked at 1.0 has to work at 1.5). So if the kernel maintained some sort of data structure on where apps are located, for example, I think that would be fine to add now. Also, while we moved `load_processes()` to the kernel for convenience, boards are free to write their own `load_processes()` function that does not rely on the implicit linked list, for example.

That all being said, this could be included as part of 2.0 of course.

lschuermann, 2020-04-02:
A few days back we had an interesting discussion on Slack regarding sharing buffers between userspace and kernel. I agree that this gives headaches, but solving the potential issues may yield some good benefits for Tock. This is a significant discussion and Slack doesn't appear to be the best medium for it. Thereby I have two questions:

- Assuming the issues can be solved, would this be a worthy goal to try to work towards for Tock 2.0?
- What would be an appropriate medium for a formal discussion? ML-thread? GitHub tracking issue?

bradjc, 2020-04-02:
>     * Assuming the issues can be solved, would this be a worthy goal to try to work towards for Tock 2.0?

Would this actually change the interface between userland and the kernel? If so then it is definitely worth discussing now. If it only changes the kernel implementation, then we can definitely still discuss it, but it wouldn't have to be (although it could be) part of 2.0.

>     * What would be an appropriate medium for a formal discussion? ML-thread? GitHub tracking issue? RFC-style PR?

I think email to tock-dev or a RFC issue are the preferred mechanisms.

cbiffle, 2020-04-03:
> As discussed today I would like to raise including a yield-no-wait syscall.

I assume the intent here is to run pending callbacks for the current task without potentially being unscheduled.

What behavior would you intend if code called from one of the callback should yield? I see two options. (I am _assuming_ code called from callbacks can yield today, but I'm not sure I've tried it.)

1. *Don't wait on this yield unless you do:* Yield in a callback works normally despite caller higher on the stack requesting don't-wait. Asking yield not to unschedule me means nothing in the presence of composition, and I can still be unscheduled.
2. *Contagious don't-wait:* Yield in any callback called directly or indirectly from a yield-no-wait also becomes yield-no-wait. (i.e. a task-level flag is stacked at each yield-to-callback transition with the outermost don't-wait winning.) Outermost caller can now reason about things, but the meaning of all other code changes unexpectedly, in ways that may break things.

Neither of these seem good, is there an alternative I haven't considered?

jrvanwhy, 2020-04-03:
> > As discussed today I would like to raise including a yield-no-wait syscall.
> 
> I assume the intent here is to run pending callbacks for the current task without potentially being unscheduled.

Yes. I don't recall all the use cases, but here are two:

1. Prioritization. An application may have some task it wants to do eventually, but which is lower priority than servicing kernel callbacks (e.g. a CPU-intensive step). The application can try `yield-no-wait` before starting that task. A long-running CPU task could also periodically `yield-no-wait` to keep the rest of the process responsive.
1. Side channel masking. An app could use `yield-no-wait` when it doesn't *need* to use the CPU but wants to mask timing/power side channel information.

> What behavior would you intend if code called from one of the callback should yield? I see two options. (I am _assuming_ code called from callbacks can yield today, but I'm not sure I've tried it.)

I can think of two reasons code might call `yield` (with waiting):

1. It is the main event loop and calls `yield` to invoke callbacks.
1. The code is synchronous, and needs a callback to run to progress.

I don't see any reason why a callback would want to invoke the event loop itself, so I presume that situation would only arise when a callback invokes a synchronous function.

> 1. _Don't wait on this yield unless you do:_ Yield in a callback works normally despite caller higher on the stack requesting don't-wait. Asking yield not to unschedule me means nothing in the presence of composition, and I can still be unscheduled.

This doesn't introduce any new problems: a callback invoking `yield` is already problematic  regardless of whether the callback was invoked by `yield` or `yield-no-wait`. In general, mixing asynchronous and synchronous code is difficult and this is just one example of that difficulty.

> 2. _Contagious don't-wait:_ Yield in any callback called directly or indirectly from a yield-no-wait also becomes yield-no-wait. (i.e. a task-level flag is stacked at each yield-to-callback [...]

This seems problematic to me, as it would change the semantics of `yield` out from under the synchronous function. Granted, all it would really do is make the function spin on the CPU rather than yielding, but it still feels like the wrong solution.

> Neither of these seem good, is there an alternative I haven't considered?

jrvanwhy, 2020-04-13:
I just opened issue #1758, which asks the question "do we want to support multi-language apps?". If the answer is yes, and we want both languages to be able to call into the kernel, then we may need some Tock 2.0 changes to make that work.

jrvanwhy, 2020-04-17:
I just opened #1769, which asks for some (relatively small, but backwards-incompatible) changes to process loading to allow for small apps.

vsukhoml, 2020-05-07:
i'm not sure if it was discussed before - i found that current model with syscalls is inefficient. most meaningful operations require 4 syscalls (allow/subscribe/command/yield) on one side, and then 1 register is used to pass only range from 0..4 in 32 bits. I guess there is some justification for this model, but what i found is semantic of allow and command are somewhere intermixed, it's not efficient to deal with syscalls which require several memory operands. My proposal - make syscall routing flatter, allow more registers.

for example, driver & operation can be encoded in u32, each being u16 - which saves 1 register unless you really expect support for more than 65536 drivers & commands.
Allocate special 'driver id' for kernel, say 0, and change encoding for memop, yield and ipc commands.
Other drivers will just use implement allow/subscribe/command as there commands. 

Extend number of registers used for arguments to 6 or 7. It's fine on RISC-V, and is ok on ARM - just syscall handler will store these extra regs on stack before passing to actual implementation.

consider putting syscall number at the register not used by C ABI - say t6 on RISC-V or r6 on ARM - this way once routing is done, syscall implementation call wouldn't require register shuffling to move data. 

It's not about changing semantic of operations, just more efficient encoding of those. using more registers will enable combined syscalls where allow & command would be in same operation.

lschuermann, 2020-05-07:
@vsukhoml I agree. The current syscalls could probably have larger payloads or map onto common operations better.

> will enable combined syscalls where allow & command would be in same operation

How would we then decide which operation to do first? I think it's important to keep `allow` and `command` cleanly separated from the driver point of view, as `allow` has very clear semantics and gets processed in the kernel (so there's no way for a capsule to access non-`allow`ed memory). Therefore, `allow` and `command` need to be routed to a driver individually.

I might want to first allow a buffer for sending to UART and then send it, or I might want to send the remaining buffer contents and then un`allow` it. Either the order is specified in the combined syscall as well, or one operation must resort to a separate `allow` call.

> driver & operation can be encoded in u32, each being u16

This would even work with my proposed changes in #1735. We are probably never going to have `2^16` driver instances on a board. However we'd then still be able to keep the old assignments and the full `u32` space for driver types (along with categories etc.)

phil-levis, 2020-06-14:
> i'm not sure if it was discussed before - i found that current model with syscalls is inefficient.

Does this matter? I'm wary of premature optimization. I.e., if we optimize in one way now, for an abstract workload, then we might be unable to optimize later, to satisfy timing or efficiency requirements on a real workload. If we have a concrete workload this is a problem for, great! Then let's use that as our benchmark which we need to satisfy. But I'd like to keep Amdahl's law in the back of my mind: optimizing something that's a tiny fraction of execution time won't have a big impact.

So, do we have a workload/use case where:
  1. it's an I/O buffer operation, such that it requires an `allow`, `subscribe`, and `command`,
  2. low latency or high throughput is required,
  3. the overhead of 3 system calls is significant (say, more than 10%).

Happy to move the sliders there if they seem a little off.

The only use case that comes to mind to me might be encryption, or, more likely, hashing/HMACing from userspace with small buffers. But there might be others? Can we nail down a concrete case?

jrvanwhy, 2020-08-07:
The threat model requires the Console (and possibly other drivers) to be mux'd. Is that something we are doing in Tock 2.0?

phil-levis, 2021-02-15:
Closing this because we now have #2429 .

closed 2021-02-15 by phil-levis