tock/tock #1626: Support deploying kernels via OpenOCD

jrvanwhy, 2020-02-25:
I'd like to deploy a kernel to the nrf52840-DK using OpenOCD rather than Segger's JLink tools. `tockloader` supports using OpenOCD already, but the Makefiles under `boards/` don't give us a way to pass the flags through to `tockloader`.

We should allow users of the `board/` Makefiles to program using OpenOCD.

jrvanwhy, 2020-02-25:
I *think* this can be accomplished in a rudimentary manner for most of the boards by changing `TOCKLOADER_JTAG_FLAGS =` to `TOCKLOADER_JTAG_FLAGS ?=` in the Makefiles.

Would anyone object to a PR that makes that change?

ppannuto, 2020-02-26:
I'm happy with a change to the conditional

bradjc, 2020-03-18:
We could add a `make flash-openocd` target (or some other name).

jrvanwhy, 2020-03-18:
Due to COVID-19, I am currently working remotely, and do not expect to return for a minimum of 3 weeks (probably much longer). I cannot work on this until I return to my desk, as that is where the dev hardware is.

jrvanwhy, 2020-03-18:
To be clear, I intend to do this eventually, but either someone else needs to do the work or it is shelved for now.

commit f40419e6 (2020-07-20): Merge #2026

2026: boards: nrf52840dk: add flash-openocd r=ppannuto a=bradjc

### Pull Request Overview

This pull request adds a make target to the nrf52840dk makefile. Fixes #1626.

### Testing Strategy

Flashing a kernel on nrf52840dk.

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
PR #2026 (2020-07-20): boards: nrf52840dk: add flash-openocd
### Pull Request Overview

This pull request adds a make target to the nrf52840dk makefile. Fixes #1626.

### Testing Strategy

Flashing a kernel on nrf52840dk.

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
closed 2020-07-20 by bors[bot]