tock/tock #1634: Question on DMAChannel::enable

fpoli, 2020-02-28:
Hi! While looking for usages of unsafe code in the Rust ecosystem, I randomly stumbled upon this line

https://github.com/tock/tock/blob/4268a351d40bf55d9507efdee78cec5be22df26e/chips/sam4l/src/dma.rs#L234

Could you help me understanding the logic here? In particular
1. Shouldn't the test be `num_enabled == 0`, to identify the first time that `DMAChannel` is enabled? `atomic_xadd` returns the value before the increment, so `num_enabled == 1` is true the _second_ time `DMAChannel` is enabled.
2. Why are `HSBClock` and `PBBClock` enabled anyway at the beginning of the `enable` method, independently from the `num_enabled` test?

It's the first time that I look at Tock, so it's possible that I'm missing some context.

fpoli, 2020-02-28:
Also, if the `DMAChannel` is meant to be used by multiple threads I think that by concurrently calling `enable` and `disable` it's possible to reach a state where `self.enabled` is false but `num_enabled` is greater than zero.

phil-levis, 2020-02-28:
Good question! This code is 5 years old, written by Amit. I've assigned it to him.

alevy, 2020-03-12:
@fpoli this seems like a bug to me.

Could be worth both switching the intrinsic out for the stable version in `core::atomic` as well as fixing the bug and testing it.

Thanks for finding this!

commit 96e62dac (2020-03-26): Merge #1699 #1701

1699: SAM4L: Update DMA peripheral driver r=alevy a=bradjc

### Pull Request Overview

Fixes #1634

### Testing Strategy

hello_loop on hail still works.

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make formatall`.

1701: Travis: Report Code Size and Memory diffs for each PR r=alevy a=hudson-ayers

### Pull Request Overview

This pull request adds two scripts and modifies Travis so that PRs are automatically updated with "statuses" (shown at the bottom of the PR [... cut]
PR #1699 (2020-03-26): SAM4L: Update DMA peripheral driver
### Pull Request Overview

Fixes #1634

### Testing Strategy

hello_loop on hail still works.

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make formatall`.
closed 2020-03-26 by bors[bot]