tock/tock #1641: Process::last_syscall is never updated

gendx, 2020-03-02:
I realized that the `last_syscall` field of `Process` is currently always left to the default value (`None`).

This got me thinking:

- Should this field rather contain a `RingBuffer` of the last N syscalls? When the last syscall is `yield`, it's not very useful to only get a history of 1 syscall.
- Should registering the last syscall(s) be gated by a flag in the `kernel/src/config.rs`, to avoid overhead when not needed? Or should the length of the ring buffer queue be configurable there for more flexibility?

bradjc, 2020-03-02:
Nice catch. I traced this back to https://github.com/tock/tock/commit/351c22cf0e4bfabdd4aff77217afa2611ae2f703 where I refactored things and this accidentally got dropped. PR on the way to at least restore the expected functionality.

Arguably, we should just remove storing last_syscall since the strace functionality is a lot more useful.

ppannuto, 2020-03-02:
> Nice catch. I traced this back to 351c22c

I think my instinct is that strace is something that will be turned on when
needed as it’s relatively high overhead, while this makes sense to leave on
by default. I think they serve slightly different use cases.

bradjc, 2020-03-03:
I have never used the last_syscall when debugging a process, and now that strace exists I don't see why I would in the future.

Has anyone used last_syscall to debug and does anyone expect to in the future?

gendx, 2020-03-03:
I've found it useful when debugging the debugging itself, in particular to try to understand what was going on with https://github.com/tock/tock/issues/1651, which completely froze the console output. The scope may be limited, so I would see it as a disabled-by-default option in `kernel/src/config.rs`.

alistair23, 2020-03-03:
I also find it useful as it indicates how far an app has progressed by knowing the last syscall. I have seen lots of cases where just enabling the syscall debug increases the binary size too much that it creates other problems so I often leave it disabled even when I would like it on.

bradjc, 2023-09-05:
Please re-open if this is still useful.

closed 2023-09-05 by bradjc