tock/tock #1652: [RFC] Panic buffer

gendx, 2020-03-03:
While debugging https://github.com/tock/tock/issues/1651, I've found it useful to be able to write to some ring buffer that is only printed in the panic handler. Together with a panic button (https://github.com/tock/tock/issues/1640), this makes it much easier to debug components related to the regular debugging itself (e.g. Segger RTT which relies on alarms).

### Proposal

Adding a `debug_panic!` macro to `kernel/src/debug.rs`, quite similar to `debug!` in terms of usage, except that instead of writing to a (virtual) UART immediately, it just appends the bytes to a ring buffer. However, contrary to `debug!`, new bytes continue to be appended when the buffer is full, so that the contents are always the most recent N bytes.

Then, inside the panic handler, the current contents of this ring buffer are printed in the diagnostics (so, the last N bytes are shown in the panic diagnostic).

### Configuration

Given that such a buffer would be useful in specific cases (in general one can use the regular `debug!`), I'd suggest to have the size of this buffer configurable in `kernel/src/config.rs` to avoid overhead in the general case. A buffer size of 0 would completely disable it.

bradjc, 2020-03-03:
This took me a second to wrap my head around, but could actually be very useful for embedded systems debugging when you don't want the runtime overhead of UART printing during some other execution. 👍 

But just like the main.rs file configures other debug parameters (like which pins to use for GPIO debugging), I think the configuration for this should also be passed in from the board's main file.

gendx, 2020-03-04:
> But just like the main.rs file configures other debug parameters (like which pins to use for GPIO debugging), I think the configuration for this should also be passed in from the board's main file.

Yes definitely! I'll send a pull-request.

commit 7e47e6f1 (2020-03-11): Merge #1667

1667: Implement a debug_enqueue! macro that writes to a ring buffer flushed only upon panicking. r=gendx a=gendx

### Pull Request Overview

This pull request fixes #1652.

### Testing Strategy

This pull request was tested on an nRF52840-DK board, writing some data with `debug_enqueue!` and triggering a panic to observe the result in the console:
- by adding a `DebugQueue` component in the board,
- without setting the `DebugQueue` buffer.

Overflowing the ring buffer and using formatting was also tested:

```rust
let buffer = static_init!([u8; 128], [0; 128]);
DebugQueueComponent [... cut]
PR #1667 (2020-03-11): Implement a debug_enqueue! macro that writes to a ring buffer flushed only upon panicking.
### Pull Request Overview

This pull request fixes #1652.

### Testing Strategy

This pull request was tested on an nRF52840-DK board, writing some data with `debug_enqueue!` and triggering a panic to observe the result in the console:
- by adding a `DebugQueue` component in the board,
- without setting the `DebugQueue` buffer.

Overflowing the ring buffer and using formatting was also tested:

```rust
let buffer = static_init!([u8; 128], [0; 128]);
DebugQueueComponent::new(buffer).finalize(());

for i i [... cut]
closed 2020-03-11 by bors[bot]