tock/tock #1654: Building Tock with stable rustc

mcy, 2020-03-03:
This issue tracks attempting to build as many Tock libraries as possible with a stable compiler, that is, minimizing the number of `#![feature(...)]` pragmas used in `lib.rs` files.

This work is especially attractive to OpenTitan's use of Tock as a secure embedded operating system. Being able to build code without exercising unstable (and, as such, untested and unproven) code paths in the compiler increases confidence in the language-level and hardware-level guarantees provided by Tock. This is not to mention that unstable features are not guaranteed to be stabilized, or to remain in the compiler indefinitely.

This work is divided up according to pragmas still present after https://github.com/tock/tock/pull/1643, which every pragma that could be removed without causing build failures. Below is a list of all pragmas Tock still uses.

- [x] (#1643) Remove all pragmas currently not required to build Tock.
- [ ] (#????) `asm` and `naked_functions`. Assembly is a hard requirement for any kernel, it is possible to move all of it into `.S` files instead. This will require build system work to invoke LLVM's assembler through `clang` or similar. There have been attempts to stabilize this recently (which I have been involved in); driving stabilization forward is also an option, though the stable syntax is likely to be incompatible with the current one, which is just a Clang/GCC dialect.
- [ ] (#????) `core_intrinsics`. Intrinsics will never be stabalized, and are subject to unannounced changes by rustc's developers. We use three classes of intrinsics:
  - `core::intrinsics::offset`, which is just pointer arithmetic. We should be using `<*mut T>::offset`, which is guaranteed to have the same effect and the same optimization characteristics.
  - `core::intrinsics::atomic_*`, various atomic intrinsics. These will be very difficult to get rid of due to how Rust decides to provide `AtomicT` types, and, in the case of `kernel`'s `AtomicUsize` will require consultation with an atomics expert.
  - `core::intrinsics::sqrtf32`, a square root intrinsic. `f32::sqrt` is not present in `core` even though it just calls the intrinsic. This is because on many platforms, the intrinsic expands into a `libm` call, though not on any platforms Tock is built for. Removing this may require implementing a polyfill in assembly.
- [ ] (#????) `lang_items`. The main purpose of this pragma is declaring the global panic handler in a `no_std` binary. Once `no_std` binaries are stabilized, we will have an upgrade path out of this pragma; until then, we are very, very stuck with it.
- [ ] (#????) `const_fn`. Most of the pertinent parts of this feature are stable, though one remains: `const` functions with a bound type parameter. `OptionalCell<T>` uses this extensively, and will be very difficult to remove. However, stabilizing `const_fn` is a priority (as far as I can tell) for Rust, so it may be worth waiting it out.
- [ ] (#1646) `in_band_lifetimes`. This is syntax sugar that makes otherwise undeclared lifetimes become defined, by adding lifetime declarations where necessary. In other words, the transformation `fn foo(_: &'a T)` -> `fn foo<'a>(_: &'a T)`. The "placeholder lifetime", `'_`, is a stable feature that acts as an in-band lifetime which does not bind a name, and which was originally part of `in_band_lifetimes`. Since the stabilization of `'_`, the stabilization of `in_band_lifetimes` has been deprioritized, and is fairly unlikely to happen in its current form.
- [x] (#1649) `concat_idents`. Used in just one macro that doesn't actually need it. Given that it is basically unusable, Rust is very reluctant to ever stabilize it.
- [ ] (#1650) `crate_visibility_modifier`. Allows `crate` in item position as sugar for `pub(crate)`. Interest in this feature appears to have died down after the 2018 module system shakeup, and provides no value other than saving five characters.
- [ ] (#1648) `ptr_internals`. Allows use of `core::ptr::Unique`. This is an internal of `Box` that will never be stabilized. `core::ptr::NonNull` is a drop-in replacement.
- [ ] (#1647) `exclusive_range_pattern`. Allows use of `a..b` as a pattern. Only used in one place (used incorrectly, too).
- [ ] (#????) `associated_type_defaults`. Allows declaring defaults for associated types in trait definitions. We only use it in two places, so it's easy to get rid of. Only partially implemented in the compiler right now, but seems to be nearing completion? https://github.com/rust-lang/rust/issues/29661
- [ ] (#????) `panic_info_message`. Used in one place by `kernel` to dump panic information. Might want to get rid of it, since it seems the API might get tossed out in favor of something else: https://github.com/rust-lang/rust/issues/66745
- [ ] (#????) `untagged_unions`. Allows putting non-Copy traits in `union`s. Using this is basically asking for weird dtor behavior and general unsoundness. We use it in one place, and can probably just plug the hole with the stable `DropManually` API (which is just an `untagged_union` without sharp edges).

alevy, 2020-03-03:
Thanks @mcy, this is excellent!

pfmooney, 2020-03-03:
Indeed.  Thanks for your work on this.

> asm and naked_functions. Assembly is a hard requirement for any kernel, it is possible to move all of it into .S files instead. This will require build system work to invoke LLVM's assembler through [...]

If there are any other features which mandate sticking with nightly in the short-to-medium term, I hope we would keep `asm` and `naked_functions` so this code could remain inline until the "native" rust option stabilizes.

mcy, 2020-03-04:
Regarding inline assembly, there seems to be some degree of contention on whether inline assembly is even a good idea, in principle (I mostly lean towards the `.S` side but that's because I think GCC/Clang constraint syntax is hot garbage). I still need to look into pushing stabilization forward; I was very involved in the most recent effort to specify a syntax (I yelled at them very loudly to explicitly support RISC-V).

We can answer the question of inline assembly later; plowing through the other bullet points will probably take me a while, and I think the fate of inline assembly is RFC-worthy.

bradjc, 2020-03-04:
The tension in the initial pull requests is, I think, getting at a deeper issue than just switching toolchains. Pushing towards using stable represents a change in how Tock views Rust. When we started, just creating _any_ architecture for an OS for embedded systems using Rust that an embedded engineer would understand and find reasonable was a challenge, given the disconnect between how embedded software is traditionally written and Rust's ownership model. But, we thought it was worth seeing if Rust could provide any benefit. As Tock developed, the attitude of "let's see what tools Rust can give us" persisted, and I think that has manifested as a willingness to embrace new features of the language and explore how they can be used, like:
- Do they help with sharing code between embedded platforms?
- Do they make it easier for a C programmer to understand Rust?
- Can we remove Tock-specific code and use Rust-provided abstractions instead?
- Can we eliminate `unsafe` and the chance for panics?

So, the question might be, is now the right time to say we expect to be able to leverage fewer Rust features to our benefit, and are better served with the stable toolchain? Or is Rust continuing to push with new features that will make it easier to develop Tock, and we shouldn't miss out/delay their use for months? Or has Tock progressed enough that its goals on this issue should change?

--------
My opinion has been that if we can't use stable anyway (because of assembly), then we shouldn't be in this halfway state where we _have_ to use the nightly toolchain, but then also don't get to try out any of its new features. Now, however, if there is a reasonable roadmap to switching to the stable compiler then it might be worth making the change. But, I'm not sure that March 2020 is the deadline to say that if a feature has not been stabilized by now, then Tock cannot use it. I would rather wait until a switch to stable is imminent. Maybe this could be part of Tock 2.0?

gendx, 2020-03-04:
Thanks for setting up the list and starting cleaning up the low-hanging fruits!

My comment would go in the opposite direction, but is also quite hypothetical at this stage. Anyway, since we're discussing unstable features here, what about *adding* more features in the future?

I'm mostly thinking about [const generics](https://github.com/rust-lang/rust/issues/44580) (which are still highly unstable and not usable for many cases yet, hence the "hypothetical"). However, I see them bringing value in Tock as well.

In particular, because dynamic memory allocation is restricted in Tock, many structures are defined as constant or static arrays of a `const` size. This means that either the size is fixed and unconfigurable, or we have to pass slices around to support tuning the size for each board or to quickly modify it. Slices have their own set of problems, notably that a suitable lifetime (usually `'static`) has to be passed around, and that the Rust compiler limits what's possible w.r.t. statics and w.r.t. borrowing.

Here are a couple of example where I see them bringing value in Tock

- It was proposed recently in #1585 to have IPC array take a const generic size.
- Another example that comes to mind is the [RingBuffer](https://github.com/tock/tock/blob/master/kernel/src/common/ring_buffer.rs), which currently takes a slice. It'd be useful to have ring buffer incorporating their own array, but that requires either separately creating a static array (which can be quite tricky given Rust's borrowing rules and limitations around statics), using something like `Pin` to be able to have a self-referential struct, or using const generics. Here are a couple examples.

```rust
// Self-referential struct
struct SelfRingBuffer<T> {
    // Easy to change the size.
    array: [T; 16],
    // Intrusive ring buffer which should reference &self.array.
    ring_buffer: RingBuffer<T>,
}

// Const-generic struct.
struct FixedRingBuffer<T, const N: usize> {
    array: [T; N],
    head: usize,
    tail: usize,
}
```

In many cases, macros could be used to simulate const-generics, but I don't think that'd be clearer (e.g. worse error messages, not the same level of type checking).

Of course, this is still highly hypothetical as const-generics are [not nearly stable enough](https://github.com/rust-lang/rust/labels/A-const-generics), but on the other hand given the value they will bring to the Rust ecosystem as a whole, I don't see them as a feature being abandoned by the community - and they will likely eventually be stabilized (although no idea when).

So my question is: what's Tock position regarding adding new unstable features in the future? What bar (current and future stability? value proposition?) should an unstable feature pass to be used in Tock?

ppannuto, 2020-03-04:
> So my question is: what's Tock position regarding adding new unstable features in the future? What bar (current and future stability? value proposition?) should an unstable feature pass to be used in Tock?

I think historically the answer has been "no bar". If it's in the officially released nightly compiler, it's fair game. We've viewed Tock as a project written in the language "unstable Rust" so anything in the official language (as defined by the compiler that exists for it) is valid.

gendx, 2020-03-04:
> I think historically the answer has been "no bar". If it's in the officially released nightly compiler, it's fair game. We've viewed Tock as a project written in the language "unstable Rust" so anything in the official language (as defined by the compiler that exists for it) is valid.

I think we should be a bit more nuanced than "whatever is possible today is valid". We regularly update the compiler version (e.g. due to compiler bugs, or to adopt more recent features). I don't think Tock should end up in a situation where:

- Some `shiny_feature` is available on week 1.
- On week 2, a new compiler has `more_shiny_feature`, supports CPU architecture `awesome` and fixes a critical bug in the nightly compiler, but `shiny_feature` isn't available anymore.

What should we do? Stay stuck on week 1's compiler? Rollback all the changes to migrate to week 2?

This is of course an extreme example, and the Rust compiler has generally been good at avoiding such situations even in nightly (especially given how many projects rely on nightly for the unstable features that it provides). But the more features we use, the riskier it is to end up in such a situation.

Another possible problem is if the nightly of day D has a critical bug, because it's nightly.

For better or worse, Rust is sometimes slow at stabilizing features, which makes nightly more appealing to many projects. But trimming down to "essential" features only (with a minimal bar for allowed features) can reduce the risks, especially as Tock grows to more stakeholders (more developers, more architectures/boards, more applications).

gkelly, 2020-03-04:
> I think historically the answer has been "no bar". If it's in the officially released nightly compiler, it's fair game. We've viewed Tock as a project written in the language "unstable Rust" so [...]

I believe that there's a strong argument to be made from a safety standpoint for using stable rust.  I believe the unstable features receive significantly less testing than the core stable set, just by virtue of fewer programs ever getting written against them.  I also worry about features rotting.  Just because it was safe at the original time of implementation to depend on that feature I don't feel comfortable depending on ones who have no clear path to inclusion in rust-stable.  If the feature does break the only recourse we have is to fix it after the fact.

pfmooney, 2020-03-04:
> * [ ]  (#????) `untagged_unions`. Allows putting non-Copy traits in `union`s. Using this is basically asking for weird dtor behavior and general unsoundness. We use it in one place, and can probably [...]

Looking at the one piece of code using this, it seems we could solve the specific case by extending tock-registers to support a dual-type mode for device registers which have different behavior/formats for read and write.

phil-levis, 2020-03-05:
> My opinion has been that if we can't use stable anyway (because of assembly),

I'll go on the record saying I've always been against inline assembly. I'd prefer to just write assembly, follow the C ABI, and then import these functions into Rust as C functions. This is what I do in my (non-Tock) embedded systems work.

bradjc, 2020-03-09:
Any thoughts on `try_trait` https://github.com/rust-lang/rust/issues/42327? That proved to be very useful in #1480.

alevy, 2020-03-09:
> Any thoughts on try_trait rust-lang/rust#42327? That proved to be very useful in #1480.

@bradjc I _believe_ this has a very high chance of becoming stable in the not far future.

bradjc, 2020-03-31:
Any thoughts on `cell_update` https://github.com/rust-lang/rust/issues/50186? It is used in #1718.

alistair23, 2020-03-31:
`cell_update` looks useful, but we can always just `take()` and then `replace()` the `Cell` value so it doesn't let us do something we couldn't already.

lschuermann, 2020-04-01:
Indeed. However it let's us write code in a way more aligned to what's commonly done with `TakeCell` and `MapCell`. I often tend to forget the `set()` part of the Cell update. It's only development ergonomics, and I'm fine either way - with or without the feature.

lschuermann, 2020-04-01:
I'm using `associated_type_defaults` https://github.com/rust-lang/rust/issues/29661 in #1727 too. Can be easily removed though, if it doesn't become stable in the near future.

ppannuto, 2020-05-22:
The question of stable rust was discussed on the core call today. Replicating the salient points from #1646:

 - Tock should start pushing towards stable Rust as policy.
 - This **does not** currently prohibit the use of unstable features.
 - However, features that we use should be on at least short-to-medium term path to stabilization.
 - Priority for work on unstable feature removal by the core team will go towards the "fundamental" blockers (i.e. the new asm interface)
 - However, we are happy to take in work from others towards the removal of unstable features that do not seem to be on the path to stabilization

----

As an example of what I think we should do moving forward, here is the note I included on #1822, which introduces the use of a new unstable feature: 

```rust
// Feature used to opt-in the new `core::Option::contains()` API.
//
// This feature can be removed if needed by manually reimplementing the
// `contains` logic for `Option`.
//
// Tock expects this feature to stabilize in the near future.
// Tracking: https://github.com/rust-lang/rust/issues/62358
#![feature(option_result_contains)]
```

gendx, 2020-06-09:
> * `core::intrinsics::sqrtf32`, a square root intrinsic. `f32::sqrt` is not present in `core` even though it just calls the intrinsic. This is because on many platforms, the intrinsic expands into a [...]

The `sqrt` function is currently [only used](https://github.com/tock/tock/search?q=sqrtf32) by the `tmp006` capsule, which I [don't see used](https://github.com/tock/tock/search?q=tmp006) in any board by default. So I see several options to get rid of it.

1. Have the capsule manually compute the square root, with one of the [many available manual methods](https://en.wikipedia.org/wiki/Methods_of_computing_square_roots).
1. Gate the definition of the `kernel::common::math::sqrtf32` function by a feature, that one has to manually opt-in when they want to use the `tmp006` capsule.
1. Depend on the [`rust-lang/libm` crate](https://github.com/rust-lang/libm), which has pure-Rust implementation of most math functions (including [sqrtf](https://github.com/rust-lang/libm/blob/master/src/math/sqrtf.rs)). This is an additional dependency for Tock, but there is a [roadmap](https://github.com/rust-lang/libm/milestone/2) to ultimately add it to libcore - although all items of the roadmap seem to already be implemented so I'm not sure what's left to get there.

In the long run, the latter option is probably the way to go as a solution supported by Rust.

hudson-ayers, 2020-06-09:
> > * `core::intrinsics::sqrtf32`, a square root intrinsic. `f32::sqrt` is not present in `core` even though it just calls the intrinsic. This is because on many platforms, the intrinsic expands into a `libm` call, though not on any platforms Tock is built for. Removing this may require implementing a polyfill in assembly.
> 
> The `sqrt` function is currently [only used](https://github.com/tock/tock/search?q=sqrtf32) by the `tmp006` capsule, which I [don't see used](https://github.com/tock/tock/search?q=tmp006) in any board by default. So I see several options to get rid of it.
> 
> 1. Have the capsule manually compute the square root, with one of the [many available manual methods](https://en.wikipedia.org/wiki/Methods_of_computing_square_roots).
> 2. Gate the definition of the `kernel::common::math::sqrtf32` function by a feature, that one has to manually opt-in when they want to use the `tmp006` capsule.
> 3. Depend on the [`rust-lang/libm` crate](https://github.com/rust-lang/libm), which has pure-Rust implementation of most math functions (including [sqrtf](https://github.com/rust-lang/libm/blob/master/src/math/sqrtf.rs)). This is an additional dependency for Tock, but there is a [roadmap](https://github.com/rust-lang/libm/milestone/2) to ultimately add it to libcore - although all items of the roadmap seem to already be implemented so I'm not sure what's left to get there.
> 
> In the long run, the latter option is probably the way to go as a solution supported by Rust.

The relevant rust tracking issue for adding math to core is here: https://github.com/rust-lang/rfcs/issues/2505 , and reading that issue it also identifies that the rust libm crate is much less efficient than the c version used by intrinsics, which takes advantage of floating point instructions if they are available (any cortex-m4f should have these, though I don't think we have the FPU enabled on anything yet).

Personally, I don't think it is worth adding our first external dependency to support a function in a capsule that no Tock boards use. I would prefer to replace it with a software square root implementation, and then revisit the issue when a more compelling use case arises.

gendx, 2020-06-09:
A relevant part from rust-lang/rfcs#2505 may be the following:

> Also, as it's not possible to use inline assembly (`asm!`) on stable the `vsqrt.f32` instruction would have to be invoked via FFI and an external assembly file.

Given that Tock using stable Rust will already depend on stable inline assembly of some form, we could reasonably expect that by then this constraint will also be removed from the `libm` crate. It'll still be a lot of work on the `libm` side to dispatch to the most efficient implementation depending on the exact CPU architecture, but there is hope to have a pure-Rust and efficient solution one day.

Besides, it's unclear to me how critical the speed of computing square roots is to the `tmp006` capsule, so even the current non-optimal implementation in `libm` could be sufficient. And if we use a software sqrt implementation for the capsule, it will likely be similar to [sqrtf.rs in libm](https://github.com/rust-lang/libm/blob/master/src/math/sqrtf.rs) anyway.

If we don't want to explicitly depend on an external crate, we can also paste the relevant `sqrtf.rs` file in the `tock/libraries` folder, as we already do for `enum_primitive`.

brghena, 2020-06-09:
It's so funny that complaints about the TMP006 are coming up again. I was trying to do a good, "modern" thing when I wrote the driver and just actually use math functions and floating point numbers since microcontrollers now (often) have FPUs. But it's come back in discussions about whether floating point should exist in the kernel, code size for the `powf` function that was also originally used, and now core intrinsics.

While I do generally believe we should be able to use `sqrt` or other math functions in the kernel and that we shouldn't be implementing them ourselves, I also don't think it's worth putting effort in now if the tmp006 is the only driver using it. The TMP006 chip itself is obsolete and deprecated, and this driver was written for the Firestorm board we originally started Tock with, which is itself not available anymore. I think a perfectly valid option would be to just delete this driver from Tock. If we do want to keep the driver, just having a custom `sqrt` function within it seems like the easiest solution, but it's going to be hard to test...

jrvanwhy, 2020-06-09:
I have no opinion on deleting vs. supporting TMP006, but I wanted to chime in on this point:

> If we do want to keep the driver, just having a custom `sqrt` function within it seems like the easiest solution, but it's going to be hard to test...

A modern desktop machine can iterate through all possible inputs to a 32-bit square root function in a few minutes, allowing us to compare our own implementation to the libm-provided implementation. I would expect a hand-written 32-bit square root function to be portable, so we can test on a amd64 machine and use it in embedded.

brghena, 2020-06-09:
Good call! I was thinking in terms of modifying the TMP006 driver, which is going to be hard to test since we don't support it on any board anymore.

bradjc, 2020-07-15:
Does anyone know how we can support `DefferedCall` (which uses AtomicUsize) on earlgrey (rv32imc) without `core_intrinsics`?

hudson-ayers, 2020-07-17:
> Does anyone know how we can support `DefferedCall` (which uses AtomicUsize) on earlgrey (rv32imc) without `core_intrinsics`?

(I might be exposing myself here as not understanding something as well as I should).

Why does `DefferedCall` require atomics? I don't think I was particularly involved at the time they were introduced, but here is my understanding of how they work:

1) They are only set from within the kernel

2) They are never set from interrupt context

3) They are never read from interrupt context

4) We only ever have 1 kernel thread, so effectively all of these things are done from the same thread.

So why is it necessary that setting deferred calls or servicing them requires atomics? Can we not just use a cell?

hudson-ayers, 2020-08-10:
I updated the top level comment with my understanding of the current state of things. To summarize remaining major blockers:

1. Transition all `llvm_asm!()` --> `asm!()`, and wait for `asm!()` to stabilize. Ideally, we should encourage new contributions to use the new macro.

2. `core_intrinsics`. To remove all uses of atomics in the Kernel, we need to modify `DeferredCall` to not use atomics. This should be safe to do given the design of the Tock kernel, but will require careful review.

3. `const_fn`. #2069 presents a design that makes removal of almost all uses of `const_fn` possible. But the tock register interface still requires it, and removing its use there may be difficult, and is IMO the only remaining open question on the journey to stable.

I think all other remaining features could be removed without too much effort, but deserve to be used for now in the event that they become stabilized before these blockers are resolved.

hudson-ayers, 2022-07-15:
Update of major blockers to stable Rust:

1. `core_intrinsics`. As before, I still believe this could be safely removed with just some implementation effort.
2. `asm_sym` + `naked_functions`: Removing either of these would be very difficult, but both have been nominated for stabilization (and `naked_functions` has been approved for stabilization), so these are no longer much of a concern.
3. `asm_const`: Only one use remains after #3083 is merged. This use in in `riscv_csr`. Removing that use would require going back to either the switch statement approach to CSRs (e.g. before https://github.com/tock/tock/pull/2472/files), or revisiting the macro-based approach that Alistair suggested (https://github.com/tock/tock/pull/2470/files), but is certainly technically possible at the expense of that file being harder to maintain.

Everything else is either in `boards/` or an option passed in our Makefile, and as such does not prevent any downstream user from using stable Rust, and as such less important to resolve.