tock/tock #1657: Running `make ci-travis` sometimes triggers "unused attribute" errors.

gendx, 2020-03-04:
While developing, I run `make ci-travis` to check my changes. Sometimes, that fails due to my code being work-in-progress, but running `make ci-travis` then triggers warnings and ultimately errors of the following kind:

```
error: unused attribute
  --> /tock/libraries/tock-cells/src/volatile_cell.rs:20:5
   |
20 |     #[inline]
   |     ^^^^^^^^^
   |
   = note: `-D unused-attributes` implied by `-D warnings`
```

Running `make clean` before re-running `make ci-travis` seems to fix the issue.

A weird thing is that there are no warnings at all on a clean repository, but somehow the warnings show up in a "dirty" one, where they turn into errors.

- As far as I remember, this is the fist time `make clean` is needed. I guess this now answers @ppannuto's question on https://github.com/tock/tock/pull/1377#issuecomment-533617734.
- This particular warning seems to come from upstream's https://github.com/rust-lang/rust/pull/65294.
- I have no idea why `make ci-travis` is not idempotent w.r.t. this warning, but this is a bug. I'm not sure where it comes from, it could be Tock's building infrastructure, Cargo, or the specific nightly compiler that we currently use.
- Enforcing warnings everywhere (https://github.com/tock/tock/pull/1568) would be helpful in enforcing stricter code hygiene.

Note that `cargo --version` and `rustc --version` match those found in Travis logs, so there doesn't seem to be a problem with `rust-toolchain` in my case.

gendx, 2020-03-06:
Am I the only one observing this?

I consistently get these errors, and the trigger path seems to be the following.

```shell
# First, make sure to have a clean repo.
make clean
# The first run works normally.
make ci-travis
# Change a file to trigger a rebuild.
# No need to actually change anything, touching is enough.
touch kernel/src/debug.rs
# The second run triggers warnings/errors about `#[warn(unused_attributes)]`.
make ci-travis
```

niklasad1, 2020-03-06:
That should be quite easy to fix with one the following solutions:

1. use `#![cfg(all(target_arch = "arm", target_os = "none"))]` in the file and don't repeat the conditional compilation for each function
2. use `#[cfg_attr(all(target_arch = "arm", target_os = "none"), inline(always))]` instead

gendx, 2020-03-06:
I'm still wondering why it only happens on the second run. This seems to indicate an issue with the Makefiles and/or Cargo.

niklasad1, 2020-03-06:
Alright, TLDR it properly.

I also saw a bunch faulty `unused attribute`, such as when using `nightly-2020-02-27`:

```   
--> tock/chips/nrf52/src/uart.rs:292:5
    |
292 |     #[inline(never)]
```

I also tested `rustc 1.43.0-nightly (96bb8b31c 2020-03-05)` and those warnings didn't appear. Thus, I think it is just a bad nightly version.

gendx, 2020-03-06:
> Thus, I think that we are just using a bad nightly version.

I could reproduce it at 8e544f40a07684b9fa824d5b2f74120b4ea2d5a0 (latest update of `rust-toolchain`), whereas it's not happening at the parent commit (3a7d6b775d972798bfd731cba8365b58fab27175). So I agree with this analysis.

bradjc, 2020-03-06:
Is there a rust issue for a fix? Does it still happen on the latest nightly? https://github.com/tock/tock/compare/update-rust-2020-03?expand=1

niklasad1, 2020-03-06:
Couldn't find any issue for it, seem to affect 26-28 Feb.

Latest nightly 2020-03-06 works fine on my machine.

commit 4d680ade (2020-03-07): Merge #1670

1670: update rust to 2020-03-06 r=niklasad1 a=bradjc

### Pull Request Overview

Update Rust version to fix #1657.

The life of running on nightly, I guess :) Probably an argument for #1654

### Testing Strategy

This pull request was tested by travis

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make formatall`.
PR #1670 (2020-03-07): update rust to 2020-03-06
### Pull Request Overview

Update Rust version to fix #1657.

The life of running on nightly, I guess :) Probably an argument for #1654

### Testing Strategy

This pull request was tested by travis

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make formatall`.
closed 2020-03-07 by bors[bot]