tock/tock #1663: [Tracking] Optimizing Tock for code size

gendx, 2020-03-05:
Given the recent interest in optimizing Tock for code size, I think it would make sense to summarize and track existing and future ways of decreasing the size.

- [x] **Capsule configuration in boards.** This is a well-established way of trimming unused code. Despite dozens of capsules being available, each board can pick those that are relevant. By default, board configurations include many drivers as a baseline, but one can easily remove unused drivers to trim down Tock's size for their particular use case.
- [x] (#1443) **Kernel's const configuration.** By tuning `const` parameters in `kernel/src/config.rs`, once can enable "niche" debugging features, such as syscall tracing or loading processes. When turned off, the Rust compiler can fold the constants and remove dead code.
- [ ] (#1660) **Disabling frame pointers.** This is work-in-progress to decrease code size by 2-10%. Frame pointers might be useful for debugging, but we could add configuration for it.
- [ ] (#1372) **Stripping debug statements.** The idea would be to replace the `debug!` macro's generated code by an empty block, under some configuration option (e.g. the kernel's const configuration). This would remove code in itself, as well as a bunch of strings. The compiler could then further optimize the rest of the code. A more aggressive pruning strategy would also remove the UART-related capsules under such as "no-debug" mode.
- [ ] (#1662) **[Unclear] Static mut doesn't help inlining?** Because the compiler can assume that static mutable data is modified elsewhere, it may not optimize (inlining, constant folding) expressions that are in fact constant.

gkelly, 2020-03-05:
I think removing the debug statements (or making a pluggable logging driver that could do things like printing offsets into `.rodata` and format arguments, for example) might be a big win, especially if results in us completely removing the `fmt` machinery.

ppannuto, 2020-03-05:
Re the first checkbox, there is still some work to do:
https://github.com/tock/tock/issues/625

gendx, 2020-03-05:
> Re the first checkbox, there is still some work to do: #625

Thanks for the pointer! I added it to the list.

alevy, 2020-03-12:
RE: debug statements, one "trivial" way to strip code down by a lot is to remove modify the `panic` handler for a board with one that doesn't output any formatted strings.

E.g., on the Hail board, the following change:

```diff
diff --git a/boards/hail/src/io.rs b/boards/hail/src/io.rs
index 4aef525e0..a3c6a5aab 100644
--- a/boards/hail/src/io.rs
+++ b/boards/hail/src/io.rs
@@ -61,12 +61,13 @@ pub unsafe extern "C" fn panic_fmt(pi: &PanicInfo) -> ! {
 
     let led_red = &mut led::LedLow::new(&mut sam4l::gpio::PA[13]);
     let writer = &mut WRITER;
-    debug::panic(
+    /*debug::panic(
         &mut [led_red],
         writer,
         pi,
         &cortexm4::support::nop,
         &PROCESSES,
         &CHIP,
-    )
+    )*/
+    loop {}
 }
```

saves 20k of code size (from 90k to 70k)

gendx, 2020-03-13:
> RE: debug statements, one "trivial" way to strip code down by a lot is to remove modify the `panic` handler for a board with one that doesn't output any formatted strings.

Definitely! I added it to the list as well.

I also think that it would be useful to have a more principled way of doing it (e.g. via some config parameter) rather than just commenting out the code (likewise for stripping `debug!` statements), so that when it becomes useful one doesn't have to remember which parts to comment out (which also has implications on unused variables, that need to be renamed to avoid compilation errors).

bradjc, 2020-03-13:
Need more investigation on the frame pointers, since removing them seems to break cortex-m4 support.

mcy, 2020-04-06:
@bradjc How would you feel about enabling this just for the RISC-V boards? After all, the frame pointer win is miniscule on ARM but big on RISC-V, which seems otherwise unaffected.

bradjc, 2020-04-06:
> @bradjc How would you feel about enabling this just for the RISC-V boards? After all, the frame pointer win is miniscule on ARM but big on RISC-V, which seems otherwise unaffected.

That sounds good to me.

vsukhoml, 2020-04-15:
where the need for relocatable kernel (-C relocation-model=dynamic-no-pic ) come from? it impacts code size and inhibits linker relaxation (use of __global_pointer and (gp) based addressing on RISC-V). 

Also, as a side note - cargo binutils's objdump is inferior compare to gcc's:
1) not all information is displayed with -dCS:
gcc's objdump:
   92754:	00000097          	auipc	ra,0x0
   92758:	b2a080e7          	jalr	-1238(ra) # 9227e <core::slice::<impl [T]>::split_at>
   9275c:	4a8d                	li	s5,3
cargo's:
   92754: 97 00 00 00                  	auipc	ra, 0
   92758: e7 80 a0 b2                  	jalr	-1238(ra)
   9275c: 8d 4a                        	addi	s5, zero, 3

2) demangling doesn’t always work:
gcc binutils:
00092638 <kernel::tbfheader::parse_tbf_header_lengths>:
000926aa <kernel::tbfheader::parse_tbf_header>:
00092b6e <<kernel::process::ProcessLoadError as core::convert::From<kernel::tbfheader::TbfParseError>>::from>:

cargo binutils:
00092638 kernel::tbfheader::parse_tbf_header_lengths::hc07c65444fc61568:
000926aa kernel::tbfheader::parse_tbf_header::h7d932551f5f43602:
00092b6e _$LT$kernel..process..ProcessLoadError$u20$as$u20$core..convert..From$LT$kernel..tbfheader..TbfParseError$GT$$GT$::from::had00201fb076e8e

Combined with loss of debug information in .elf file, it's very hard to trace where code comes from. I found that --emit=asm gives better tracing, though less convenient.

vsukhoml, 2020-04-15:
When I removed -C relocation-model=dynamic-no-pic - it seems there is some redistribution between .text and .data, but simple kernel i tried worked. It didn't enable gp-based addressing on RISC-V though.

vsukhoml, 2020-04-16:
it seems [linker relaxation](https://content.riscv.org/wp-content/uploads/2019/03/11.15-Shiva-Chen-Compiler-Support-For-Linker-Relaxation-in-RISC-V-2019-03-13.pdf) for RISC-V doesn't work in llvm/ld. When adding -C target-feature=+relax - i don't see any traces of using gp-based addressing in the code. Actually using -C target-feature=-relax results in different code than not having this option at all, which possibly points to issue in linker (or unsupported functionality? https://reviews.llvm.org/D77694)

lenary, 2020-04-16:
Linker relaxation is not supported in LLD. It is supported by binutils ld, which should be tried.

For gp-based addressing, you also need to define `__global_pointer$` in the same output section as the data you want to address, and you have to initialize gp to its address at program startup.

vsukhoml, 2020-04-20:
Yes, i actually tried with __global_pointer$ pointing to .data and to .rodata to see where it works better. With binutils ld there is another issue that negative offsets are encoded as positive:
compare encodings (binutil's objdump) for:
 8484c:	1001a103          	lw	sp,256(gp) # 13100 <dir_entries+0xf8>
and:
848b4:	1021a023          	sw	sp,256(gp) # 12900 <task_state_current>
both put +256
but refer to different objects
first one is correct with gp = 0x13000
second is not - it should be -2048+256
So,  it only works if __global_pointer$ is at start of data segment for now. 

llvm's ld also generates crappy map files which I usually use to check values of constants, sizes of code/data, etc. When LTO is enabled there are many 0-sized symbols with no names, which makes it hard to analyze:
   84410    84410        0     1
   84410    84410        0     1                          
   84410    84410        0     1
   84410    84410        0     1
   84410    84410       1c     1                 _start                         
   8442a    8442a        0     1
   8442c    8442c        0     1
   8442c    8442c        0     1
   8442c    8442c        0     1
   8442c    8442c       54     1         . = DEFINED ( _start_trap ) ? ALIGN ( 128 ) : ALIGN ( 1 )

lenary, 2020-04-21:
> Yes, i actually tried with __global_pointer$ pointing to .data and to .rodata to see where it works better. With binutils ld there is another issue that negative offsets are encoded as positive: [...]

Can you send me a link to a branch with code where this happens? If nothing else, it would be useful to have a reproducer to file against binutils ld for this bug.

> llvm's ld also generates crappy map files which I usually use to check values of constants, sizes of code/data, etc. When LTO is enabled there are many 0-sized symbols with no names, which makes it [...]

I would need more time to look into this issue - I'm not familiar with LLD to point to something immediately. 

I'm interested to know how you have enabled LTO for RISC-V (or have I got my wires crossed). This is not supported by LLVM yet, for some complex reasons - but we're getting there with support for it.

vsukhoml, 2020-04-21:
> Can you send me a link to a branch with code where this happens? 
This is on non-Tock based home-grown project. I didn't try well to make ld work with Tock :)  let me try to make a simple reproducible sample out of that.  

> I'm interested to know how you have enabled LTO for RISC-V
i didn't know that it doesn't work... just enabled lto=true in cargo and it seems did the trick. So far it was a bring-up for the board and it worked, though suppose there can be other issues. You can also look at opentitan board.

vsukhoml, 2020-04-22:
>  let me try to make a simple reproducible sample out of that.

I found creating that sample challenging. on simple example generated code was correct. need more investigation what was specific in mentioned example.