tock/tock #1676: How to use TockOS on the board not in the support list of Tockloader?

bling206, 2020-03-09:
Hi 

I have a HiFive1 board and want to try the applicatrions on TockOS. However, all the methods about how to install an applicaton on TockOS are using the Tockloader. How to install an application without Tockloader ?

Thank you.

alistair23, 2020-03-09:
You will have to include the app as part of the binary that you are flashing.

There are steps here on how to do it for another board: https://github.com/tock/tock/blob/master/boards/nucleo_f446re/README.md#flashing-app

bradjc, 2020-03-09:
If you want to run a single app, you need to flash the rv32i.tbf (I think that is what it is called for risc-v) file to the correct address in flash on your board. Any flashing tool that works should be able to do that.

Tockloader also takes a number of arguments for customizing how it uses tools like openocd, so if you are using openocd or jlinkexe then you might be able to pass options to tockloader to make it work in your case.

bling206, 2020-03-11:
Hi 

Thank you for replying
I still have some questions.
1. If I download the kernal on the github  ,  just go to the board directory  and make a BIN files, will this BIN files include the example application such as blink? Or I need to download the libtock-c and make the example BIN files and then flashing it on my board?
2. Is the command "tockloader install blink"  to load the appication to ram and run it if this applicaton is already included in the Bin files of flash?  If Tockloader doesn''t support my board, how do I run the applicaiton?

Thank you.

alistair23, 2020-03-11:
1. No, that will just be the Tock kernel (without any apps). You need to load the apps. I posted a link earlier about `objcopy`ing it to the binary file you flash to the board. That won't require Tock loader, instead Tock will just load the app when it starts. You can also use Tockloader which supports the HiFive1 (earlier I thought it didn't).
2. I don't think so, Tockloader should copy the file over. It looks like Tockloader does support the HiFive1 though, as you can see here: https://github.com/tock/tockloader/blob/master/tockloader/board_interface.py#L76

bling206, 2020-03-11:
Hi Alistair23

Thank you.
I just installed tockloader last week , but I can't see HiFive1 in my support list like the picture below: https://imgur.com/a/sfa0ej2.
I will try to reinstall it.

bling206, 2020-03-11:
After I uninstall tockloader(pip3 uninstall tockloader) and reinstall again( pip3 install tockloader --user), the support list remains the same which does not support HiFive1.  

By the way, how does TockOS would know to load which APP to run when it starts if there is more than one app installed in TockOS

Thank you.

alistair23, 2020-03-11:
Maybe it's not in the released version. Just install Tockloader from the git repo.

If you include more then 1 app it will load all of them. I doubt you will be able to load more then 1 on the hifive1 though as it's too resource constrained.

bling206, 2020-03-12:
I try to install it from the github, but there is some error messages about "unknow type https".
I'm trying to figure it out.
Anyway, thanks a lot for answering questions. It helps a lot.

bling206, 2020-03-12:
After I update my Python to Python3.7, I can build tockloader from the github successfully and I have HiFive1 in the support list.
Thank you.

bling206, 2020-03-12:
Although I see https://github.com/tock/libtock-c/issues/44 that there was no way to build a RISCV app, is there any way now?
Thank you.

alistair23, 2020-03-12:
You can't with mainline `libtock-c`. You can try my branch here: https://github.com/alistair23/libtock-c/tree/alistair/riscv

bling206, 2020-03-13:
I download your branch and try to make example but there are errors about"/bin/sh: 1: -Wbad-function-cast: not found" and "make: riscv32-oe-elf-ar: Command not found".
I thought that it might be my riscv toolchain not official .(I download it from SiFive website). Then, I go to https://github.com/riscv/riscv-gnu-toolchain and make the toolchains. However, it doesn't work.

bling206, 2020-03-13:
I find that RISCV.md  mentions about to upate Configuration.mk where toolchain is, but I have no idea where to modify in Configuration.mk.

alistair23, 2020-03-13:
I must have some hardcoded values for the OpenEmbedded toolchain.

The SiFive toolchain is not the official toolchain and neither is the RISC-V fork. Please just use upstream GCC and friends. If you aren't using the OE toolchain you will need to change some names. Just grep around and change the names in their to paths that you are using.

bling206, 2020-03-16:
After I change the names of my toolchains , the error messages about "make:riscv32-oe-elf-ar: Command not found" disappear, but the messages about "bin/sh: 1: -Wbad-function-cast: not found" remains the same, like the picture, https://imgur.com/a/VjTaTGr.

alistair23, 2020-03-16:
It looks like you still aren't setting the toolchain correctly then.

In future please attach text logs and not pictures.

ppannuto, 2020-04-09:
Can you paste the text output of `make V=1`, that will report what all of the variables are being set to.

In this case, the `CC` variable, which controls the compiler command, seems to be blank. The first argument given to `CC` is `CFLAGS`, which begins with `-Wbad-function-cast`. The error  is coming from the missing `CC`

closed 2020-06-03 by bradjc