tock/tock #1679: Transparency and policies: approved vs. merged pull-requests?

gendx, 2020-03-10:
There are currently 14 approved pull-requests in the [backlog](https://github.com/tock/tock/pulls), some of which have been so for many days (or even a week) and/or don't seem any controversial. Given the recent growth in the number of contributors and pull-requests throughput, I don't think this is a healthy state of development.

In the context of transparency and policies, I think it would be good to clarify the rules that allow a pull-request to go from the approved state to the merged state, so that developers know what to expect when sending a pull-request. Without a policy, it may demotivate developers who spent time working on a pull-request and addressing the review comments to make it reach an approved state, but then may feel that it has been "forgotten" in the approved state.

Also, some future pull-request(s) can build upon a current pull-request, which means that holding back on merging the currently approved pull-request blocks the future pull-request(s) from happening (notably to avoid having to handle merge conflicts).

Some ideas:
- Should there be at least N approvals? For now, N=1 shows the pull-request as approved in GitHub's UI, but GitHub should allow to increase the threshold if a lack of diverse reviews is the most common issue.
- Depending on the affected directory, should some specific "maintainers" give approval? E.g. the `kernel/` is maintained by reviewers X and Y, the `capsules/` by X and W, etc.? So any pull-request affecting `kernel/` should be approved by at least X or Y.
- Should an approved pull-request stay in the "last call" state for N days after approval in order to be merged? Or until the next core talk? Some pull-requests are definitely more complex than others and a "last call" vs. merging right away makes sense, so there should be some flexibility, but having a policy - and making sure to consistently tag such pull-requests as "last call" and respect the agreed-upon N days of delay - would again be helpful.

ppannuto, 2020-03-10:
We do have a written policy on this that I we have been imperfect in
adhering to:
https://github.com/tock/tock/blob/master/doc/CodeReview.md

While we have sort of stopped explicitly labeling things P-Upkeep,
P-Significant tags are still serving their intended role. We've just fallen
into an 'upkeep unless explicitly mentioned otherwise' model (note: the
argument against this likely still holds, we should be proactively
identifying things as Upkeep or Significant).

With that background, I believe the current policy in short is: Upkeep N=1
+ no minimum time, Significant N=2 + minimum 7 day (waived if N=all of
core).

---

Sometimes, I approve things but don't want to merge them immediately only
because I want to give others a little time to chime in if it makes sense.
It would be good to codify this behavior pattern somehow. I have made use
of "d+" with bors sometimes; I think this is a mechanism we can/should lean
on more.

bradjc, 2020-03-10:
14 _seemed_ high, but looking closer the number is more realistically 7 (which is still a lot). Seven approved PRs are actually:

- 1 Merge conflict
- 1 Very recently approved
- 3 With unaddressed/pending comments
- 2 Not approved by any core team member

I tend to not merge nontrivial PRs if I am the only approver.

This isn't a super helpful comment, but I agree that it is hard to keep track of the state of PRs once the count is > 10.

gendx, 2020-03-10:
> We do have a written policy on this that I we have been imperfect in adhering to: https://github.com/tock/tock/blob/master/doc/CodeReview.md

Thanks for the pointer! My bad for not checking the docs properly...

> While we have sort of stopped explicitly labeling things P-Upkeep, P-Significant tags are still serving their intended role. We've just fallen into an 'upkeep unless explicitly mentioned otherwise' [...]

The labels could definitely be useful if the number of pending pull-requests remains as high as the current peak.

> Sometimes, I approve things but don't want to merge them immediately only because I want to give others a little time to chime in if it makes sense. It would be good to codify this behavior pattern [...]

This can also be useful.

> 14 *seemed* high, but looking closer the number is more realistically 7 (which is still a lot).

I definitely agree.

This answers my questions, so I'm leaving the issue open for just a few days to give the chance for others to chime in, otherwise will close it. :)

closed 2020-03-13 by gendx