tock/tock #1682: Building Rust's standard library from source

mcy, 2020-03-12:
At the moment, building a Tock board just links in the copy of the standard library (well, just the libcore part) that rustup downloads. In particular, this means it is not built with `-C opt-level=z`.

However, thanks to an unstable cargo flag, we can do this. Passing `-Z build-std=core` in `boards/Makefile.common` causes cargo to build libcore in the same optimization configuration as everything else. We also need to provide memcpy, memcmp, and memset implementations, since LLVM expects those to exist.

Applying all of this to the opentitan board... resulted in a size increase of about a kilobyte. This seems quite bad, and hopefully means I've done something wrong.

See: https://github.com/rust-lang/rfcs/pull/2663

alevy, 2020-03-12:
> Applying all of this to the opentitan board... resulted in a size increase of about a kilobyte. This seems quite bad, and hopefully means I've done something wrong.

Is it possible that's somehow from the implementations of `memcpy`, `memcmp`, `memset`?

Otherwise this is pretty surprising.

Having this as an option certainly makes sense, so thanks for exploring.

mcy, 2020-03-12:
I did some more digging. It seems like almost all of the bloat is from... formatting functions. I'm not kidding. I did a diff of symbols that have different lengths across both binaries. The addr1/len1 side is with -Z build-std=core, while the addr2/len2 side is head. Note that the vast majority of the size increase is from symbols only present when building libcore ourselves, while the majority of changes in existing symbols are size decreases.

Have a giant wall of data.
```
   addr1/   addr2     len1/    len2 delta name
2000be52/???????? 00000510/???????? 01296 __udivmoddi4
2000db16/???????? 00000198/???????? 00408 anon.bdfa12fed23b7e9d886db6439dafafd9.75.llvm.5773251433037376239
2000d8f0/???????? 0000013a/???????? 00314 anon.bdfa12fed23b7e9d886db6439dafafd9.72.llvm.5773251433037376239
2000d7cb/???????? 00000125/???????? 00293 anon.bdfa12fed23b7e9d886db6439dafafd9.71.llvm.5773251433037376239
200054e4/???????? 00000108/???????? 00264 core::unicode::printable::check
20004f84/???????? 000000bc/???????? 00188 <core::str::Chars as core::iter::traits::iterator::Iterator>::next
2000da70/???????? 000000a6/???????? 00166 anon.bdfa12fed23b7e9d886db6439dafafd9.74.llvm.5773251433037376239
20004a6a/???????? 00000090/???????? 00144 <core::fmt::num::UpperHex as core::fmt::num::GenericRadix>::digit
20005b8e/???????? 00000078/???????? 00120 core::fmt::Formatter::padding
2000661a/???????? 00000078/???????? 00120 core::fmt::builders::DebugTuple::finish
20004bd4/???????? 00000072/???????? 00114 core::fmt::num::<impl core::fmt::UpperHex for usize>::fmt
20004afa/???????? 00000072/???????? 00114 core::fmt::num::GenericRadix::fmt_int
20004b6c/???????? 00000068/???????? 00104 core::fmt::num::<impl core::fmt::LowerHex for usize>::fmt
200058ea/???????? 0000005e/???????? 00094 core::fmt::getcount
20006122/???????? 00000058/???????? 00088 core::fmt::builders::DebugStruct::finish
2000d779/???????? 00000052/???????? 00082 anon.bdfa12fed23b7e9d886db6439dafafd9.70.llvm.5773251433037376239
2000da2a/???????? 00000046/???????? 00070 anon.bdfa12fed23b7e9d886db6439dafafd9.73.llvm.5773251433037376239
20005948/???????? 00000040/???????? 00064 core::fmt::PostPadding::write
20004e7a/???????? 0000003a/???????? 00058 core::fmt::num::<impl core::fmt::Debug for usize>::fmt.472
2000e870/???????? 00000039/???????? 00057 str.0.llvm.724120425618518905
2000eda0/???????? 00000039/???????? 00057 str.0.1141
20005b0c/???????? 00000036/???????? 00054 <core::str::Chars as core::iter::traits::iterator::Iterator>::count
200051ec/???????? 00000032/???????? 00050 <[A] as core::slice::SlicePartialEq<A>>::equal
2000c394/???????? 00000032/???????? 00050 compiler_builtins::int::shift::Lshr::lshr
2000c362/???????? 00000032/???????? 00050 compiler_builtins::int::shift::Ashl::ashl
20006a1c/???????? 00000028/???????? 00040 core::iter::range::<impl core::iter::traits::iterator::Iterator for core::ops::range::Range<A>>::next.847
200065f4/???????? 00000026/???????? 00038 core::str::traits::<impl core::slice::SliceIndex<str> for core::ops::range::RangeFrom<usize>>::index::{{closure}}.721
200065d4/???????? 00000020/???????? 00032 core::str::traits::<impl core::slice::SliceIndex<str> for core::ops::range::RangeTo<usize>>::index::{{closure}}.720
2000260c/???????? 0000001c/???????? 00028 core::ptr::drop_in_place.33
2000e91d/???????? 0000001b/???????? 00027 anon.3ab98ec4a94ce25548100b5d2e5b18e1.5.llvm.12220577744171697338
20004db8/???????? 0000001a/???????? 00026 core::fmt::num::imp::<impl core::fmt::Display for u8>::fmt
2000e9a0/???????? 00000019/???????? 00025 str.0.1017
20004c5e/???????? 00000018/???????? 00024 core::fmt::num::<impl core::fmt::UpperHex for u32>::fmt
20004dea/???????? 00000018/???????? 00024 core::fmt::num::imp::<impl core::fmt::Display for usize>::fmt
2000be3a/???????? 00000018/???????? 00024 __udivdi3
2000c3dc/???????? 00000016/???????? 00022 __lshrdi3
2000c3c6/???????? 00000016/???????? 00022 __ashldi3
200073c2/???????? 00000010/???????? 00016 core::ptr::drop_in_place.963
20008fc2/???????? 00000010/???????? 00016 core::ptr::drop_in_place.1122
200056fa/???????? 00000010/???????? 00016 core::ops::function::FnOnce::call_once
2000de87/???????? 00000005/???????? 00005 anon.49fff9cda440e45261a3c76e5a2a58c8.14.llvm.9003057635970185926
2000c714/???????? 00000004/???????? 00004 anon.9a48013de569c534a83c6ad935d1d670.0.llvm.5251483950559478545
2000c71c/???????? 00000004/???????? 00004 anon.49fff9cda440e45261a3c76e5a2a58c8.13.llvm.9003057635970185926
2000df1e/???????? 00000003/???????? 00003 anon.ffca1f2bf3bf35cbc827f6f5b39bab8f.26.llvm.10170687242704130045
2000df15/???????? 00000003/???????? 00003 anon.ffca1f2bf3bf35cbc827f6f5b39bab8f.22.llvm.10170687242704130045
2000df24/???????? 00000002/???????? 00002 anon.ffca1f2bf3bf35cbc827f6f5b39bab8f.32.llvm.10170687242704130045
2000df22/???????? 00000002/???????? 00002 anon.ffca1f2bf3bf35cbc827f6f5b39bab8f.31.llvm.10170687242704130045
2000df1c/???????? 00000002/???????? 00002 anon.ffca1f2bf3bf35cbc827f6f5b39bab8f.25.llvm.10170687242704130045
2000df1a/???????? 00000002/???????? 00002 anon.ffca1f2bf3bf35cbc827f6f5b39bab8f.24.llvm.10170687242704130045
2000df18/???????? 00000002/???????? 00002 anon.ffca1f2bf3bf35cbc827f6f5b39bab8f.23.llvm.10170687242704130045
2000d40b/???????? 00000002/???????? 00002 anon.faec563d3e6f00308162078a72da3ec0.0.llvm.10134526121450097957
2000df27/???????? 00000001/???????? 00001 anon.ffca1f2bf3bf35cbc827f6f5b39bab8f.34.llvm.10170687242704130045
[... cut ...]
20004c46/20006766 00000018/00000084 -0108 core::fmt::num::<impl core::fmt::UpperHex for i32>::fmt
2000574e/20005d0a 0000019c/00000214 -0120 core::fmt::write
2000521e/2000537a 000002c6/0000034a -0132 core::str::slice_error_fail
20005988/20004dca 00000184/00000232 -0174 core::fmt::Formatter::pad_integral
200062f6/20005f8c 000002de/000001ec 00242 <core::fmt::builders::PadAdapter as core::fmt::Write>::write_str
20005c06/20005124 00000148/00000256 -0270 core::fmt::Formatter::pad
20005d84/200056e2 00000344/000004e4 -0416 <char as core::fmt::Debug>::fmt
```

gendx, 2020-03-13:
Nice idea! I added it to the list in #1663, hopefully you'll figure out how to decrease the size :)

vsukhoml, 2020-04-19:
i made it work with risc-v board by providing implementation of memcpy/memset/memmove and setting -Z build-std=core. It solved a problem with missing debug symbols and I was able to trace what contributes to size. My findings are that most stuff comes from core::fmt and Display trait. One idea to solve it is to move conversion to text from target to host by using custom terminal application to decode binary stream. That what we used in home-grown OS to get rid of many strings and conversions from firmware.

ppannuto, 2020-04-21:
> One idea to solve it is to move conversion to text from target to host by using custom terminal application to decode binary stream.

We've talked about this before, but it really harms the accessibility of the platform. We generally try to minimize the _requirements_ for using custom tooling (i.e. tockloader makes life easier, but you can use anything that can copy bytes to get kernels and apps on to boards). I think there would be similar resistance to requiring a custom serial terminal to decode debugging messages [particularly since I imagine that would be complicated to integrate with faux-serial mechanisms like RTT].

vsukhoml, 2020-04-21:
> minimize the _requirements_ for using custom tooling

This is a good point. However, this can be implemented as a configurable feature, so it won't hurt out of the box experience if you don't need it, but will provide means for code size reduction for those in need.

ppannuto, 2020-04-21:
For an out-of-tree platform that wanted to customize, I could see that, but I don't think it is an idea we would take into Tock mainline. Our experience from a combined few decades of embedded OS work is that allowing `ifdef`-like compilation results in an exponential number of build configurations, many of which do not actually work in practice. This is especially true of space-saving measures, where eventually the only way the kernel fits into the required size is if all of them are turned on, and they are no longer truly optional.

bradjc, 2020-04-21:
> One idea to solve it is to move conversion to text from target to host by using custom terminal application to decode binary stream.

This is an interesting idea. I agree with @ppannuto absolutely that we need to keep both the "out of the box" experience and new users firmly in mind when considering something so essential as printf, but there may be a tipping point where a custom serial interface for tock presents so many opportunities it is hard to ignore. In particular, muxing the serial console so that process console doesn't interfere with app printf() would be nice, and it would open up the possibility of numerous onboard debugging consoles which would be a fantastic debugging tool. Add to that doing formatting host-side which could enable significant size wins makes using a custom serial protocol pretty compelling.

I have a habit of comparing Tock to smartphones, and I'm not going to stop now: Android has its logcat tool which is the standard way of getting debug messages from android, and Tockloader could be that same tool for us. It is definitely less convenient than just being able use any serial tool, but if it is easy to install and provides significant value, then it might be worth it.

On the other hand, even Linux provides UART debugging with no magic. Kind of hard to argue with that. Also, TinyOS had [java libraries for communicating with devices](http://tinyos.stanford.edu/tinyos-wiki/index.php/Mote-PC_serial_communication_and_SerialForwarder_(TOS_2.1.1_and_later)), and they were, in my experience, not exactly user friendly.

jrvanwhy, 2020-04-21:
I've also had ideas along the lines of "print a binary stream and decode on the host" for debugging. That said, I think that conversation is off-topic for this issue.

If we'd like to continue discussing that idea, lets open a new issue or a thread on tock-dev.

bradjc, 2021-04-23:
Is this still something Tock should have on its radar?

hudson-ayers, 2021-09-27:
This is quite straightforward today. If we want to build the std library from source, and prefer not to vendor our own memcpy etc., all that is required is to add `-Z build-std=core, compiler_builtins`. (compiler_builtins means you dont have to supply your own memcpy etc.).

I dug some more into the size increases that mcyoung found, and I believe that the size increase is the result of the core library being built with debug information, including panic paths, such that all the formatting functions that panic now have a bunch of path strings to a local path on your machine wherever the core library is located in your rust installation. Using remap_path_prefix to remap the bulk of this path to just `/core/` leads to a binary smaller than without using `-Z build-std=core`. This also gets us back to reproducible binaries (I think, have not tested).

The one disadvantage of using build-std=core is that it is an unstable compiler option with no timeline to stabilization. That said, it isn't really possible for on-device code to depend on this feature being present, so I consider it low harm to include it.

closed 2021-09-30 by bors[bot]

vsukhoml, 2021-10-20:
@hudson-ayers, it doesn't look like memset of opt-level=3 is optimized to use word copies. It just became a bit larger and probably a bit slower.
Sure, code is trivial and basically replica from C code: https://gist.github.com/vsukhoml/7002ce73da372d060dd1ad549cae56e9, written to be compiled into code I like under opt-level=z - had to tune specific ways to write expressions to avoid unnecessary register spills, but this probably changes with compiler advances. This version is not needed for x86 where simple rep movsb is now optimized and versions which use SSE/AVX registers can offer better performance, but it is generic to use register size copies on all platforms. I didn't have bandwidth to propose it to compiler builtins https://github.com/rust-lang/compiler-builtins/tree/master/src/mem, which seems to provide optimized version only for x86_64 unlike say newlib (risc-v example - https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libc/machine/riscv/memcpy.c;hb=HEAD).

commit 5448e8b0 (2021-09-30): Merge #2847

2847: use -Zbuild-std=core to build the std library with our optimization settings r=bradjc a=hudson-ayers

### Pull Request Overview

This pull request modifies the Makefile to pass `-Z build-std=core, compiler_builtins` to Cargo. This tells Cargo to rebuild the std library using our (size focused) optimization settings, and also means that the std library is built with debug information, which makes tracing panics in core library code easier and makes debugging using GDB easier. It uses remap-path-prefix to reduce the code size overhead of core library panic paths being included [... cut]
PR #2847 (2021-09-30): use -Zbuild-std=core to build the std library with our optimization settings
### Pull Request Overview

This pull request modifies the Makefile to pass `-Z build-std=core, compiler_builtins` to Cargo. This tells Cargo to rebuild the std library using our (size focused) optimization settings, and also means that the std library is built with debug information, which makes tracing panics in core library code easier and makes debugging using GDB easier. It uses remap-path-prefix to reduce the code size overhead of core library panic paths being included in the binary, and to try to keep builds re [... cut]