tock/tock #1685: Release 1.5

bradjc, 2020-03-13:
We are targeting a new release of Tock for March 2020. The following tracking issues contain the milestones that will be included in the release:

- [ ] #995 (Tracking: /arch to New Register Interface)
- [ ] #1464 (Tracking: Kernel Documentation Round 2)
- [ ] Further RISC-V support
  - [ ] Add HiFive1(b) board
- [ ] Add size tool to Travis-CI

As we get closer to all of the above checked then we will decide which open PRs to include before testing the release.

## Testing

To test the release we use the following applications and run them on all applicable boards and ensure they execute successfully. This list should be copied for each platform and will be tested once a release candidate is ready.

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.4, and might work for 1.5.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
 

- Hail
  - [ ] `examples/tests/hail`
  - [ ] `examples/witenergy`
  - [ ] `examples/accel_leds`
  - [ ] `examples/ble-uart`

- Miscellaneous
  - [ ] Test the apps compiled at 1.0
  - [ ] process console works as expected
  - [ ] restarting apps works as expected

## Sign off

Once the release candidate has been tested there will be check boxes for each maintainer to sign off on release 1.5.

gendx, 2020-03-18:
I see that the list of milestones to include in Tock 1.5 was discussed in last week's call, which I couldn't participate in (Friday evening is an inconvenient time in Europe).

I have another item to add to the list:

- [ ] Correctly implement alarms. There are multiple pending bugs related to alarms. I remember at least: #1691, #1651 and #1513.

I see timers and alarms as an essential component of Tock, upon which multiple capsules depend on. IMO, forcing users to migrate to Tock 2.x to have functional alarms would essentially mean that the 1.x branch is in fact not stable (and should instead be more clearly named as 0.1.x or something like that). For example, Segger RTT debugging via USB on nRF52840-DK totally depends on it.

As discussed in multiple places, this won't be an easy fix, involving multiple design decisions (muxing virtual alarms, setting an alarm a few ticks from now, wrapping on 24-bit timestamps, specifics of MachineTimer on RISC-V https://github.com/tock/tock/pull/1521#issuecomment-580381788), but it's not less essential.

If this is too big of a refactoring for Tock 1.5, I'd suggest creating a Tock 1.5.1 or 1.6 version back-porting the fix of the alarms.

cc @phil-levis @alevy.

gendx, 2020-03-18:
I'd also add that fixing #1697 (corrupted console output) would be nice (if it indeed comes from Tock).

alevy, 2020-03-18:
@gendx re: timers

I'm in favor of fixing alarms in either a 1.5, 1.5.1 or 1.6 release _as long as it doesn't change the system call interfaces_. This is not to argue that these interfaces are somehow precious or necessarily the _right_ design, but instead that if we're going to change major interfaces those should go into 2.0.

I hear your point about not forcing people to migrate away from a major version in order to get simply correct functionality, but I think we're still at a sweet spot in the "adoption" curve where more or less everyone that might be affected is in my recent e-mail history :)

Nonetheless, let's see what we can do in the confines of 1.x releases, since I agree that timer bugs are really really problematic.

re: console, I agree.

bradjc, 2020-03-18:
What implementation is required for 1.x patch related to timers? Is there a tracking issue? It seems like adding timer stack changes would make it difficult to do a 1.5 release soon. But it's hard for me to tell.

gendx, 2020-03-18:
> What implementation is required for 1.x patch related to timers?

An implementation that fixes the known bugs (at least #1691, #1651 and #1513).

bradjc, 2020-03-18:
Added. Like all "release blockers", we can always push them past 1.5 if necessary.

ppannuto, 2020-04-08:
Can I suggest that we resolve https://github.com/tock/tock/issues/630 one way or another for 1.5 as well?

alevy, 2020-04-16:
Having a real heck of a time compiling and getting C apps to run on the HiFive qemu board, but the kernel seems to work fine.

bradjc, 2020-04-20:
Ok 1.5-rc1 is tagged: https://github.com/tock/tock/releases/tag/release-1.5-rc1. I know we are waiting on some hifive documentation, but I don't want to block testing on that.

Testing commit: 32e58617fb753eeddcd1de0837eb5a8c65e013cd

If you are willing to test for a board, please copy the list of application tests from the first comment in this thread into a new comment with the name of the board, and check off tests as you verify they work as expected.

bradjc, 2020-04-20:
# Hail

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.4, and might work for 1.5.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
 

- Hail
  - [ ] `examples/tests/hail`
  - [ ] `examples/witenergy`
  - [ ] `examples/accel_leds`
  - [ ] `examples/ble-uart`

- Miscellaneous
  - [ ] Test the apps compiled at 1.0
  - [ ] process console works as expected
  - [ ] restarting apps works as expected

alistair23, 2020-04-20:
# OpenTitan

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.4, and might work for 1.5.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send

hudson-ayers, 2020-04-20:
# Imix

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.4, and might work for 1.5.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
 

- Imix-specific
  - [ ] `examples/tests/imix`
  - [ ] `examples/ble-uart`
  - #### In-Kernel Tests
  - [ ] `boards/imix/src/test/aes_ccm_test.rs`
  - [ ] `boards/imix/src/test/aes_test.rs`
  - [ ] `boards/imix/src/test/i2c_dummy.rs`
  - [ ] `boards/imix/src/test/icmp_lowpan_test.rs`
  - [ ] `boards/imix/src/test/ipv6_lowpan_test.rs`
  - [ ] `boards/imix/src/test/linear_log_test.rs`
  - [ ] `boards/imix/src/test/log_test.rs`
  - [ ] `boards/imix/src/test/rng_test.rs`
  - [ ] `boards/imix/src/test/spi_dummy.rs`
  - [ ] `boards/imix/src/test/spi_slave_dummy.rs`
  - [ ] `boards/imix/src/test/udp_lowpan_test.rs`
      - [ ] `start()`
      - [ ] `start_rx()`
      - [ ] `start_with_app()`
      - [ ] `start_dual_rx()`
  - [ ] `boards/imix/src/test/virtual_uart_rx_test.rs`

gendx, 2020-04-21:
Since almost no tests have been reported yet, could we update the `rust-toolchain` for this release?

ppannuto, 2020-04-22:
# nrf52840dk

---

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.5 (console capsule not fully virtualized), and might work for 1.6.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0

bradjc, 2020-04-22:
I just tagged RC2: https://github.com/tock/tock/tree/release-1.5-rc2

Please use this for testing. The changes from RC1 are pretty minor, but double checking that risc-v platforms aren't affected would be good.

hudson-ayers, 2020-04-22:
I am finding that the catch-all `Imix` app only works when the app is first flashed, or after unplugging the board and plugging it back in. running `tockloader listen` or pressing the reset button causes it to stop working. I also see this behavior on past releases, and on two different Imix boards (a v2.0 and a v3.0c). I feel I must be doing something wrong, as this test passed for previous releases, but I also seem to remember us having troubles with this app in particular in the past...

The primary issue seems to be the bluetooth driver failing after a reset. I suspect some hardware state is being cleared by a power cycle that is not cleared by the reset handler.

brghena, 2020-04-22:
@hudson-ayers Does imix have a GPIO connected to reset the nRF51822? Otherwise, it'll get into a bad state and you'll have to powercycle the board each time you want to run something that interacts with it. Resetting via the button probably only resets the SAM4L. On Hail, we have a GPIO line that we toggle in the board setup.

bradjc, 2020-04-22:
Removing R6 on the PCB should fix the issue.

bradjc, 2020-04-23:
# 1.5 Signoff

- [ ] Niklas Adolfsson, @niklasad1
- [ ] Hudson Ayers, @hudson-ayers
- [ ] Brad Campbell, @bradjc
- [ ] Branden Ghena, @brghena
- [ ] Philip Levis, @phil-levis
- [ ] Amit Levy, @alevy
- [ ] Pat Pannuto, @ppannuto
- [ ] Johnathan Van Why, @jrvanwhy

jrvanwhy, 2020-04-23:
(tock-on-titan)[https://www.github.com/google/tock-on-titan] is now on Tock-1.5-rc2 and all our tests pass.

gendx, 2020-04-24:
I'll take a look at updating OpenSK to Tock-1.5-rc2.

bradjc, 2020-04-24:
I just tagged rc3: https://github.com/tock/tock/releases/tag/release-1.5-rc3

Need testing:
- launchxl
- acd52832

Would be nice to have testing:
- arty-e21
- one or more stm32 boards
- nrf52dk, nrf dongle

hudson-ayers, 2020-04-27:
Ran the `virtual_uart_rx_test` on Imix on rc3, with the process console disabled (as requested on Friday's call). The output seems correct (I typed 'a' into the console 14 times):

```
Listening for serial output.
Starting virtual reads.
Starting receive of length 3
Starting receive of length 7
Initialization complete. Entering main loop
Virtual uart read complete: SUCCESS:
61
61
61
Starting receive of length 3
Virtual uart read complete: SUCCESS:
61
61
61
Starting receive of length 3
Virtual uart read complete: SUCCESS:
61
61
61
61
61
61
61
Starting receive of length 7
Virtual uart read complete: SUCCESS:
61
61
61
Starting receive of length 3
Virtual uart read complete: SUCCESS:
61
61
61
Starting receive of length 3
Virtual uart read complete: SUCCESS:
61
61
61
61
61
61
61
Starting receive of length 7
```
cc @phil-levis

lschuermann, 2020-04-27:
> nrf52dk

For some reason took that board with me in Corona times, so I can test in the next 2 days.

alevy, 2020-04-27:
# acd52832

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.5 (console capsule not fully virtualized), and might work for 1.6.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0

bradjc, 2020-04-29:
Last call for testing so we can make our April goal!

@tock/core-wg Please check off https://github.com/tock/tock/issues/1685#issuecomment-618437881 or comment what PRs we need still.

ppannuto, 2020-04-29:
I know @lschuermann has at least started with nrf52 testing from a brief conversation on slack, so hopefully we can get the results of that in before signing off.

rajivr, 2020-04-29:
> Would be nice to have testing:
> 
> * arty-e21
> * one or more stm32 boards

Unfortunately I am stuck in the Bay Area, with my Nucleo board in Bangalore. So, won't be able to test for this release cycle.

hudson-ayers, 2020-04-29:
I unfortunately only have a single nrf52840 board with me, but if someone was able to test the 15.4 tx and rx apps on that board that would be nice (interop with Imix fails, but that is probably future work)

lschuermann, 2020-04-29:
# nrf52dk
(_Explanations shamelessly stolen from @ppannuto's nRF52840DK testing_)

* All Boards
  * [x]  `examples/sensors`
  * [x]  `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    * Fails with error code 16385 as expected (no serialization)
  * [x]  `examples/c_hello` and `examples/tests/printf_long`
  * [x]  `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
  * [x]  `examples/blink`
  * [x]  `examples/rot_client` and `examples/rot_service`
    * [x] With only client: `No rot13 service`
    * [x] With only service:
    * [x] With both: `12: Uryyb Jbeyq!  12: Hello World!`
  * [x]  `examples/blink` and `examples/c_hello` and `examples/buttons`
  * [x]  `examples/lua-hello`
  * [x]  `examples/tests/console_timeout`
  * [x]  `examples/tests/malloc_test01`
  * [x]  `examples/tests/stack_size_test01`
  * [x]  `examples/tests/stack_size_test02`
  * [x]  `examples/tests/mpu_stack_growth`
    One loop:

          ```
          stack: 0x200043e0 - buffer: 0x200043e0 - at_least: 0x   0
           write to 0x200043e0
          read from 0x200043e0
              value 33
          <panic>
          ```
  * [x]  `examples/tests/mpu_walk_region`
     regular loop does not crash, button presses crash both reads as expected
  * [x]  `examples/tests/multi_alarm_test`
     slow, peaceful blinking of each LED in order
  * [x]  `examples/tests/adc`
     `No ADC driver!`
  * [x]  `examples/tests/adc_continuous`
     `No ADC driver!`
  * [x]  `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
     `Number of LEDs: 4` .. then each LED randomly turns on and off
  * [x]  `examples/tests/gpio` with mode set to 0
     Touch `P0.03` to Vdd (randomly tested with a few other pins to _not_ fire an interrupt):

          ```
          GPIO Test Application
          Print userspace GPIO 0 pin reading whenever its value changes
          Jump pin high to test
          	GPIO Interrupt!
          	GPIO Interrupt!
          ```
* nRF specific
  * [x]  `examples/ble_advertising`
    [image]

  * [x]  `examples/ble_passive_scanning`
    Address and data redacted for privacy reasons, addresses cross-checked with Nordic nRF Connect app

        [Tutorial] BLE Passive Scanner
        --------------------------LIST-------------------------
        PDU Type: 0 ADV_IND
        PDU TxAdd: 1
        PDU RxAdd: 0
        PDU Length: 24
        Address: XX XX XX XX XX XX
        Data: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX
        --------------------------END---------------------------

lschuermann, 2020-04-29:
Wondering why the `ble_advertising` and `ble_passive_scanning` did work on the `nRF52DK` but presumably not on the `nRF52840DK` I did a quick check on my own board. I can't reproduce the errors of @ppannuto but I could very well be doing something wrong in my testing methodology. I'd appreciate a third test as an "arbiter", quickly checking whether the BLE examples work or I'm doing something wrong? @hudson-ayers as you have a nRF52840DK, would you mind trying that?

hudson-ayers, 2020-04-29:
I can confirm that `ble_passive_scanning` works completely on my nrf52840dk on release-1.5-rc3 and current libtock-c master.

`ble_advertising` successfully generates advertisements, but my ble scanner app fails to connect to it (that may be a result of my using a cheap app I found on the app store). I do not see the errors printed that @ppannuto saw.

ppannuto, 2020-04-29:
Seems I'll have to debug my hardware / setup at some point; but that's an issue for another day. I'm willing to call two working 52840's enough and have checked the box above.

--- 

Actually, I'm also going to edit that sign-off to the top comment of this issue, because it's a bit annoying to hunt for it in the middle of the chain.

bradjc, 2020-04-29:
# arty-e21

- [x] kernel loads and debug message is printed out
- [x] blink works from libtock-c/riscv
  - note: have to slow down the rate to `delay_ms(5000)`

brghena, 2020-04-30:
What is our promise to libtock-c 1.4 or earlier applications, with regards to backwards compatibility?
The [change to nrf-serialization resetting](https://github.com/tock/tock/commit/262f166d38ee33be9a4148a383ddc887c90ba2ea#diff-8b173947534fbd3252bca3879c4b1cc3L413) means that the 1.0 and 1.4 versions of the `examples/tests/hail` app do not work on kernel release 1.5.

This isn't a problem with the system call boundary, and it really does just effect nrf-serialization, but I thought it would be good to bring up the question.

bradjc, 2020-04-30:
> What is our promise to libtock-c 1.4 or earlier applications, with regards to backwards compatibility?
>
> The [change to nrf-serialization [...]

I think as long as syscall 1.0 APIs still work, that is the only promise we made. The others can evolve over time, and old versions of the apps might not work.

closed 2020-04-30 by bradjc