tock/tock #1692: Persistent Key-Value Store

gkelly, 2020-03-16:
One area that was identified by the Tock on OpenTitan WG is the lack of a persistent key value store (or filesystem) in Tock.  The current model of having apps modify their own flash to store values is very constrained and has a few issues:
- If the kernel and apps are signed as part of a monolithic signature then modifying the app's flash region breaks signing.
- Any update system breaks the storage.  A/B updates with a monolith mean it's a whole new app getting run on the other slot.  In place updates probably overwrite storage unless careful consideration is paid to the section.  Side-by-side updated apps come up as whole new apps.
- Flash topology tolerance.  Most (all?) systems that Tock targets are XIP from flash. During flash erases a bank is usually unavailable for reads/XIP.  A kv-store/filesystem that is aware of flash topology could avoid writing/erasing banks that are currently in use, or even just defer flash layout questions to the platform itself.

To that end, I believe it makes sense to consider a persistent key-value store API for Tock.  I think reasonable goals are to allow the platform to provide an implementation of the lower-level details (bank management, flash controller details) but provide common things like access control (isolating apps to only reading things that they wrote, size limits, for example).

To get the conversation started I'd like to propose an API:
- Read(&str) -> &[u8]
- Write(&str, &[u8])
- Delete(&str)

The keys would be string-named, the data stored is [u8].  The semantics are that no function will return until the corresponding action is visible to subsequent calls referring to that key.  To begin with, I think apps should be isolated from interacting with each other via this API.

bradjc, 2020-03-16:
This is only tangentially related, but Tock's bootloader supports key-value "attributes" that we use to automatically identify what kind of board is connected (https://github.com/tock/tock/blob/master/doc/Memory_Layout.md#flash-1).

Also, a while ago we discussed "kernel attributes" that would be persistent key,value settings or state relevant to the kernel (like the address of apps), rather than the physical board (https://github.com/tock/tock/issues/695). We ended up deciding it wasn't worth the effort to implement. But, if boards have a key-value mechanism already, might be worth revisiting.

labbott, 2020-03-17:
can we make keys also be `&[u8]` ? The extra overhead of utf-8 validation in a str doesn't seem to be beneficial, especially based on other discussions we've had about str overhead in tock already.

gkelly, 2020-03-19:
> can we make keys also be `&[u8]` ? The extra overhead of utf-8 validation in a str doesn't seem to be beneficial, especially based on other discussions we've had about str overhead in tock already.

That's a really good point, thanks for raising it!  I'm definitely on board with `&[u8]` keys.  I can't see any reason why not.

vsukhoml, 2020-03-28:
In order to make it useful in multi-application scenario, filesystem/store should also implement access control lists for keys and persistent ids for applications to define acls. 
Also, regarding keys - does it really make sense to have variable sized bytes? Having a 32-bit integer as a key can be enough, and apps can implement their own mapping of strings if needed at all via simple hashing. Most keys will probably be just a counters from known base.

lschuermann, 2020-04-02:
In solving a similar issue, I have implemented a partitioned storage driver on top of the NonvolatileStorage HIL, taking an identifier of the app and mapping that to a partition. Think of a logical volume manager, but much less dynamic. :) Currently the app requests the partition size once during initialization, but I could imagine using the tock app header for that. I _think_ I've already started implementing transparent encryption support also using #1720, but that was some time ago.

This would actually solve the ACL issue - for accessing data from another application we used IPC, that way we have one data owner controlling access.

I'm not necessarily saying that this solves the same issues as an in-kernel userspace accessible key/value store, but certainly allows one to implement a per app key value store purely in userspace.

If there is interest, I could probably spin up a draft PR in a couple of days.

ppannuto, 2020-04-09:
I'd like to quickly flag the semi-parallel discussion going on on tock-dev for the folks watching this issue: https://groups.google.com/forum/#!topic/tock-dev/aduN7fHWXdI 

Any persistent storage API is going to need to have a sense of persistent application identifiers, which I think we'll need to resolve first

ia0, 2020-05-22:
If permissions to persistent storage are defined in the application header, then application IDs are not required, so we can resume the discussion here.

Let me recap the current state as I see it, and add what I think is missing.

### API

The store is essentially a partial map from keys to values.

#### Keys

We need to define the notion of key we want to use. There are at least 2 solutions: length-bounded slices or bounded integers. The trade-off is that slices are expressive while integers are flash-efficient.

#### Values

Probably everyone agrees that values should be length-bounded slices. The question is on the bound which may be on the order of magnitude of a page.

#### Operations

As described in the first post, one wants at least the following operations (using Rust terminology):
- `get` to test if a key is defined and access the associated value
- `insert` to overwrite the value of a key if present and insert it otherwise (may return the previous value)
- `remove` to remove a key if present and nothing otherwise (may return the previous value)

However, if keys are slices or for optimization purposes if keys are integers, one may also want `iter` to iterate over a range of keys. This could be optimized in many ways to avoid the cost of repeated syscalls (e.g. return batch of handles where a handle is a key and a position hint in the storage with dynamic invalidation support). This operation is useful for applications that want to represent a set using the store. They would reserve a range of keys for the set. They would then need to iterate over the range to know the elements.

One may also want `transaction` to bundle independent updates (`insert` and `remove`) together. And for optimization purposes, one may also want `clear` to remove a range of keys.

Finally, one may also want `capacity` and `lifetime` to get information about the application consumption of those quantities (see Permissions section). This is useful if applications have to report those numbers to the outside world.

### Properties

The store should be power-loss resilient in the sense that if power is lost in the middle of a mutable flash operation (write or erase), the store should be able to recover from the corruption. It may be acceptable if the content is corrupted (new, deleted, or modified values), but see next point.

Ideally, mutable operations (`insert`, `remove`, `transaction`, and `clear`) are atomic (they either succeed or don't modify anything) even when power is lost.

### Permissions

Since Tock aims to support distrustful applications, we need to define a notion of permissions to prevent the following scenario:
- An application accesses the entries of another application without permission (note that the key-value store could be used to share state between applications)
- An application consumes all the store capacity preventing other applications to insert/modify values.
- An application consumes all the lifetime of the flash preventing other applications to insert/modify values.

We would naturally need the following permissions:
- Read access (`get` and `iter`) to a range of keys (cumulable)
- Write access (`insert`, `remove`, `transaction`, `clear`) to a range of keys (cumulable)
- Maximum capacity
- Maximum lifetime

We would store those permissions in the header of applications. We would need 3 distinct mechanisms:
- One to check that applications don't claim permissions they should not have (the header could be signed or something, and this is checked at process creation time)
- One to check that the cumulative capacity and lifetime fit the store (this is done at process creation time)
- One to check that an application respects its permissions (this is done by the driver for each syscall)

closed 2023-08-18 by bradjc