tock/tock #1730: Inefficiencies in the system call path on RISC-V

alphan, 2020-04-02:
I recently took a close look at Tock OS system call performance on OpenTitan using Verilator and noticed some inefficiencies in the system call path.

To measure the system call performance, I implemented a simple capsule with a `command` system call that returns the value of the `mcycle` register and triggered this system call from a libtock-rs application. Both tock and libtock-rs were built with `opt-level = 3` and `lto = true`.

According to the verilator execution trace, roundtrip latency of this system call is 5172 cycles.

Looking at the execution trace and disassembly of the kernel side by side I was able to find a couple of places with room for improvement. I share them below because they seem to be relatively safe:
- In `Process::swtich_to()` (line 1124) using `self.stored_state.take()` instead of `self.stored_state.get()` saves 1225 cycles.
- Skipping MPU configuration if returning to the same process saves an extra 2751 cycles.
- Commenting out debug_* calls in `Process::switch_to()` and `Kernel::do_process()` saves an extra 102 cycles.

I hope I didn't miss anything obvious and please let me know if I can provide any additional information.

bcantrill, 2020-04-02:
Good stuff! @alphan, if you haven't seen it, you might be interested in [Tockilator](https://github.com/oxidecomputer/tockilator), a (very) crude tool we built to facilitate this kind of analysis...

alphan, 2020-04-03:
Thanks @bcantrill! I wish I had seen it before, looks pretty cool! I'll definitely check it out.

gendx, 2020-04-06:
> * Commenting out debug_* calls in `Process::switch_to()` and `Kernel::do_process()` saves an extra 102 cycles.

I've been thinking about having a more principled way of removing debugging statements, see #1372 if you find it useful.

And I've been tracking code-size improvements in #1663. This issue could be generalized to all kinds of performance improvements, or someone could add another tracking issue for saving CPU cycles.

gkelly, 2020-04-07:
I can't assign this to @tock/opentitan-wg for some reason, @bradjc is assigning this to you-in-lieu as the chair of that WG a reasonable alternative?

vsukhoml, 2020-04-15:
i think there is potential to reduce syscall overhead down  to 50-100 cycles by streamlining storing/restoring context and reducing it to RISC-V ABI conventions on syscalls (don't save caller-saved regs).

hudson-ayers, 2020-09-10:
cc #2101 which identifies inefficiencies (~130 cycles) on `allow()` calls and on any system calls that access grant regions.