tock/tock #1736: single-byte console writes never complete

bcantrill, 2020-04-06:
Currently, the <a href="https://github.com/tock/libtock-rs/blob/master/examples/timer.rs">timer example from libtock-rs</a> hangs, after
emitting a single `[`:

```
Version:    opentitan-snapshot-20191101-1-887-ge849d530
Build Date: 2020-04-03, 11:01:10
I00000 boot_rom.c:44] Boot ROM initialisation has completed, jump into flash!
OpenTitan initialisation complete. Entering main loop
[
```

I looked at a trace under <a
href="https://github.com/oxidecomputer/tockilator">Tockilator</a> (and because
it's instructive, I have also pushed everything needed to reproduce this to
the <a
href="https://github.com/oxidecomputer/tockilator/tree/master/example/libtock-rs/timer">example/libtock-rs/timer
subdirectory</a>).

For starters, here is the output showing just system calls:

```
% tockilator ./trace_core_00000000.log
169204   => SYSCALL MEMOP operand=0 (brk) arg0=0x10003004
187307   <= SYSCALL MEMOP operand=0 (brk) arg0=0x10003004
187310   => SYSCALL MEMOP operand=10 (update-stack-start) arg0=0x10003400
202583   <= SYSCALL MEMOP operand=10 (update-stack-start) arg0=0x10003400
202586   => SYSCALL MEMOP operand=11 (update-heap-start) arg0=0x10003004
217281   <= SYSCALL MEMOP operand=11 (update-heap-start) arg0=0x10003004
217358   => SYSCALL MEMOP operand=0 (brk) arg0=0x10003404
234333   <= SYSCALL MEMOP operand=0 (brk) arg0=0x10003404
235019   => SYSCALL COMMAND driver=0 (timer) subdriver=0 arg0=0x0 arg1=0x0
250366   <= SYSCALL COMMAND driver=0 (timer) subdriver=0 arg0=0x0 arg1=0x0
250373   => SYSCALL COMMAND driver=0 (timer) subdriver=1 arg0=0x0 arg1=0x0
265330   <= SYSCALL COMMAND driver=0 (timer) subdriver=1 arg0=0x0 arg1=0x0
265339   => SYSCALL SUBSCRIBE driver=0 (timer) subdriver=0 cb=0x20032006 data=0x1000331c
281047   <= SYSCALL SUBSCRIBE driver=0 (timer) subdriver=0 cb=0x20032006 data=0x1000331c
281059   => SYSCALL COMMAND driver=0 (timer) subdriver=2 arg0=0x0 arg1=0x0
296016   <= SYSCALL COMMAND driver=0 (timer) subdriver=2 arg0=0x0 arg1=0x0
297243   => SYSCALL ALLOW driver=1 (console) subdriver=1 addr=0x100032d4 len=1
314794   <= SYSCALL ALLOW driver=1 (console) subdriver=1 addr=0x100032d4 len=1
314805   => SYSCALL SUBSCRIBE driver=1 (console) subdriver=1 cb=0x20031294 data=0x10003168
330553   <= SYSCALL SUBSCRIBE driver=1 (console) subdriver=1 cb=0x20031294 data=0x10003168
330559   => SYSCALL COMMAND driver=1 (console) subdriver=1 arg0=0x1 arg1=0x0
346462   <= SYSCALL COMMAND driver=1 (console) subdriver=1 arg0=0x1 arg1=0x0
346479   => SYSCALL YIELD
```

So we're doing a one-byte write to the console -- and if the subscribe
callback is called, it's certainly not making any other system calls.  For
starters, it's helpful to confirm this by specifying the Tock application
`timer.elf` binary to Tockilator:

```
% tockilator -e ./timer.elf ./trace_core_00000000.log
...
297041      | write_fmt<libtock::console::Console>
297068      -> write
297111         | iter<core::fmt::ArgumentV1>
297111           | add<core::fmt::ArgumentV1>
297111             | offset<core::fmt::ArgumentV1>
297134         | zip<core::slice::Iter<core::fmt::rt::v1::Argument>,core::slice::Iter<&str>>
297134           | new<core::slice::Iter<core::fmt::rt::v1::Argument>,core::slice::Iter<&str>>
297134             | new<core::slice::Iter<core::fmt::rt::v1::Argument>,core::slice::Iter<&str>>
297134               | min<usize>
297134                 | min<usize>
297134                   | min_by<usize,fn(&usize, &usize) -> core::cmp::Ordering>
297139         | next<core::slice::Iter<core::fmt::rt::v1::Argument>,core::slice::Iter<&str>>
297139           | next<core::slice::Iter<core::fmt::rt::v1::Argument>,core::slice::Iter<&str>>
297148        -> write_str<libtock::console::Console>
297208           | copy_from_slice<u8>
[... 783 lines cut ...]
@@ -219,7 +227,6 @@ impl<'a> MuxUart<'a> {
                     });
                 });
                 node.operation.clear();
-                self.inflight.set(node);
             });
         }
     }
```

With this fix, the libtock-rs timer example runs correctly:

```
Version:    opentitan-snapshot-20191101-1-887-ge849d530
Build Date: 2020-04-03, 11:01:10
I00000 boot_rom.c:44] Boot ROM initialisation has completed, jump into flash!
OpenTitan initialisation complete. Entering main loop
[0] Waited roughly 0.000s. Now is 0.304s = 0x00000be5 ticks (None ticks since last time at 10000 Hz)
[1] Waited roughly 0.500s. Now is 6.438s = 0x0000fb7c ticks (Some(61335) ticks since last time at 10000 Hz)
[2] Waited roughly 1.000s. Now is 12.582s = 0x0001eb7e ticks (Some(61442) ticks since last time at 10000 Hz)
[3] Waited roughly 1.500s. Now is 18.922s = 0x0002e326 ticks (Some(63400) ticks since last time at 10000 Hz)
[4] Waited roughly 2.000s. Now is 24.875s = 0x0003cbb7 ticks (Some(59537) ticks since last time at 10000 Hz)
[5] Waited roughly 2.500s. Now is 31.215s = 0x0004c35f ticks (Some(63400) ticks since last time at 10000 Hz)
...
```

gendx, 2020-04-06:
Nice catch and detailed investigation! I wonder if you've encountered #1651 or #1697 as well?

I think we really need better ways of avoiding these kinds of scheduling logic errors (other example: #1636), maybe with better abstractions and/or unit tests?

gendx, 2020-04-06:
Btw, the frequency seem off on your platform.

> ```
> OpenTitan initialisation complete. Entering main loop
> [0] Waited roughly 0.000s. Now is 0.304s = 0x00000be5 ticks (None ticks since last time at 10000 Hz)
> [1] Waited roughly 0.500s. Now is [...]

There are roughly 6 seconds between loop iterations, whereas the waiting time should be of 500ms. Or is that a side-effect of Tockilator's tracing, which slows down everything?

I don't see any 10KHz frequency in https://github.com/tock/tock/blob/master/kernel/src/hil/time.rs.

bradjc, 2020-04-06:
I believe the "bug" is in the lowrisc uart driver, specifically this block:

https://github.com/tock/tock/blob/b02fe523c971abeb8d3f52e3c1747819ac39102b/chips/lowrisc/src/uart.rs#L211-L218

That client callback is in the `transmit_buffer` code path, but callbacks should only be issued in response to interrupts. In virtual_uart, the line `self.inflight.set(node);` _can_ expect to run before the callback occurs. That is why I wrote "bug", since this is a product of Tock's particular concurrency model.

We also have the `DeferredCall` mechanism, which `virtual_uart` uses to make sure that a callback only happens after an "interrupt"(simulated with a deferred call), even when an error occurs.

bcantrill, 2020-04-06:
> Btw, the frequency seem off on your platform.
>
>
> There are roughly 6 seconds between loop iterations, whereas the waiting time should be of 500ms. Or is that a side-effect of Tockilator's tracing, [...]

It's a side effect of running Ibex in Verilator, which (as you can imagine!) does indeed distort time. To run under Verialtor, you need to set `CHIP_FREQ` to `500_000` and `UART_BAUD` to `9_600`. @pfmooney has a patch to do this under a config option (which is what we use to compile), but a separate issue should be opened on this...

bcantrill, 2020-04-06:
> I believe the "bug" is in the lowrisc uart driver, specifically this block:
>
> https://github.com/tock/tock/blob/b02fe523c971abeb8d3f52e3c1747819ac39102b/chips/lowrisc/src/uart.rs#L211-L218
>
> That [...]

Well, considering that one of the most basic libtock-rs examples doesn't currently run on OpenTitan/Earl Grey, I would say that it's *just* a bug -- air quotes definitely not required!

I will leave to others whether the bug is in the UART driver or in the virtual UART, but if the bug is believed to be in the UART vs. the virtual UART, I would add that:

1. It seems like the use of `self.inflight` may still be incorrect as written as it will be set even if nothing is sent to the underlying device.

2. The contract between the virtual UART and the UART is clearly more intricate than documented -- and in particular, the implicitly depended upon context for `transmitted_buffer` should be spelled out much more clearly.  Here is that comment currently, which makes no mention of `transmitted_buffer` only be called asynchronously with respect to a call to `transmit_buffer`:

https://github.com/tock/tock/blob/b02fe523c971abeb8d3f52e3c1747819ac39102b/kernel/src/hil/uart.rs#L224-L242

3. This model seems incredibly brittle, as per @gendx's observation that there are (or have been) quite a few issues in this area -- and the system does little to prevent them.

alevy, 2020-04-06:
To summarize:

- Tockilator is still a very cool tool
- There is bug (no air-quotes) in the libtock-rs + OpenTitan/EarlyGrey stack
- The bug currently stems from the low-level UART driver "incorrectly" (air-quotes required) issuing a callback in the same call stack as the down-call to `transmit`.
- These kinds of bugs have shown up in the past, _particularly_ when developers who have yet to be burned by these kinds of bugs contribute otherwise wonderful code. It's currently: not well explained anywhere, not well tested, and (empirically at least) quite hard to catch during code-review.

Suggestions:
- [x] Continue using Tockilator
- [x] Acknowledge there is a bug
- [ ] The bug seems fixable: don't call `tx_progress` from the `transmit_buffer` and instead only begin the transmission---rely on interrupts, even in the single-byte case, to relay callbacks to the client.
- [ ]
  - [ ] Document this requirement better
  - [ ] Figure out how to test for these behaviors
  - [ ] Perhaps come up with abstractions that make it harder to make these "mistakes"

lschuermann, 2020-04-06:
> These kinds of bugs have shown up in the past, _particularly_ when developers who have yet to be burned by these kinds of bugs contribute otherwise wonderful code

Speaking from experience here, been tearing my head out on one of these bugs for weeks: Many developers run into these issues way before even trying to contribute. Requiring a callback from a different call stack (using deferred calls or interrupts) is something very common within Tock, which is as @alevy beautifully outlined not something obvious and even worse, not particularly good or __visibly__ documented. This is a core concept every Tock kernel developer should know about and requires careful and beginner friendly explanation somewhere near the beginning of a getting started guide. I don't know the best place to put it but given a pointer can certainly _try_ to draft a documentation.

commit 53ae5dbb (2020-05-11): Merge #1817

1817: Fix one byte UART transmissions for lowrisc (maybe) r=ppannuto a=bradjc

### Pull Request Overview

This pull request attempts to fix #1736. The issue is that the lowrisc UART TX driver can call a callback directly after a request is made, but for Tock's concurrency model that callback must be issued in response to an interrupt being handled so that the original caller can complete before the callback executes.

This change forces a TX interrupt to always happen, and only issues the callback in the handle_interrupt call if all bytes have been transmitted.

### Testing Strate [... cut]
PR #1817 (2020-05-11): Fix one byte UART transmissions for lowrisc (maybe)
### Pull Request Overview

This pull request attempts to fix #1736. The issue is that the lowrisc UART TX driver can call a callback directly after a request is made, but for Tock's concurrency model that callback must be issued in response to an interrupt being handled so that the original caller can complete before the callback executes.

This change forces a TX interrupt to always happen, and only issues the callback in the handle_interrupt call if all bytes have been transmitted.

### Testing Strategy

This is where I need help. I don't h [... cut]
closed 2020-05-11 by bors[bot]