tock/tock #1745: Kernel fault at allow syscall

alexandruradovici, 2020-04-10:
I am trying to run any app on nucleof429zi and stm32f3discovery. Since updating to the latest master, I get a kernel fault when using the allow syscall. I traced the error back to https://github.com/tock/tock/blob/cfa0757432e1b61e85f3ad49fb4baffd12c21344/capsules/src/console.rs#L201. 

Here is the fault output:

```
Initialization complete. Entering main loop
Loading processes from flash=0x08020000 into sram=[0x20004000:0x2000C000]
Loaded process[0] from flash=[0x08020000:0x08020400] into sram=[0x20004000:0x20006000] = Some("c_hello")
[0] function_call @0x8020055(0x802002c, 0x20004000, 0x2000, 0x20004c00)
[0] memop(0, 0x200049fc) = 0x0 = SUCCESS
[0] memop(10, 0x20004800) = 0x0 = SUCCESS
[0] memop(11, 0x200049fc) = 0x0 = SUCCESS

Kernel panic at arch/cortex-m4/src/lib.rs:297:
        "Kernel HardFault.
        Kernel version release-1.4-1018-gcfa07574
        r0  0x5f3efb0
        r1  0x8005fd1
        r2  0x6f490e80
        r3  0x8005fd1
        r12 0x0
        lr  0x800cb2f
        pc  0x800d5c6
        prs 0x21000000 [ N 0 Z 0 C 1 V 0 Q 0 GE 0000 ; ICI.IT 0 T true ; Exc 0-Thread Mode ]
        sp  0x20000a48
        top of stack     0x20001000
        bottom of stack  0x20000000
        SHCSR 0x0
        CFSR  0x8200
        HSFR  0x40000000
        Instruction Access Violation:       false
        Data Access Violation:              false
        Memory Management Unstacking Fault: false
        Memory Management Stacking Fault:   false
        Memory Management Lazy FP Fault:    false
        Instruction Bus Error:              false
        Precise Data Bus Error:             true
        Imprecise Data Bus Error:           false
        Bus Unstacking Fault:               false
        Bus Stacking Fault:                 false
        Bus Lazy FP Fault:                  false
        Undefined Instruction Usage Fault:  false
        Invalid State Usage Fault:          false
        Invalid PC Load Usage Fault:        false
        No Coprocessor Usage Fault:         false
        Unaligned Access Usage Fault:       false
        Divide By Zero:                     false
        Bus Fault on Vector Table Read:     false
        Forced Hard Fault:                  true
        Faulting Memory Address: (valid: false) 0x45F6EE38
        Bus Fault Address:       (valid: true) 0x45F6EE38
"
        Kernel version release-1.4-1018-gcfa07574

---| No debug queue found. You can set it with the DebugQueue component.

---| Fault Status |---
Precise Data Bus Error:             true
Forced Hard Fault:                  true
Bus Fault Address:                  0x45F6EE38
Fault Status Register (CFSR):       0x00008200
Hard Fault Status Register (HFSR):  0x40000000

---| App Status |---
App: c_hello   -   [Running]
 Events Queued: 0   Syscall Count: 4   Dropped Callback Count: 0
                                                                 Restart Count: 0
                                                                                  Last Syscall: Some(ALLOW { driver_number: 1, subdriver_number: 1, allow_address: 0x20004820, allow_size: 15 })

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20006000═╪══════════════════════════════════════════╝
             │ ▼ Grant         985 |    985
  0x20005C27 ┼───────────────────────────────────────────
             │ Unused
  0x200049FC ┼───────────────────────────────────────────
             │ ▲ Heap            0 |   4651               S
  0x200049FC ┼─────────────────────────────────────────── R
             │ Data            508 |    508               A
  0x20004800 ┼─────────────────────────────────────────── M
             │ ▼ Stack          64 |   2048
  0x200047C0 ┼───────────────────────────────────────────
             │ Unused
  0x20004000 ┴───────────────────────────────────────────
             .....
  0x08020400 ┬─────────────────────────────────────────── F
             │ App Flash       980                        L
  0x0802002C ┼─────────────────────────────────────────── A
             │ Protected        44                        S
  0x08020000 ┴─────────────────────────────────────────── H

  R0 : 0x00000001    R6 : 0x00000000
  R1 : 0x00000001    R7 : 0x20004000
  R2 : 0x20004820    R8 : 0x00000000
  R3 : 0x0000000F    R10: 0x00000000
  R4 : 0x0000000F    R11: 0x00000000
  R5 : 0x0802009D    R12: 0x00000000
  R9 : 0x20004800 (Static Base Register)
  SP : 0x200047C0 (Process Stack Pointer)
  LR : 0x080200D9
  PC : 0x080201E2
 YPC : 0x080201E2

 APSR: N 0 Z 0 C 1 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00

 Cortex-M MPU
  Region 0: [0x20004000:0x20006000], length: 8192 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20004000:0x20004400], Enabled
    Sub-region 1: [0x20004400:0x20004800], Enabled
    Sub-region 2: [0x20004800:0x20004C00], Enabled
    Sub-region 3: [0x20004C00:0x20005000], Disabled
    Sub-region 4: [0x20005000:0x20005400], Disabled
    Sub-region 5: [0x20005400:0x20005800], Disabled
    Sub-region 6: [0x20005800:0x20005C00], Disabled
    Sub-region 7: [0x20005C00:0x20006000], Disabled
  Region 1: [0x08020000:0x08020400], length: 1024 bytes; ReadOnly (0x6)
    Sub-region 0: [0x08020000:0x08020080], Enabled
    Sub-region 1: [0x08020080:0x08020100], Enabled
    Sub-region 2: [0x08020100:0x08020180], Enabled
    Sub-region 3: [0x08020180:0x08020200], Enabled
    Sub-region 4: [0x08020200:0x08020280], Enabled
    Sub-region 5: [0x08020280:0x08020300], Enabled
    Sub-region 6: [0x08020300:0x08020380], Enabled
    Sub-region 7: [0x08020380:0x08020400], Enabled
  Region 2: Unused
  Region 3: Unused
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug, run `make debug RAM_START=0x20004000 FLASH_INIT=0x8020055`
in the app's folder and open the .lst file.
```

Any feedback would be great.

ppannuto, 2020-04-10:
Given that's inside of the `Grant.enter(||` closure, the fault is almost certainly an error in https://github.com/tock/tock/pull/1648

I won't have time look into this until Monday at the earliest, but maybe you can bisect down?

bradjc, 2020-04-10:
I'm looking into this.

bradjc, 2020-04-10:
Try again on current master?

Sorry about this. Hopefully sometime soon we can have CI support that actually runs the Tock kernel.

alexandruradovici, 2020-04-10:
It doesn't fault anymore, but seems to be stuck after loading the processes.

```
Initialization complete. Entering main loop
Loading processes from flash=0x08020000 into sram=[0x20004000:0x2000C000]
```

alexandruradovici, 2020-04-10:
Seems to work though, ignore my last comment. Most probably the build tree did no clean up everything.

closed 2020-04-10 by bradjc