tock/tock #1758: [RFC] Do we want to support multi-language binaries? (e.g. mixed C and Rust code in userspace)

jrvanwhy, 2020-04-13:
Currently, you can build C apps with [libtock-c](https://www.github.com/tock/libtock-c) and Rust apps with [libtock-rs](https://www.github.com/tock/libtock-rs). However, there is no mechanism to build apps that mix C and Rust code together in a single binary. It is nontrivial to do so (I elaborate below), so the question is: do we want to do so?

At a minimum, to support multi-language apps, we would need an app whose `main()` is in language A to make FFI calls into language B. There are some other interactions we may wish to support as well:

1. **Can `main()` be in either language?** We could require apps that contain mixed C and Rust code use `libtock-c` as their runtime and have a C `main()`. We could require mixed apps to use `libtock-rs` as their runtime and have a Rust `main()`. We could allow either option (as is done for Rust on Linux).
1. **Is `malloc()` available to language B?** Should language B be able to call into language A's memory allocator?
1. **Can language B make `allow`/`command`/`subscribe`/`yield` syscalls?** Do we allow language B to interact with most of Tock's system call interfaces?
1. **Can we share a single driver between languages?** Can language A and language B both interact with a single system call API? E.g. can they both set timers independently?
1. **Can language B call `memop`? If language B can call other system calls, then should it be able to call `memop` as well? `memop` may not be useful if the runtime is provided by language A.

There are a number of technical challenges involved:

1. **Build system:** Currently `libtock-c` apps and `libtock-rs` apps have very different build systems. Do we teach Cargo to invoke `libtock-c`'s Makefiles? Do we teach `libtock-c`'s Makefiles to invoke Cargo? Do we teach both build systems to build shared libraries and let the application's Makefiles link them together?
1. **Relocation:** On ARM, `libtock-c` uses a dynamic relocation form that `rustc` doesn't support. What relocation models do we want to use? Currently it seems that static linking is the only option, but that may change in the future.
1. **Runtime component compatibility:** We can't have both entry points active, so language A will need to do the stack setup. Do we need to do anything to make that compatible? If we want language B to be able to access dynamic memory allocation, then we'll need to implement shims that support B's allocation mechanism using language A's heap, and those shims will need to be included at the correct times.
1. **Kernel-level system call virtualization:** If we allow both languages to access the same system call driver, how is that mediated? Does the kernel system call interface need to provide a virtualization mechanism to (allow the runtime libraries to) mediate between the languages? **May impact Tock 2.0**.
1. **Preemption:** There is an ongoing discussion about preemption (started by #1619). If we allow the kernel to preempt apps, do we need to do anything to make that work cross-language (i.e. language A preempted by language B and vice-versa)?

If we decided that we should pursue mixed-language apps, I'll open a tracking issue to cover these decisions and work items. Note that this is not a priority for me, but I am opening the question now because it may impact Tock 2.0 plans and may affect the design of libtock-rs (which is a priority for me).

alistair23, 2020-04-14:
I can't think of any real use case for a mixed language app, do you have any in mind?

ppannuto, 2020-04-14:
Quick instinct for where polyglot might be useful would be a mostly Rust app that wants to leverage an extant C library -- easy example would be a Rust app on Hail that wants to take advantage of the ble_serialization library.

alistair23, 2020-04-14:
Tock apps are so small that it almost just makes sense to write the app in C in that case.

ppannuto, 2020-04-14:
Maybe? I could see a reasonably complex sensing+processing app just wanting to use BLE advertisements to transmit data -- in that case the bulk of the logic has nothing to do with BLE.

However, this is also literally the example we use for demo-ing IPC, so maybe that's the answer / additional restriction: would anything need to be polyglot that couldn't be solved with IPC (i.e. where IPC would be too much overhead)?

Nothing as immediately comes to mind. Again my guess would be "some C library you want to use"; perhaps a ML model or something that's black box but operates on streaming sensor data as part of a larger application?

alistair23, 2020-04-14:
Ok, so the main reason to do this would be to support some C library with a Rust app where IPC won't cut it. Is that use case common/important enough to justify all of the above effort?

bradjc, 2020-04-14:
I can't come up with a use case for this.

If I go into academic mode, then the idea that there is a lot of C code out there that could be useful to non-C app is somewhat compelling, but the issues you are describing would require very complex interactions that would not come up in the "use an existing c library" case.

vsukhoml, 2020-04-15:
Use cases:
1) on target coverage, profiling, hwasan - this will be done using Tock/platform specific library linked to application as replacement for compiler-rt
2) link with closed-source libraries written in C (say, crypto for certification needs will be closed source).
3) link to device emulation library written in C for host-emulation of target platform.

Forget about malloc(), stdlib - it's embedded world.  need to invoke other makefiles, link to library, possibly generate bindings via cbindgen/rust-bingen.

alistair23, 2020-04-15:
1. I don't understand what you mean here.
 2. Ok, so you are saying that a use case is that someone wants to use their Rust app with some closed source C library (let's say crypto). Why couldn't the closed source C lib be moved to a different process and the two apps can communicate with IPC?
 3. What are you trying to emulate in C in a rust app? I think I am missing something here.

vsukhoml, 2020-04-16:
because that C lib will need direct access to hardware, so should be part of Tock kernel.

regarding emulation - i want to run target-specific code on host to get branch coverage, run fuzzing, etc. I can't do that on target/VM due to code/memory constraints. To stay as close as possible to original code I can change register interface in https://github.com/tock/tock/blob/master/libraries/tock-register-interface/src/registers.rs to call hooks which would emulate required peripheral devices, while rest of the code will be compiled for host. So, I can get same branch coverage, mostly same functional behavior, use debuggers, sanitizers, etc. That library which would provide device emulation will be most probably written in C, because it can be used with/based on QEMU too due to similarity of interfaces.

alistair23, 2020-04-16:
Ok, now it makes more sense :)

Neither of those use cases are running combined Rust/C in a Tock userspace app. So they don't really apply to what @jrvanwhy is asking.

Linking C libraries into Tock is a whole different problem and for the emulation it's probably better just to emulate the device in QEMU instead of using Tock kernel hooks.

jrvanwhy, 2020-04-16:
At this point, I think we have answers to the following questions (let me know if you disagree):

3. **Can language B make `allow`/`command`/`subscribe`/`yield` syscalls?** No, in all of the use cases we have considered, the language B code is a platform-independent library.
4. **Can we share a single driver between languages?** Not applicable -- language B cannot make any syscalls.
5. **Can language B call `memop`?** No, same reasoning as for the other system calls.

Answering these questions removes the need to consider **kernel-level system call virtualization** and **preemption**. I don't think this issue is relevant to the Tock 2.0 discussion anymore so I'll remove that label.

vsukhoml, 2020-04-20:
what prevents language B making a syscall? if language B is an assembler or C and it's not even planning to use libtock? After studying libtock-c a bit, I think that what is currently is https://github.com/tock/libtock-c/blob/master/libtock/crt0.c should become part of Tock core for task preparation, and make the rest as shim layer on top of syscalls via 'static inline' definition of basic syscalls, defining all types/constants. no 'libc' functionality. This would make it 'header-only' library.

As for Rust version - i'd also try to follow same idea, though it's probably not very idiomatic.

What I'm concerned though is duplication of code between apps. Somehow I'd try to make a common implementation of library as part of Tock. This is pretty easy with apps linked with Core, but need relocation fixup  one way or another in case of loadable apps. 

Specific use-cases I'm looking for are for security applications, with no loadable apps at all, so having a mechanism to support static linking and shared run-time would be an important feature. May be some of the functionality different between static & relocatable apps can be made configurable.

jrvanwhy, 2020-04-20:
> what prevents language B making a syscall? if language B is an assembler or C and it's not even planning to use libtock?

It's not that language B is actively prevented from making syscalls, but that they would not be deconflicted with language A's system calls. There may be a workable solution if language A and language B talk to distinct sets of drivers, but it's still unclear how `subscribe`/`yield` would interact between languages.

> After studying libtock-c a bit, I think that what is currently is https://github.com/tock/libtock-c/blob/master/libtock/crt0.c should become part of Tock core for task preparation, and make the rest [...]

I don't understand what you are asking for here. Would you mind explaining the connection with multi-language binaries?

> As for Rust version - i'd also try to follow same idea, though it's probably not very idiomatic.

I'm trying to keep `libtock-rs` higher-level than system calls.

> What I'm concerned though is duplication of code between apps. Somehow I'd try to make a common implementation of library as part of Tock. This is pretty easy with apps linked with Core, but need relocation fixup one way or another in case of loadable apps.

Tock does not currently support shared libraries, and it is very nontrivial to do so (and expensive in terms of code size). Please clarify what you mean by "apps linked with Core" -- applications built directly in the kernel?

ppannuto, 2020-04-21:
>  I think that what is currently is https://github.com/tock/libtock-c/blob/master/libtock/crt0.c should become part of Tock core for task preparation

If I understand correctly, here you are suggesting that the Tock kernel should set up bss and data on behalf of processes. That was actually the original design of Tock, however, when we introduced the Rust runtime, which had different initialization desires, we had to introduce an option to let processes specify their startup routines. For a little bit, we had it set up as an option specified in the TBF header, but in practice there was no use of the in-kernel loading, so we phased that out.

For an analog in Linux, the loader is a userspace process, not a kernel module, which sets up the new process memory segment and then `execve`'s itself into the started process. For Tock applications, the loader is effectively embedded into the process itself.

From a security perspective, I don't see an advantage to having the kernel act as a loader. If the loader is compiled into your application, then you have exact knowledge and control over what it is doing.

ppannuto, 2020-06-17:
I believe this RFC was resolved as of @jrvanwy's comment here: https://github.com/tock/tock/issues/1758#issuecomment-614924843

The follow-on nuance is that this RFC expresses what Tock positively aims to support, not necessarily what it aims to enforce. That is, Tock intends its syscall interface to be correct when exposed to a single runtime. The kernel does not prevent a polyglot app from making syscalls from multiple languages. A polyglot app is responsible for ensuring that any syscalls dispatched by a second language do not interfere with the primary language. Such coordination is outside of Tock's scope.

closed 2020-06-17 by ppannuto