tock/tock #1761: About the overhead and constraints of dynamic dispatch (`&dyn` references)

gendx, 2020-04-14:
Currently, the Tock kernel makes extensive use of "trait objects" (recognizable by `&dyn` references). One such example is the `Process` type, which implements the `ProcessType` trait. The rest of the kernel references processes via `&dyn ProcessType`.

The use of such a dynamic trait object is motivated by the fact that the actual `Process` type is parametrized by a `Chip`, so using a `&Process<ChipFoo>` everywhere instead of `&ProcessType` would make the code more verbose because the `Chip` parameter would have to be recursively given by all structs/traits referring to a process.

However, I see multiple problems with this approach.
- **Dynamic trait objects** have in principle some **overhead**. This is because in principle each object refers to a vtable to resolve at runtime all of the functions in the trait. The reason being that there can be multiple implementations of the trait, and `&dyn ProcessType` could be any struct that implements the trait.
  However, in practice, for a given board there is a single chip and therefore a single implementation of a `Process`. So all the methods in a process could instead be resolved at compile time, and even optimized (inlined by the compiler, etc.).
- Because of the constraints of storing all methods in a vtable, traits used by trait objects have some programming limitations. For example, the **methods in such traits cannot themselves be parametrized** by other types. This problem came up in https://github.com/tock/tock/pull/1648#discussion_r405022427. Essentially, we cannot have the following `alloc` function:
  ```rust
  pub trait ProcessType {
      fn alloc<T: Sized>(&self) -> Option<NonNull<T>>;
  }
  ```
  and then use `&dyn ProcessType`. Essentially, this is because the compiler cannot give a dynamic vtable pointer for such a generic `alloc` function - as it would need to have a function pointer for every possible `T` one might instantiate `alloc` with. This limitation is shown by the following compiler error.
  ```
  error[E0038]: the trait `process::ProcessType` cannot be made into an object
     --> tock/kernel/src/sched.rs:327:18
      |
  327 |         process: &dyn process::ProcessType,
      |                  ^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `process::ProcessType` cannot be made into an object
      |
     ::: tock/kernel/src/process.rs:183:11
      |
  183 | pub trait ProcessType {
      |           ----------- this trait cannot be made into an object...
  ...
  350 |     fn alloc<T: Sized>(&self) -> Option<NonNull<T>>;
      |        ----- ...because method `alloc` has generic type parameters
      |
      = help: consider moving `alloc` to another trait
  ```

In essence, broad use of such dynamic objects allows to remove verbosity, at the expense of performance and expressiveness. As such, it is quite similar to using `'static` lifetimes everywhere instead of carefully propagating lifetimes through HILs (#1074, #1628). The limitations of `'static` lifetimes have been shown in #1069 and #1145 for example.

As an example, removing the `dyn` part of `ProcessType` (and removing this trait altogether) would mean that the kernel would take an additional chip parameter, to be able to write the following.

```rust
pub struct Kernel<C: Chip> {
    processes: &'static [Option<&'static process::Process<C>>],
}
```

This would mean that the `Chip` parameter would have to be propagated to all structures using the kernel, so the changes would be similar to the generic kernel lifetime proposal (#1628). And indeed, to continue the analogy, `'static` is to a generic lifetime `'ker` what `&dyn ProcessType` (dynamic dispatch) is to `&Process<C>` (static dispatch).

What do you think?

ppannuto, 2020-04-14:
I think my instinct is supportive of this. Conceptually, it makes sense that the kernel is instantiated for a particular type of chip. I imagine this fits in nicely with the growing suite of widespread changes that I would imaging being part of a 2.0 overhaul (removing the `'static`s from HILs; the `'ker` lifetime; some of the more substantial feature removal ideas [#1650, #1646]; is the list that comes to mind).

It's unfortunate that many of these changes increase the verbosity of code. I'm less decided on whether that's a net win or loss of readability (verbose => explicit, which can be good).

At some point (after the 1.5 release), I think we'll need to make a plan for how these series of "touch everything" changes will go through.

hudson-ayers, 2020-04-14:
Thank you for writing this up. I have been running into a number of similar issues while working on a generic scheduler interface, and had been starting to wonder whether a change like this might be worth it after all. I have a few comments:

1. For the `Chip` example in particular, it is a little frustrating that we have to add a generic type parameter given that there will only ever actually be one chip for any compiled binary. It would be nice if there were some way to have Chip just be a struct, and multiple `Chip` structs are defined in different crates (`sam4l`, etc.), but which Chip struct is actually pulled into `sched.rs` is determined by where `make` is called from -- I can't immediately think of any good way to achieve this though

2. I believe that there are actually two reasons for having a `ProcessType` trait. One is the reason you mentioned -- that `Process` is parameterized by a `Chip`. The second is that there was desire to keep the door open to adding new `Process-` types, with the possibility of in-kernel processes being one particular example that might motivate this. Of course, this is till possible without dynamic dispatch -- the `Kernel` struct could also be parameterized by a generic `ProcessType` (though this adds even more boilerplate).

3. I think if we do this, I would want to move my WIP scheduler trait to use this approach as well, so at that point we could be looking at a Kernel struct like:
    ```
    pub struct Kernel<'ker, C: Chip, P: ProcessType, S: Scheduler,> {
        processes: &'ker [Option<&'ker P<C>>],
        scheduler: &'ker S,
        //...
    }
    ```
    which of course would also then be the signature of `AppId`, `Grant`, `Callback`, all capsules, all 
    components.
    
    I definitely think that is a significant net loss in readability -- it will make little sense to new Tock 
    users why every single capsule is parameterized by all of these seemingly unrelated things.
    
    However I think it might be worth it for the performance benefits (code size, RAM, CPU cycles), so it would be nice to quantify those to some extent.

4. I think our use of `dyn` references to pass around Capabilities is another example where we are paying an artificial cost. I think if we were to replace all `&dyn ProcessManagementCapability` arguments with capabilities passed by value, I think the compiler would be able to entirely optimize those arguments away in a manner that I suspect it is unable to right now.

gendx, 2020-04-14:
To take back your example:

>    ```rust
>    pub struct Kernel<'ker, C: Chip, P: ProcessType, S: Scheduler> {
>        processes: &'ker [Option<&'ker P<C>>],
>        scheduler: &'ker S,
>        //...
>    }
>    ```

I'm wondering to which extent we could leverage associated types to make things more readable. I haven't tried the following code, but there could be something like:

```rust
trait KernelTypes {
    type C: Chip;
    type P: ProcessType;
    type S: Scheduler;
}
```

and then we define the kernel with only one parameter:

```rust
pub struct Kernel<'ker, T: KernelTypes> {
    processes: &'ker [Option<&'ker T::P>],
    scheduler: &'ker T::S,
    //...
}
 ```

hudson-ayers, 2020-04-14:
> ```rust
> trait KernelTypes {
>     type C: Chip;
>     type P: ProcessType;
>     type S: Scheduler;
> }
> ```

another big advantage of that approach is that adding additional types (after the initial change) only requires changing a couple of lines of code instead of a couple thousand

bradjc, 2020-04-14:
Would a change like this propagate outside of the kernel crate? When I added `ProcessType`, my initial attempts required changing something, which lead to changing Grant, which lead to changing capsules. My concern is that invasive syntactic overhead would be much more than just some verbosity, and would make Tock very difficult to read, and therefore difficult to maintain/develop/use.

If it can be captured in the kernel crate then it sounds like an improvement.

hudson-ayers, 2020-04-14:
As proposed, this would propagate well outside of the kernel crate.

I have been thinking about this some for the scheduler example, and I came up with one way to prevent this from propagating outside of the kernel crate is to use default type parameters + conditional compilation + cargo features:

in `kernel/Cargo.toml`, list the scheduler options
```
[features]
round_robin = []
priority = []
```

then in `boards/imix/Cargo.toml`, add the scheduler you want:
```
[dependencies]
kernel = { features = ["round_robin"], path = "../../kernel" }
```
and finally, sched.rs:
```rust
//! sched.rs

#[cfg(feature = "round_robin")]
type DefaultChip = RoundRobin;
#[cfg(test)]
type DefaultSched = Priority;

pub struct Kernel<S: Scheduler = DefaultSched> {
    scheduler: S,
}
```

This is still a little clunky, but *way* less clunky than parameterizing 1000 lines of code throughout Tock with the scheduler type. Unfortunately, for the process/chip example it is a little trickier, because the actual `Chip` structs are created outside of the Kernel crate, in different crates that depend on the Kernel crate (rather than the other way around), but it may be possible to reorganize such that this is solvable..

gendx, 2020-04-15:
Another example that I just remembered with dynamic dispatch is the `ProcessRestartPolicy`.

Again, even though each board is instantiated with (at most) only one restart policy, this policy is stored in a dynamic reference, which may prevent compiler optimizations.

https://github.com/tock/tock/blob/1088a52b977ac1ef8da5d03788ae08e1b3bc944d/kernel/src/process.rs#L572

For example, the "always restart" policy has a trivial predicate which returns `true`.

https://github.com/tock/tock/blob/1088a52b977ac1ef8da5d03788ae08e1b3bc944d/kernel/src/process.rs#L481-L485

This should be propagated to eliminate the following branch of code if this predicate is well inlined by the compiler. If dynamic dispatch prevents that, this incurs some overhead due to missed optimization opportunities (on top of the vtables).

https://github.com/tock/tock/blob/1088a52b977ac1ef8da5d03788ae08e1b3bc944d/kernel/src/process.rs#L854-L856

hudson-ayers, 2020-04-15:
I think that changing `ProcessRestartPolicy` to not use dynamic dispatch could be done without changing code outside of the kernel crate, because it is already hidden behind a `ProcessType` trait object

gendx, 2020-04-15:
> Would a change like this propagate outside of the kernel crate? When I added `ProcessType`, my initial attempts required changing something, which lead to changing Grant, which lead to changing [...]

Another question to ask is why so many objects (grants, app ids, etc.) are storing a reference to the whole kernel. Can't they be re-designed in such a way that the kernel reference is instead passed to them in function calls that need it?

This could simplify a lot of things - for a start the kernel lifetime (#1628) and this.

This could also reduce the memory overhead - instead of having N grants/appids/foos/bars objects all storing a reference to the same kernel, there would be a single reference passed around in function calls that need it.

bradjc, 2020-04-15:
The references to the kernel are to provide access to the list of processes that all of those types need in some way. AppId, in particular, is shared with capsules, and capsules can't have a reference to the kernel (and shouldn't have to have a reference to the kernel).

ppannuto, 2020-04-15:
>  and capsules can't have a reference to the kernel (and shouldn't have to have a reference to the kernel).

Is that actually still true? None of the `Kernel` struct members are `pub` and all of the `pub` methods now require a capability of some kind. Maybe capsules can/should have a kernel reference.

bradjc, 2020-04-15:
> Is that actually still true? None of the `Kernel` struct members are `pub` and all of the `pub` methods now require a capability of some kind. Maybe capsules can/should have a kernel reference.

It might be ok for capsules to have the reference, but it seems like an extra developer burden for them to need to keep track of it and pass it around, while never actually using it.

gendx, 2020-04-16:
This may start to side-track a bit from the original issue.

But regarding the references to the kernel, a typical capsule look like the following.

```rust
struct FooDriver {
    apps: Grant<FooApp>,
    ...
}

struct FooApp {
    cb1: Option<Callback>,
    cb2: Option<Callback>,
    slice1: Option<AppSlice<_, _>>,
    slice2: Option<AppSlice<_, _>>,
}
```

The layout of such a capsule looks like the following in memory.

```
+ FooDriver (allocated with static_init! macro)
  + Grant<FooApp>
    - &'static Kernel     <=====
    - usize
  - ...

+ FooApp (allocated via Process::alloc, in each process' grant memory section)
  + Callback
    + AppId     <=====
      - &'static Kernel     <=====
      - usize
      - usize
    - CallbackId
    - usize
    - NonNull<*mut ()>
  + Callback
    - AppId     <=====
    - ...
  + AppSlice
    - AppId     <=====
    - NonNull<T>
    - usize
  + AppSlice
    - AppId     <=====
    - ...
```

From this memory layout, we can see that there are many redundant references.
- Each capsule contains a reference to the kernel, via the `Grant`.
- For each capsule times each process times each Callback/AppSlice, there is an `AppId`, itself containing a reference to the kernel and information about the app ID. Yet the `AppId` should be the same for all the grants within a process, right?

Although potentially decreasing the memory overhead of grants would be a different question, I think it's something worth considering if we end up re-designing `AppId` and related structs.

gendx, 2020-04-16:
> The references to the kernel are to provide access to the list of processes that all of those types need in some way. AppId, in particular, is shared with capsules, and capsules can't have a reference to the kernel (and shouldn't have to have a reference to the kernel).

I've also though of another way of tackling this problem. Instead of passing a kernel reference around, there could be a `static KERNEL: Kernel` object (given that there is only a single kernel anyway). Then the kernel would expose static methods, that themselves retrieve the static instance of the kernel and do some operations on it.

The major drawback I see with this approach is that the kernel cannot be generic in this way - there's no such thing as `static KERNEL: Kernel<T>` because that would mean a static object for every possible `T`.

---

Apart from that, there is an inherent reason why some of these kernel objects need to have a reference to the kernel (rather than a reference passed to them from function calls): the `Drop` implementation.

https://github.com/tock/tock/blob/010ffa33e4bbd5d5277c178fe10e2d2f91514b2e/kernel/src/grant.rs#L59-L68

To implement such a `drop`, either `Owned<T>` has a reference to the kernel (current implementation), or there is a way to retrieve the static instance of the kernel (see the beginning of this comment).

However, the only place where an `Owned<T>` is created (and then dropped) is inside `AppliedGrant::enter`, so if I'm not mistaken the drop could be some explicit call there instead:

https://github.com/tock/tock/blob/010ffa33e4bbd5d5277c178fe10e2d2f91514b2e/kernel/src/grant.rs#L25-L35

The more complicated case, is `AppSlice`, internally using `AppPtr`, which implements `Drop`.

https://github.com/tock/tock/blob/010ffa33e4bbd5d5277c178fe10e2d2f91514b2e/kernel/src/mem.rs#L51-L59

Capsules can obtain an `AppSlice` via `allow`, and then drop it whenever they want, at which point the above `drop` must run and use the kernel.

https://github.com/tock/tock/blob/010ffa33e4bbd5d5277c178fe10e2d2f91514b2e/kernel/src/process.rs#L1097-L1101

By the way, what prevents a "malicious" capsule from "collecting" an `AppSlice` when an app calls the `allow` syscall, and then keeping the `AppSlice` forever and continue using it, even after the app has disallowed it? How does this fall into the threat model? Shouldn't the capsules instead ask the kernel every time they need to access an application slice - so that disallowed slices are not accessible anymore?

hudson-ayers, 2020-08-18:
One note: getting rid of trait objects is not possible for all instances where trait objects are used to allow for the creation of linked lists of different underlying types (all virtualizers).