tock/tock #1764: Lack of Error Signaling when insufficient app memory available

hudson-ayers, 2020-04-16:
By default, libtock-c applications request 8kB of memory. On a board like Hail, `NUM_PROCS=20`, implying that 20 processes can be loaded. However `APP_MEMORY` on hail is set to 49 kB, so attempting to load more than 6 processes fails. Currently, this manifests as the first 6 processes loaded onto the board work as expected, and subsequently flashed apps are treated as if they do not exist. No error messages are printed and the only signal something is wrong is that some apps do not execute.

A developer encountering this problem can solve it by increasing `APP_MEMORY`, or by decreasing the amount of memory requested by apps in libtock-c -- but it can be confusing to debug.

Ideally, when a board is initialized, it should at least print a debug message if there is not enough memory to load all apps, or perhaps even panic. Alternatively, perhaps tockloader could print an error if an app is loaded that there is not enough memory for?

alistair23, 2020-04-16:
If you enable the process loading debug does that give you enough information to figure out what is going on? Maybe that should be enabled by default?

hudson-ayers, 2020-04-16:
> If you enable the process loading debug does that give you enough information to figure out what is going on? 

Ah, yes it does! it says '[!] flash=[0x00041000:0x00041400] process=Some("t5") - couldn't allocate memory region of size >= 0xFC8'

>Maybe that should be enabled by default?

I don't think we want to print all the process loading information in the default case -- I think the default should be to only print anything if there is an error

hudson-ayers, 2020-04-16:
Which would be a pretty easy fix (take the error reporting debug statements outside of the `if config::CONFIG.debug_load_processes`  block) but I wonder if others will object to that? Perhaps the error should also be updated to suggest reducing the amount of memory used by the app? Perhaps error-reporting debug statements that happen only at initialization should be moved behind a different config flag that is enabled by default?

bradjc, 2020-04-16:
The line that is relevant here:

https://github.com/tock/tock/blob/5b61be4a41e39cbe1e44f6f7539f6201680f391a/kernel/src/process.rs#L1659

Pre-TBF parsing changes addressing this issue would have been tricky, but now we can propagate an error to main.rs which will can then handle it as desired (right now we would print a message).

I think I didn't want to change too much in the TBF refactor, but looking at what is and is not an error when discovering and creating apps is something worth looking into. For example, should load_processes keep trying apps after not being able to load one? I think propagating an error in that case would be more difficult.

----

> Which would be a pretty easy fix (take the error reporting debug statements outside of the `if config::CONFIG.debug_load_processes` block) but I wonder if others will object to that?

I would, but I think we already support a better solution using `Result`.

hudson-ayers, 2020-04-16:
Ah, I was working off a slightly outdated version of master that pre-dated the TBF parsing changes. Propagating that error to main.rs seems like the right solution.

PR #1770 (2020-04-30): propagate additional process loading errors to main.rs
### Pull Request Overview

This pull request modifies `process::create()` to return an error in 3 cases where previously it just returned an empty process slot -- if there is an error initializing the process, if there is an error allocating the MPU region for that process's flash, or if there is not enough memory to meet the amount requested by that process (or the minimum amount currently required by the kernel).

It adds one new `ProcessLoadError` variant: `ProcessLoadError::NotEnoughMemory`. Failures with allocating an MPU region or in [... cut]
closed 2020-04-30 by bradjc