tock/tock #1769: Remove guarantees on memory size and stack pointer value at app startup (Tock 2.0)

jrvanwhy, 2020-04-17:
The Tock kernel currently implements the following:

1. Apps receive at least [`min_ram_size`](https://github.com/tock/tock/blob/master/doc/TockBinaryFormat.md#1-main) at startup.
1. Apps receive at least 3 KiB of RAM at startup.
1. The stack pointer is set 3 KiB after the start of the app's memory region at startup.

We should allow apps to use less than 3 KiB of RAM. However, doing so would break apps that rely on the above behaviors, and therefore needs to wait until Tock 2.0.

jrvanwhy, 2020-04-17:
I propose we replace the replace the existing behavior (at outlined in the issue behavior) with the following behavior:

1. Apps receive the minimum RAM they can receive that is greater than or equal to their `min_ram_size` value (subject to MPU constraints).
1. Apps are started with the highest valid stack pointer value (i.e. just before the initial app break, subject to alignment constraints).

Keeping the RAM allocation size close to `min_ram_size` bytes will help diagnose apps that give a too-small value for `min_ram_size`. Setting the stack pointer to the top of the app's RAM region gives room for interrupts to push frames onto the stack (I know this is required for ARM, I don't know about RISC-V). We would expect the userspace runtimes to perform their own (better) stack setup.

ppannuto, 2020-04-17:
I might suggest that apps start with an undefined stack pointer.

I don't think either the Rust or C runtime currently have any expectations or requirements on the initial stack pointer, I will speak authoritatively that [the C runtime starts with assembly that sets `sp` before doing anything else](https://github.com/tock/libtock-c/blob/dc87cc2d76cd7e528cc85d52d2dec990995ae4b1/libtock/crt0.c#L58).

alistair23, 2020-04-19:
The Rust assembly is basically the same as the C assembly (at least for RV32) and I'm pretty sure we also have no expectations on the stack pointer being set up.

bradjc, 2020-04-20:
> I might suggest that apps start with an undefined stack pointer.
> 
> I don't think either the Rust or C runtime currently have any expectations or requirements on the initial stack pointer, I will speak authoritatively that [the C runtime starts with assembly that sets `sp` before doing anything else](https://github.com/tock/libtock-c/blob/dc87cc2d76cd7e528cc85d52d2dec990995ae4b1/libtock/crt0.c#L58).
> 
> Given that nothing relies on the stack pointer value presently, I'd err on the side of maximizing future flexibility by leaving it explicitly undefined.

But the Cortex-M Userland-Kernel boundary implementation requires that the app have a stack pointer.

ppannuto, 2020-04-21:
> But the Cortex-M Userland-Kernel boundary implementation requires that the app have a stack pointer.

Yes, eventually the stack pointer must be well-defined. Specifically, there has to be a valid stack by the time that the userland calls into the kernel.

As an implementation artifact, the kernel happens to set up a stack, so that the switch_to routine can assume there was already one present. However, the first thing that C runtime does is completely ignore all the data on that "stack" and set up a new one anyway.

I think my statement is still true: neither the C runtime nor the Rust runtime make assumptions about the value of the `sp` register at startup.

vsukhoml, 2020-04-21:
> Yes, eventually the stack pointer must be well-defined.

Cant' agree more, and I'd move this duty to OS. Being under OS control enables OS to check that it wasn't pivoted to other place on one side, and reduce size of run-time, which otherwise have to do setup. 

> kernel happens to set up a stack

If this artifact exist, it can be repurposed and reduce run-time code size. What are the benefits for app to be responsible for it's own stack? Are there any use case where it may be needed and can't be addressed in other way? 

> neither the C runtime nor the Rust runtime make assumptions

That is true for run-time, but application itself may have assumptions about available stack. Different apps have different needs. This requirement may come from Tbf or be known to OS in other way (say, if apps and OS are built together statically).

jrvanwhy, 2020-04-21:
> > kernel happens to set up a stack
> 
> If this artifact exist, it can be repurposed and reduce run-time code size. What are the benefits for app to be responsible for it's own stack? Are there any use case where it may be needed and can't be addressed in other way?

Ostensibly, it really shouldn't be the kernel's responsibility to do application runtime setup. For example, the kernel shouldn't need to know what language an app is written in, but different languages have different runtime setups (e.g. on ARM, libtock-c performs dynamic relocations while libtock-rs doesn't).

> That is true for run-time, but application itself may have assumptions about available stack. Different apps have different needs. This requirement may come from Tbf or be known to OS in other way [...]

This is a good reason for stack setup to be done from userspace.

bradjc, 2020-04-21:
But for the kernel to start the process the first time the UKB implementation for cortex-m requires a stack pointer to push a stack frame to which is how the HW knows where in the app to start executing. I'm just noting that either the kernel has to pick a stack pointer or the cortex-m UKB implementation has to change.

vsukhoml, 2020-04-22:
> Ostensibly, it really shouldn't be the kernel's responsibility to do application runtime setup.

There are few common sections - .text, .data, .rodata, .bss. Loader of application should move data from .data into RAM (copy from LMA to VMA) , and initialize .bss.  All languages so far follow this convention. Stack size is configured for application by developer. Loader can configure it.
This is common stuff performed for every application. Having it as part of application is just a waste of memory and doesn't bring any benefits. 

I understand desire to have rich functionality, ease of use for loadable applications, and thus a need to support dynamic relocations on some platforms. On Linux dynamic loader is shared among all apps. I'd prefer same model for Tock. Implementing shared libs on Tock in my opinion has some use cases [like CardOS for Smart cards], but it also creates new corner cases (what to do if symbol is not found, ABI changed, etc). So far it's not supported. 
The other approach to make loader shared is to have loader as part of OS and make sure all parameters are in Tbf, same way as they are in ELF, including entry point, stack size, .data section, etc. The benefit is that common code is not duplicated. 

I'm not a big fan of dynamic relocations on embedded target due to extra memory requirements (relocation tables, less efficient addressing modes, run-time support). Specific use cases I'm looking at are around several static apps, updated with OS core and provided as single image. As such, I'm looking for board-specific implementation which will support this model, but it require changes in core, which I wouldn't like to fork. Another optimization I'm looking for is that given a static list of applications and requirements, some of the internal tables can be computed at compile time.  If you know that you only have X apps with specified stack/heap/data requirements, many parts can be simplified [hopefully with just LTO]. 

> This is a good reason for stack setup to be done from userspace.

What makes you think so? I'd do some extra hardening on syscalls and check that stack pointer is always where it should be, return address is in .text segment [so we don't execute stack/data]. This can't be done if OS is not in control.
Stack pointer would be set to lower address in RAM so stack overflow would generate access violation, while .data would be close to top of range so buffer overflow would also cause access violation. This is a common part. All the rest can be language run-time specific. The less control application have over this, the better it is for security.

ppannuto, 2020-04-22:
> But for the kernel to start the process the first time the UKB implementation for cortex-m requires a stack pointer to push a stack frame to which is how the HW knows where in the app to start executing. **I'm just noting that either the kernel has to pick a stack pointer or the cortex-m UKB implementation has to change.** [emph mine]

Yes, I think we're agreeing with each other. The kernel has to pick something, and that's fine. I'm just saying that what the kernel chooses to pick should not be part of the ABI contract with userspace, i.e. from the userspace perspective Tock makes no guarentees about what the kernel chooses and the kernel is free to choose something different in the future without notice.

ppannuto, 2020-04-22:
> Having it as part of application is just a waste of memory and doesn't bring any benefits.

For libtock-c, the entire `_start` routine is 60 bytes on ARM. I didn't do the math for risc-v as I don't know their instruction coding as well, but it's similar. The memory waste is minimal, and it does afford a benefit of application-level flexibility.

---

> On Linux dynamic loader is shared among all apps. I'd prefer same model for Tock.

I think a userspace loader process would be interesting to explore; more generally, we've talked a bit about a "supervisor" process (similar to linux's init process) before that could incorporate the functionality. I think we'd be open to exploring that direction.

---

> I'm not a big fan of dynamic relocations on embedded target due to extra memory requirements 

How else do you support general purpose applications on arbitrary platform targets? Re-usable applications (i.e. *not* compiled for a specific embedded board) was and is a first-order design goal of Tock. Platforms can certainly opt-out on a case-by-case basis, but reusable apps are central to Tock.

---

> The less control application have over this, the better it is for security.

I don't understand this line of argument. Security of what? The kernel? It already considers processes wholly untrusted, letting them set up their memory space arbitrarily doesn't change its threat model in any way. Of the process? The current design gives processes the most control.

---

> Certification ... loading order for security

Tock recently has added a formal thread model: https://github.com/tock/tock/tree/master/doc/threat_model -- I think if we want to make higher-level guarantees like this, we should discuss them as potential updates to the threat model.

bradjc, 2020-04-22:
> Yes, I think we're agreeing with each other. The kernel has to pick something, and that's fine. I'm just saying that what the kernel chooses to pick should not be part of the ABI contract with [...]

We should probably add test apps that mess with the stack pointer and make sure the kernel handles it OK. It certainly would be easier to have the kernel have nothing to do with an app's stack pointer, but that might not be possible/worth it.

jrvanwhy, 2020-04-22:
> We should probably add test apps that mess with the stack pointer and make sure the kernel handles it OK. It certainly would be easier to have the kernel have nothing to do with an app's stack [...]

Has #1257 been fixed? #1257 was triggered by having an app set its stack pointer to an invalid value.

jrvanwhy, 2020-04-22:
> What makes you think so? I'd do some extra hardening on syscalls and check that stack pointer is always where it should be, return address is in .text segment [so we don't execute stack/data]. This [...]

Except for guard pages, I am not aware of stack pointer checks being a successful security mechanism. Regardless, the kernel already knows enough to enforce that even though it doesn't set up the stack itself.

Write-xor-execute is already implemented via the MPU, which has the advantage of preventing exploits that don't trigger system calls.

> Stack pointer would be set to lower address in RAM so stack overflow would generate access violation, while .data would be close to top of range so buffer overflow would also cause access violation. [...]

For `libtock-rs` (and I *believe* `libtock-c`), the stack is already at the bottom of the application's address space for exactly that reason. For `libtock-c` the heap is at the top so that it grows towards the grant region -- this minimizes the size of the MPU region the app has access to. I intend to do the same in `libtock-rs` but that hasn't happened yet.

`libtock-c` and `libtock-rs` should provide secure setups by default (and to implement Rust's memory safety guarantees, `libtock-rs` must make the stack grow towards an MPU boundary). There is no security benefit to performing runtime setup in the kernel rather than in the apps.

> PP099 defines Installer: The installer is the Java Card System component dealing with loading, linking and installation of new packages. The installer is the on-card application responsible for the
> installation of applets on the card. It may perform (or delegate) mandatory security checks according to the card issuer policy (for bytecode-verification, for instance), loads and link packages (CAP
> file(s)) on the card to a suitable form for the Java Card VM to execute the code they contain. It is a subsystem of what is usually called “card manager”; as such, it can be seen as the portion of
> the card manager that belongs to the TOE.

Tock uses the term "[Application Loader](https://github.com/tock/tock/blob/master/doc/threat_model/Application_Loader.md)" to refer to the same concept. Note that in Tock, the application loader is not necessarily an application.

> Beyond functional issues, the installer ensures at least a property that matters for security: the
> loading order shall guarantee that each newly loaded package references only packages that
> have been already loaded on the card. The linker can ensure this property because the Java
> Card platform does not support dynamic downloading of classes.

The PDFs aren't loaded for me. What is the purpose of the loading order check? It's unclear how that particular security property would translate over to Tock.

bradjc, 2020-04-23:
This issue is making me think that the fact that process.rs tracks a stack pointer is for two reasons:
1. To help with debugging by letting users know how big the stack has grown.
2. Because it is used for the cortex-m syscall implementation, which was more tightly coupled with process.rs in the past.

Perhaps it makes sense to make the stack pointer entirely contained in UKB. That way process.rs doesn't have to be responsible for it at all. However, UKB would still need to check the stack pointer if it is going to be using it, so we would have to pass in the app's memory bounds to UKB interface (which we are already kind of doing).

vsukhoml, 2020-04-28:
i have very specific use case now why having kernel to set up stack is better - for ChromeOS we want to disable loadable apps mechanism completely as it creates potential security issues as we don't yet have protection against malicious apps, no authentication of apps, etc. So, we want to have apps as 'libs' added as a dependency to board project and linked together. and then 'loaded' using Process::create directly instead of kernel::procs::load_processes as last step of board initialization. It would be nice to have design a bit better suited for this, like variant of load_processes which gets a struct and skip parsing Tbf headers.

rajivr, 2020-04-28:
> i have very specific use case now why having kernel to set up stack is better - for ChromeOS we want to disable loadable apps mechanism completely as it creates potential security issues as we don't yet have protection against malicious apps, no authentication of apps, etc.

@vsukhoml We too have a very similar requirement where we don't need loadable apps mechanism. In our case authenticity and integrity is ensured at the image (kernel + apps) level and not at the app level.

The approach that I am currently experimenting with is with custom Tbf headers. In our custom Tbf headers I am also specifying the MPU regions for both SRAM and Flash. (This approach coupled with no support for `brk`, `sbrk` system calls greatly simplifies the MPU trait and MPU code). External tooling (that is not yet written) can read the app `.elf` files and make sure that MPU constrains of the underlying architecture is correctly met. We would run this tool as a part of the build system.

vsukhoml, 2020-04-29:
@rajivr This sounds interesting! My idea is to refactor kernel::procs::load_processes and Process::Create so that Tbf header parsing will be a separate step, producing list of processes and process configuration, while there will be an API to start process given already parsed information. So, both use cases would be supported. 

If apps are compiled into relocatable objects (ldd -r) and linked with OS, then it's easy to define symbols in linker script marking start/end of each process and it's data, and build process table statically. Today on RISC-V one have to place applications manually as PIC is not properly supported. Earlier I used autogenerated linker scripts and still exploring this option. But even when PIC would be supported, it's use creates an overhead due to addressing modes, and can be a problem for constrained environment.

rajivr, 2020-04-29:
> If apps are compiled into relocatable objects (ldd -r) and linked with OS, then it's easy to define symbols in linker script marking start/end of each process and it's data, and build process table [...]

@vsukhoml I would really like to learn how you combined the kernel and app (`.o` files) into one linker script. Would you be able to share an example of that? That could greatly simplify things and make the library model very workable.

Currently in Tock, the approach that we take to load apps is to either use Tockloader or use [`objcopy`](https://github.com/tock/tock/tree/master/boards/nucleo_f446re#flashing-app). I was planning on relying on `objcopy` approach with some additional checks.

vsukhoml, 2020-04-29:
@rajivr I'm still looking on best way to do it - either rename sections in app .o file using objcopy or try to control section names on compiler side. key thing - make sure you can separate app's code and data from Tock's code and data, and from code/data of other apps to configure memory protection. The only reliable way i see so far - different section names.

hudson-ayers, 2021-02-12:
@jrvanwhy can this be closed now that #2383 is merged? Or is there still a memory size guarantee that needs to be addressed?

jrvanwhy, 2021-02-12:
> @jrvanwhy can this be closed now that #2383 is merged? Or is there still a memory size guarantee that needs to be addressed?

#2383 is *almost* all the way there. It has the Tock 2.0 semantics commented out (see [`initial_process_app_brk_size`](https://github.com/tock/tock/pull/2383/files#diff-3e14d5caed71206a0a09f1bdbefd86f3715ce9bf5838affe03990eeeee21bd61)).

I think the remaining work is to merge `master` into `tock-2.0-dev` then uncomment the Tock 2.0 semantics in `arch/{cortex-m,rv32i}/src/syscall.rs`. I'm not comfortable doing the merge myself, but I'll be happy to send the uncomment PR when I see a merge has occurred.

phil-levis, 2021-02-17:
> #2383 is _almost_ all the way there. It has the Tock 2.0 semantics commented out (see [...]

I'll take a look tomorrow (2/18).

hudson-ayers, 2021-02-17:
FYI: The relevant code has been merged into `tock-2.0-dev` already as part of #2407 , but it has not yet been switched over to the 2.0 approach

jrvanwhy, 2021-02-17:
> FYI: The relevant code has been merged into `tock-2.0-dev` already as part of #2407 , but it has not yet been switched over to the 2.0 approach

I'll send a PR later today to switch to the 2.0 approach. I have RISC-V and ARM entry points for `libtock-rs` that depend on the new behavior, and can manually test that they work with the new kernel.

hudson-ayers, 2021-02-18:
Closed by #2441

closed 2021-02-18 by hudson-ayers