tock/tock #1829: [QEMU] hifive1 QEMU is quickly blocked after boot

dpinthinker, 2020-05-06:
I am just trying to run hifive1 image with qemu. 
The output shows tock os is running. 
But it is blocked. It show no interactive info or something else. To exit it, I have to kill current ssh session.

Did I use the qemu in a wrong way?

Below is the whole outputs:
```
➜  hifive1 git:(master) qemu-system-riscv32 -M sifive_e -kernel ../../target/riscv32imac-unknown-none-elf/release/hifive1.elf  -nographic
HiFive1 initialization complete.
Entering main loop.
interrupt triggered 3
```

ppannuto, 2020-05-06:
Are there any applications loaded on your image? It looks like the kernel boots successfully, then goes to a the "wait for interrupts" state waiting for some 'hardware' event to happen. I think things are working as expected.

alistair23, 2020-05-06:
That is working as expected. Tock is running in the main loop, you will need to load an app to get more functionality.

dpinthinker, 2020-05-07:
Thanks, I can run the hello_world.tpf with QEMU now.

1. But the other examples output just one line, like below:
```
LowLevelDebug: App 0x0 alert code 0x1
```
and then nothing else continues. 

Is it expected?

2. Can we run the qemu with graphic?

alistair23, 2020-05-07:
Great!

You are probably hitting this issue: https://github.com/tock/libtock-rs/issues/156

You can run QEMU without the `-nographic` option, that will run QEMU with graphics.
I'm not sure what that will get you though. The sifive_e has no way to do display out, no expandable bus (like PCIe) and AFAIK Tock doesn't support displays either.

closed 2020-05-10 by dpinthinker