tock/tock #1834: [RFC] `map_must` and other internal consistency support

ppannuto, 2020-05-07:
This came up recently in #1826, but variations have come up before (i.e. in the rf233 driver if memory serves) of "what do we do when fairly fundamental maps / takes fail?".

The current behavior is to silently drop operations. I wonder if something like a `map_must` might make sense. Borrowing from #1826, it would look like this:

```diff
unsafe fn set_syscall_return_value(&self, return_value: isize) {
     let mut stored_state = self.stored_state.get();
-    self.stored_state.map(|stored_state| {
+    self.stored_state.map_must(|stored_state| {
            self.chip
                .userspace_kernel_boundary()
                .set_syscall_return_value(self.sp(), stored_state, return_value);
     });
}
```

The idea being that in a release build nothing would change, but in a debug build `map_must` would `panic!` if the map were empty.

## Pros:

 - Improved semantics (the author is assuming this will / must work)
 - A path to removing silent failures
 - No overhead or added runtime panics for production

## Cons:

 - More differences between release/debug builds always make me nervous
 - Still fail silently at runtime in release builds; that may cause problems elsewhere

## Alternatives:

 - Writing explicit `_or` conditions everywhere that a map might be empty?
 - Doing nothing (silent failures in all cases)

I think there would be an analogous `take_must` and perhaps a few other situations where this idiom might make sense as well.

bradjc, 2020-05-07:
This could also go the other way: have a `map_maybe()` for times when it is expected that the cell might be empty, and have `map()` panic on empty.

But I think this is a good, but tricky, question. I agree it definitely feels weird to be forced to do the check, but then just completely ignore the failure case of the check. On the other hand, this is almost really just a language quirk since it doesn't seem to result in bugs very often (perhaps ever?). I've lost a lot of time to debugging and checking if maps fail, and it has never been the underlying issue. It seems to me at this point that the empty map case is a bit of a red herring.

I do wonder what would happen if we make empty `map()`s and the like panic while testing. I just tried with hail and the hail test app, and it's working fine.

alexandruradovici, 2020-05-08:
I can relate to the issue with map silently failing, I had this issue several times when writing capsules. As capsules are state machines, it is very easy to forget to replace a buffer, even more if there are several callbacks (client) functions that signal completed actions.

I started using ``map_or_else`` and ``panic!`` when else. I see two problems here:
1. if it panics and the system is deployed in an industrial environment and a non-essential capsule (like a screen, or status led) fails due to a unsuccessful map, the system might be able to continue, but should report the error.
2. if a map fails, the capsule might not call a callback function, thus locking an application or another capsule that is waiting for the result. A way of automatically calling some client callback upon failure would be useful. Not sure though how to implement it yet.

gendx, 2020-05-18:
I'm supportive of this idea of panicking on an empty map in debug/test builds! But I also wonder if the API is really suited for the purpose? It seems to me that it's around to circumvent borrow checker rules, when in most cases we can prove at compile time that there's a value?

Also, shouldn't we use `RefCell` instead of rolling our own `MapCell`? `RefCell` allows to check borrowing at runtime with `.borrow()`. If the value is already borrowed then `.borrow()` will panic, which sounds like the semantics we want. There is also `try_borrow` for a non-panicking version.

IMO it's better to use battle-tested types from the standard library rather than rolling our own cell types and realizing they are buggy or whatnot.

> This could also go the other way: have a `map_maybe()` for times when it is expected that the cell might be empty, and have `map()` panic on empty.

Regarding existing Rust semantics, `Option::map` works well with `None`, so I'd find it weird for it to panic. On the other hand, I like the idea of panicking by default to catch potential logic bugs. But I find the `map_must` naming weird.

Maybe we should be explicit, and have both `map_must` and `map_maybe`, but no `map`?

> But I think this is a good, but tricky, question. I agree it definitely feels weird to be forced to do the check, but then just completely ignore the failure case of the check. On the other hand, [...]

Citation needed? I think this can lead to bugs if the semantics aren't clear enough. Even if we don't see such bugs often in the master branch (but have we tested all capsules well to check that there are no such bugs?), there can be bugs "under the iceberg". For example, someone new to the project tries to use this `MapCell` type for the first time and wastes times debugging on their machine, but their bug doesn't appear on Github at all. I'd still consider this "bug while developing" a bug, because a better panic or a better API would have avoided this bug from happening (or avoided long debugging times).

> I've lost a lot of time to debugging and checking if maps fail, and it has never been the underlying issue. It seems to me at this point that the empty map case is a bit of a red herring.

The panic message in debug/test builds (whatever that is) would rule this out quickly, and would have avoided you losing a lot of time debugging and checking if maps fail ;)

gendx, 2020-05-18:
Also, panicking would invalidate the current purpose of `MapCell`: https://github.com/tock/tock/blob/3dd8ea4b08521b41fb5d25180576ef4284612eaa/libraries/tock-cells/src/map_cell.rs#L7-L8

And `RefCell` can also check borrowing at runtime without possible panics, as long as one uses `.try_borrow()`.

Besides, the memory layout is pretty much the same as `RefCell` (https://doc.rust-lang.org/stable/src/core/cell.rs.html#575-578), so I don't see the advantage of `MapCell` over `RefCell`. We could have some `MapCell` type that wraps a `RefCell`, but I don't think we should have `unsafe` code ourselves, rather than deferring the `unsafe` part to `RefCell`.