tock/tock #1835: Kerbnel panic when running c_hello on nucleu429zi

alexandruradovici, 2020-05-07:
I am getting the following panic when running any app on the nucleo429zi. It seems to have a problem with an allow to the console driver.

Running an empty kernel seems to work.

```
Initialization complete. Entering main loop

Kernel panic at arch/cortex-m4/src/lib.rs:297:
        "Kernel HardFault.
        Kernel version release-1.5-91-ge3c8b2e3
        r0  0x20000aec
        r1  0x20000acc
        r2  0x14
        r3  0xfffffff8
        r12 0x0
        lr  0xfffffff9
        pc  0xf101900a
        prs 0x8000001e [ N 1 Z 0 C 0 V 0 Q 0 GE 0000 ; ICI.IT 0 T false ; Exc 30-IRQn ]
        sp  0x20000a50
        top of stack     0x20001000
        bottom of stack  0x20000000
        SHCSR 0x0
        CFSR  0x1
        HSFR  0x40000000
        Instruction Access Violation:       true
        Data Access Violation:              false
        Memory Management Unstacking Fault: false
        Memory Management Stacking Fault:   false
        Memory Management Lazy FP Fault:    false
        Instruction Bus Error:              false
        Precise Data Bus Error:             false
        Imprecise Data Bus Error:           false
        Bus Unstacking Fault:               false
        Bus Stacking Fault:                 false
        Bus Lazy FP Fault:                  false
        Undefined Instruction Usage Fault:  false
        Invalid State Usage Fault:          false
        Invalid PC Load Usage Fault:        false
        No Coprocessor Usage Fault:         false
        Unaligned Access Usage Fault:       false
        Divide By Zero:                     false
        Bus Fault on Vector Table Read:     false
        Forced Hard Fault:                  true
        Faulting Memory Address: (valid: false) 0xE000ED34
        Bus Fault Address:       (valid: false) 0xE000ED38
"
        Kernel version release-1.5-91-ge3c8b2e3

---| Debug buffer not empty. Flushing. May repeat some of last message(s):
[0] function_call @0x8040055(0x804002c, 0x20004000, 0x2000, 0x20004c00)
[0] memop(0, 0x200049fc) = 0x0 = SUCCESS
[0] memop(10, 0x20004800) = 0x0 = SUCCESS
[0] memop(11, 0x200049fc) = 0x0 = SUCCESS

---| No debug queue found. You can set it with the DebugQueue component.

---| Fault Status |---
Instruction Access Violation:       true
Forced Hard Fault:                  true
Fault Status Register (CFSR):       0x00000001
Hard Fault Status Register (HFSR):  0x40000000

---| App Status |---
App: c_hello   -   [Running]
 Events Queued: 0   Syscall Count: 4   Dropped Callback Count: 0
                                                                 Restart Count: 0
                                                                                  Last Syscall: Some(ALLOW { driver_number: 1, subdriver_number: 1, allow_address: 0x20004820, allow_size: 15 })

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20006000═╪══════════════════════════════════════════╝
             │ ▼ Grant         984 |    984
  0x20005C28 ┼───────────────────────────────────────────
             │ Unused
  0x200049FC ┼───────────────────────────────────────────
             │ ▲ Heap            0 |   4652               S
  0x200049FC ┼─────────────────────────────────────────── R
             │ Data            508 |    508               A
  0x20004800 ┼─────────────────────────────────────────── M
             │ ▼ Stack          64 |   2048
  0x200047C0 ┼───────────────────────────────────────────
             │ Unused
             .....
  0x08040400 ┬─────────────────────────────────────────── F
             │ App Flash       980                        L
  0x0804002C ┼─────────────────────────────────────────── A
             │ Protected        44                        S
  0x08040000 ┴─────────────────────────────────────────── H

  R0 : 0x00000001    R6 : 0x00000000
  R1 : 0x00000001    R7 : 0x20004000
  R2 : 0x20004820    R8 : 0x00000000
  R3 : 0x0000000F    R10: 0x00000000
  R4 : 0x0000000F    R11: 0x00000000
  R5 : 0x0804009D    R12: 0x00000000
  R9 : 0x20004800 (Static Base Register)
  SP : 0x200047C0 (Process Stack Pointer)
  LR : 0x080400D9
  PC : 0x080401E2
 YPC : 0x080401E2

 APSR: N 0 Z 0 C 1 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00
       ThumbBit true

 Cortex-M MPU
  Region 0: [0x20004000:0x20006000], length: 8192 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20004000:0x20004400], Enabled
    Sub-region 1: [0x20004400:0x20004800], Enabled
    Sub-region 2: [0x20004800:0x20004C00], Enabled
    Sub-region 3: [0x20004C00:0x20005000], Disabled
    Sub-region 4: [0x20005000:0x20005400], Disabled
    Sub-region 5: [0x20005400:0x20005800], Disabled
    Sub-region 6: [0x20005800:0x20005C00], Disabled
    Sub-region 7: [0x20005C00:0x20006000], Disabled
  Region 1: [0x08040000:0x08040400], length: 1024 bytes; ReadOnly (0x6)
    Sub-region 0: [0x08040000:0x08040080], Enabled
    Sub-region 1: [0x08040080:0x08040100], Enabled
    Sub-region 2: [0x08040100:0x08040180], Enabled
    Sub-region 3: [0x08040180:0x08040200], Enabled
    Sub-region 4: [0x08040200:0x08040280], Enabled
    Sub-region 5: [0x08040280:0x08040300], Enabled
    Sub-region 6: [0x08040300:0x08040380], Enabled
    Sub-region 7: [0x08040380:0x08040400], Enabled
  Region 2: Unused
  Region 3: Unused
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug, run `make debug RAM_START=0x20004000 FLASH_INIT=0x8040055`
in the app's folder and open the .lst file.

```

Luiz-Monad, 2020-05-07:
Kerbal Panic would be a funny thing.

bradjc, 2020-05-07:
Hmm `0xE000ED34` is a very high address. Could this be a DMA issue?

alexandruradovici, 2020-05-07:
The fault is present when running an empty program as well.

bradjc, 2020-05-07:
What about no app?

Also I messed up reading my own debug output. That faulting address isn't relevant.

There does seem to be an issue with UART interrupts...the strace debug prints should be showing up. That they aren't typically indicates that the UART interrupt isn't be handled, which then causes the tx_done call in kernel/src/debug to not happen, which means more debugging output isn't printed. Is this on master, or, more specifically, is https://github.com/tock/tock/pull/1808 included in what you are running?

alexandruradovici, 2020-05-07:
> What about no app?
> 
Using make flash (kernel, no app) runs just fine, no fault.
> Also I messed up reading my own debug output. That faulting address isn't relevant.
>
> There does seem to be an issue with UART interrupts...the strace debug prints should be showing up. That they [...]

I am running the latest master.

There is no change in the dma1.rs file compared to a working version (before the chip crate refactor).

alexandruradovici, 2020-05-07:
I figured it out, the generic_isr vector is not in the right place section. Putting it right after the BASE_VECTOR seems to work. I'm trying to fix it.

commit 2d1bbdd7 (2020-05-11): Merge #1836

1836: Added init() function to stm32f4 crates r=ppannuto a=alexandruradovici

### Pull Request Overview

This pull request:
  1. adds an `init` function to the stm32f4 sub crates so that the object file is not ignored by the compiler and IRQS are included in the .irqs section
  2. moves the IRQS in the lib.rs file, they are ignored by the compiler otherwise

This fixes #1835.

It seems to be related to https://github.com/rust-lang/rust/issues/56639.

### Testing Strategy

This pull request was tested with a nucleo429zi board.

### TODO or Help Wanted

### Documentation Updated

-  [... cut]
PR #1836 (2020-05-11): Added init() function to stm32f4 crates
### Pull Request Overview

This pull request:
  1. adds an `init` function to the stm32f4 sub crates so that the object file is not ignored by the compiler and IRQS are included in the .irqs section
  2. moves the IRQS in the lib.rs file, they are ignored by the compiler otherwise

This fixes #1835.

It seems to be related to https://github.com/rust-lang/rust/issues/56639.

### Testing Strategy

This pull request was tested with a nucleo429zi board.

### TODO or Help Wanted

### Documentation Updated

- [x] updates are required.

### Formatting

- [x] Ra [... cut]
closed 2020-05-11 by bors[bot]