tock/tock #1843: "make flash" fails after flashing tock to board ST Nucleo F446RE

dpinthinker, 2020-05-12:
I can't flash image to board after I have flashed tock image to my new board ST Nucleo F446RE.

It outputs following Errors, when I flash again.
```
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f1x.cpu failed, GDB will be halted. Polling again in 100ms
```

After searched the internet, I got one way to fix it.

1. connect RESET and GND pin with jumper wire
2. execute flash command
3. disconnect jumper wire
4. execute flash command again

And there is a article shows that it may be caused by low power mode.
https://www.jsykora.info/2014/10/error-jtag-status-contains-invalid-mode-value-communication-failure-solved/

fhars, 2020-05-12:
"Works for me". Are you on the most current master, there were some STM32 changes committed yesterday evening that at least fixed the problems I had (hard fault when starting an app on the Nucleo-F446RE).

dpinthinker, 2020-05-12:
> "Works for me". Are you on the most current master, there were some STM32 changes committed yesterday evening that at least fixed the problems I had (hard fault when starting an app on the Nucleo-F446RE).

Yes, I used codes synced about six day before. After resynced, it works for me, too!

Thanks

dpinthinker, 2020-05-12:
If I disconnect the usb with board, and flash again, it still has the following issue.
If not disconnect the usb, it still works.

```
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f1x.cpu failed, GDB will be halted. Polling again in 100ms
```

fhars, 2020-05-13:
Intersting, after switching my computer off over the night, my setup now answers this to make flash:

openocd -f openocd.cfg -c "init; reset halt; flash write_image erase /home/hars/src/tock/target/thumbv7em-none-eabi/release/nucleo_f446re.elf; verify_image /home/hars/src/tock/target
/thumbv7em-none-eabi/release/nucleo_f446re.elf; reset; shutdown"
Open On-Chip Debugger 0.10.0                                                                                                                                                          
Licensed under GNU GPL v2                                                                                                                                                             For bug reports, read                                                                      
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz                                                                    
adapter_nsrst_delay: 100                                                                                                                                                              
none separate                                                                                                                                                                         
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2                                                                 
Info : Target voltage: 3.259513
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f4x.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Info : Previous state query failed, trying to reconnect       
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f4x.cpu failed, trying to reexamine        
Examination failed, GDB will be halted. Polling again in 300ms
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
Error: mem2array: Read @ 0xe0042004, w=4, cnt=1, failed
/usr/bin/../share/openocd/scripts/mem_helper.tcl:6: Error:                                                                                                                            
in procedure 'reset' 
in procedure 'ocd_bouncer' 
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 248
in procedure 'stm32f4x.cpu' called at file "embedded:startup.tcl", line 299
in procedure 'ocd_bouncer' 
in procedure 'mmw' 
in procedure 'mrw' called at file "/usr/bin/../share/openocd/scripts/mem_helper.tcl", line 25
at file "/usr/bin/../share/openocd/scripts/mem_helper.tcl", line 6

in procedure 'reset' 
in procedure 'ocd_bouncer'

make: *** [Makefile:17: flash] Fehler 1
```
Holding down reset while trying to flash results in:

```
openocd -f openocd.cfg -c "init; reset halt; flash write_image erase /home/hars/src/tock/target/thumbv7em-none-eabi/release/nucleo_f446re.elf; verify_image /home/hars/src/tock/target
/thumbv7em-none-eabi/release/nucleo_f446re.elf; reset; shutdown"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.262673
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
Error: timed out while waiting for target halted
[... 78 lines cut ...]
adapter speed: 1800 kHz                                                                    
Error: mem2array: Read @ 0xe0042004, w=4, cnt=1, failed
/usr/bin/../share/openocd/scripts/mem_helper.tcl:6: Error: 
in procedure 'reset'                       
in procedure 'ocd_bouncer'                                                                 
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 248
in procedure 'stm32f4x.cpu' called at file "embedded:startup.tcl", line 299
in procedure 'ocd_bouncer' 
in procedure 'mmw' 
in procedure 'mrw' called at file "/usr/bin/../share/openocd/scripts/mem_helper.tcl", line 25
at file "/usr/bin/../share/openocd/scripts/mem_helper.tcl", line 6

Info : Previous state query failed, trying to reconnect
Error: jtag status contains invalid mode value - communication failure
free(): double free detected in tcache 2
[... cut ...]
and the status LED is yellow. If I then try to flash again with reset held, if fails with the "timed out while waiting for target halted", and the status LED is green.

Another observation: The status LED turns read as soon as the USB connector is plugged in again. But if I hold reset while plugging in, it first blinks read and green until a timeout expires or I release the reset button, so the STLink seems to differentiate between a target that is not running at all, and a target that somehow configures itself to refuse to talk to the STLink.

So I think there is both a double free bug in openOCD an an unfortunate default chip setup in tock.

fhars, 2020-05-13:
I've reported the double free against my distribution's openocd package: https://bugs.launchpad.net/ubuntu/+source/openocd/+bug/1878404

fhars, 2020-05-13:
This has never worked, every commit I tested since https://github.com/tock/tock/commit/4b00dfda4f064f26e11c3d877be77d6602e9be26 had this behaviour.

It is also not the low-power issue suggested above.

fhars, 2020-05-13:
Here is a shorter way to get the board working again from the "red status LED" state:

First, run:
```
openocd -f boards/nucleo_f446re/openocd.cfg -c 'init; shutdown'
```
Now the status LED is yellow.

Press reset on the board.

Run
```
openocd -f boards/nucleo_f446re/openocd.cfg -c 'init; shutdown'
```
again.

If you get the timing right, pressing reset immediately after starting the first `openocd` call when the status LED starts to blink is sufficient, too.

fhars, 2020-05-13:
OK, it _is_ the `chip.sleep()` call in `kernel_loop`, and it looks like the processor does only honor the `DBG_SLEEP` bit in `DBGMCR_CR` if there is an active debug connection before entering `WFI` (the documentation isn't quite clear on that). So if you unplug and reconnect the board, you must start a debug session before tock enters `board_kernel.kernel_loop()`, which seems to happen after less than a millisecond.

If I change the code to wait ten seconds before entering the main loop, I can do a 
```
openocd -f openocd.cfg -c  'init; shutdown'
```
during those ten seconds, and everything works from then on, if I wait longer, debugging and flashing fails.

closed 2023-08-18 by bradjc