tock/tock #1844: crash_dummy doesn't output well on board  ST Nucleo F446RE

dpinthinker, 2020-05-12:
Just run the crash_dummy and press user button on board  ST Nucleo F446RE.
Then the crash_dummy application crashed as expected, but the outputs of UART is not good, just like following:
```
Initialization complete. Entering main loop

Kernel panic at <::core::macros::panic macros>:5:
	“Process crash_dummy had a fault”
	Kernel version release-1.5-116-g2d1bbdd7

—| No debug queue found. You can set it with the DebugQueue component.

—| Fault Status |—
Data Access Violation:              true
Forced Hard Fault:                  true
Faulting Memory Address:            0x00000000
Fault Status Register (CFSR):       0x00000082
Hard Fault Status Register (HFSR):  0x40000000

—| App Status |—
App: crash_dummy   -   [Fault]
 Events Queued: 0   Syscall Count: 6   Dropped Callback Count: 0
 Restart Count: 0
 Last Syscall: Some(YIELD)

 âââââââââââââ¤âââââââââââââââââââââââââââââââââââââââââââ
 â  Address  â Region Name    Used | Allocated (bytes)  â
 â0x20004000ââªâââââââââââââââââââââââââââââââââââââââââââ
             â â¼ Grant        1016 |   1016          
  0x20003C08 â¼âââââââââââââââââââââââââââââââââââââââââââ
             â Unused
  0x200029F0 â¼âââââââââââââââââââââââââââââââââââââââââââ
             â â² Heap            0 |   4632               S
  0x200029F0 â¼âââââââââââââââââââââââââââââââââââââââââââ R
             â Data            496 |    496               A
  0x20002800 â¼âââââââââââââââââââââââââââââââââââââââââââ M
             â â¼ Stack          56 |   2048          
  0x200027C8 â¼âââââââââââââââââââââââââââââââââââââââââââ
             â Unused
  0x20002000 â´âââââââââââââââââââââââââââââââââââââââââââ
             …..
  0x08040400 â¬âââââââââââââââââââââââââââââââââââââââââââ F
             â App Flash       976                        L
  0x08040030 â¼âââââââââââââââââââââââââââââââââââââââââââ A
             â Protected        48                        S
  0x08040000 â´âââââââââââââââââââââââââââââââââââââââââââ H

  R0 : 0x00000000    R6 : 0x080402E4
  R1 : 0x00000001    R7 : 0x20002000
  R2 : 0x00000000    R8 : 0x00000000
  R3 : 0x00000000    R10: 0x00000000
  R4 : 0x08040030    R11: 0x00000000
  R5 : 0x20002800    R12: 0xFFFFFFFF
  R9 : 0x20002800 (Static Base Register)
  SP : 0x200027C8 (Process Stack Pointer)
  LR : 0x080401B5
  PC : 0x080400AA
 YPC : 0x080401B4

 APSR: N 0 Z 1 C 1 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00
       ThumbBit true 

 Cortex-M MPU
  Region 0: [0x20002000:0x20004000], length: 8192 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20002000:0x20002400], Enabled
    Sub-region 1: [0x20002400:0x20002800], Enabled
    Sub-region 2: [0x20002800:0x20002C00], Enabled
    Sub-region 3: [0x20002C00:0x20003000], Disabled
    Sub-region 4: [0x20003000:0x20003400], Disabled
    Sub-region 5: [0x20003400:0x20003800], Disabled
    Sub-region 6: [0x20003800:0x20003C00], Disabled
    Sub-region 7: [0x20003C00:0x20004000], Disabled
  Region 1: [0x08040000:0x08040400], length: 1024 bytes; ReadOnly (0x6)
    Sub-region 0: [0x08040000:0x08040080], Enabled
    Sub-region 1: [0x08040080:0x08040100], Enabled
    Sub-region 2: [0x08040100:0x08040180], Enabled
    Sub-region 3: [0x08040180:0x08040200], Enabled
    Sub-region 4: [0x08040200:0x08040280], Enabled
    Sub-region 5: [0x08040280:0x08040300], Enabled
    Sub-region 6: [0x08040300:0x08040380], Enabled
    Sub-region 7: [0x08040380:0x08040400], Enabled
  Region 2: Unused
  Region 3: Unused
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug, run `make debug RAM_START=0x20002000 FLASH_INIT=0x8040059`
in the app’s folder and open the .lst file.

```

ppannuto, 2020-05-12:
What terminal are you using? Tock's debug prints include UTF-8 characters; that looks like your terminal is not rendering them correctly.

dpinthinker, 2020-05-13:
@ppannuto Thanks, it works since switched to UTF-8.

closed 2020-05-13 by dpinthinker