tock/tock #1849: How to install multiple apps on the board stm32 nucleo-f446re

takoyakiyo, 2020-05-13:
I tried to use the following command twice to install two different applications, but the second use overwrote the first .app section, and only one application can be displayed, so what should I do if I want to install two applications, thank you.
```
$ arm-none-eabi-objcopy  \ 
    --update-section .apps=../../../libtock-c/examples/c_hello/build/cortex-m4/cortex-m4.tbf \. 
    target/thumbv7em-none-eabi/debug/nucleo_f446re.elf \
    target/thumbv7em-none-eabi/debug/nucleo_f446re-app.elf
$ sudo opened -f opened.cfg -c "init; reset halt; flash write_image erase target/thumbv7em-none-eabi/debug/nucleo_f446re-app.elf; verify_image target/thumbv7em-none-eabi/debug/nucleo_f446re-app.elf; reset; shutdown"
```

alistair23, 2020-05-13:
You can just cat the tbf files into a single binary and then use that for the `.apps` section.

So something like:
```shell
cat app1 > all_apps
cat app2 >> all_apps
arm-none-eabi-objcopy --update-section .apps=all_apps ...
```

bradjc, 2020-05-13:
Hmm, maybe this is something tockloader should help with? A `tockloader consolidate-binary` (or something) command? I think that `tockloader consolidate-binary --arch cortex-m4 kernel.elf app1.tab app2.tab` would be pretty doable. One advantage would be tockloader could apply its rules about ordering and spacing apps. One downside would be the paths for all of the files would still be really messy, so it wouldn't exactly be easy to use.

takoyakiyo, 2020-05-14:
> You can just cat the tbf files into a single binary and then use that for the `.apps` section.
> 
> So something like:
> 
> ```shell
> cat app1 > all_apps
> cat app2 >> all_apps
> arm-none-eabi-objcopy --update-section .apps=all_apps ...
> ```

Now I can load two applications on the board, thank you very much.
I want to experiment whether an application has an impact on other applications when it crashes. But when I load the two applications cxx_hello(print hello in a loop) and crash_dummy(press user button to generate an application fault), I press the button and get the following application memory debug information. The status of cxx_hello is displayed as yielded and I can’t  get the “hello” output no longer. When I replace crash_dummy with a simple crash program of my own(divide 0 in the cxx_hello_crash), the status of cxx_hello is displayed as running and I can’t get the “hello” output either. I notice that crash_dummy makes a svc call in the code, but I don’t quite understand. Is cxx_hello still running normally? How can I verify it?
```
---| App Status |---
App: cxx_hello   -   [Yielded]
 Events Queued: 0   Syscall Count: 95   Dropped Callback Count: 0
 Restart Count: 0
 Last Syscall: Some(YIELD)
App: crash_dummy   -   [Fault]
 Events Queued: 0   Syscall Count: 6   Dropped Callback Count: 0
 Restart Count: 0
 Last Syscall: Some(YIELD)
```
```
---| App Status |---
App: cxx_hello   -   [Running]
 Events Queued: 0   Syscall Count: 5   Dropped Callback Count: 0
 Restart Count: 0
 Last Syscall: Some(MEMOP { operand: 1, arg0: 436 })
App: cxx_hello_crash   -   [Fault]
 Events Queued: 0   Syscall Count: 3   Dropped Callback Count: 0
 Restart Count: 0
 Last Syscall: Some(MEMOP { operand: 11, arg0: 536889828 })
```

takoyakiyo, 2020-05-14:
> Hmm, maybe this is something tockloader should help with? A `tockloader consolidate-binary` (or something) command? I think that `tockloader consolidate-binary --arch cortex-m4 kernel.elf app1.tab [...]

Because I find that tockloader doesn't support the board which I use, so I choose the method mentioned in the documentation of my board.

ppannuto, 2020-05-14:
`cxx_hello` only prints hello once and then stops. I would suggest choosing an application that runs continuously, `blink` is probably easiest or `hello_loop` if you want to see something on the serial port.

takoyakiyo, 2020-05-14:
> `cxx_hello` only prints hello once and then stops. I would suggest choosing an application that runs continuously, `blink` is probably easiest or `hello_loop` if you want to see something on the serial port.

`cxx_hello` prints hello cyclically like `hello_loop`, and `c_hello` prints only once, maybe you mean `c_hello`. But I will try `blink` or `hello_loop`. Thanks for your reply.

bradjc, 2020-05-14:
By default, most boards in Tock panic and print the full debugging information you see when _any_ app crashes. The other apps are no longer running, and the chip is either sleeping or more likely in a while(1) loop.

It sounds like what you want to do is change the FAULT_RESPONSE in the board main.rs file. You can look at the hail board for an example of one that does not default to panic immediately.

Also, this is a great thing to look into and I would like to see what you find. We just recently added robust support for apps crashing while not disturbing other apps, but there are probably more issues yet to find.

takoyakiyo, 2020-05-15:
> By default, most boards in Tock panic and print the full debugging information you see when _any_ app crashes. The other apps are no longer running, and the chip is either sleeping or more likely in [...]

1.May I ask how to understand that when I use `crash_dummy` to get an error, the status of another normal application is displayed as yielded, and when I use the application which contains a divide-by-zero operation to get an error, the status of another normal application is displayed as running.
2.The current situation is that after an application is crashed, other normal applications also stop running. I wonder if I understand correctly? And Is the system still in normal？Or it has crashed and stopped?
Thanks.

bradjc, 2020-05-15:
1. The status of the other app when the kernel calls `panic!()` is not related to _why_ the first app failed. It is related to _when_ the first app failed, and you are just seeing the current state of the process as tracked by the kernel.

    Note that the `RUNNING` state is true for a process both when the process is actually executing (which is not the case you are seeing since a context switch back the kernel has happened for it to call `panic!()`) and when the process is ready to execute (the case that you are seeing). It just so happens that the second process has a callback enqueued for it and therefore has useful work to do when the first process crashes in some cases, and in other cases it is simply yielded and waiting on some event to occur.

2. If `FAULT_RESPONSE` is set to panic, then all processes and the kernel stop when any process crashes.

takoyakiyo, 2020-05-18:
> * The status of the other app when the kernel calls `panic!()` is not related to _why_ the first app failed. It is related to _when_ the first app failed, and you are just seeing the current state of the process as tracked by the kernel.
>   Note that the `RUNNING` state is true for a process both when the process is actually executing (which is not the case you are seeing since a context switch back the kernel has happened for it to call `panic!()`) and when the process is ready to execute (the case that you are seeing). It just so happens that the second process has a callback enqueued for it and therefore has useful work to do when the first process crashes in some cases, and in other cases it is simply yielded and waiting on some event to occur.
> * If `FAULT_RESPONSE` is set to panic, then all processes and the kernel stop when any process crashes.

When I set `FaultResponse` to `Stop`, after one process crashes, other processes are still running and continue to output. Does this mean that after one process crashes, other processes are not affected.

ppannuto, 2020-05-18:
> Does this mean that after one process crashes, other processes are not affected.

Yes. Tock is designed to provide process isolation (to the extent possible). We would generally consider it a bug if one process's behavior, including crashing, were to affect another process.

takoyakiyo, 2020-05-20:
> When I set `FaultResponse` to `Stop`, after one process crashes, other processes are still running and continue to output. Does this mean that after one process crashes, other processes are not [...]

When the crashed app was stopped, was the ram area he used released? Is there a way to verify this

Loadprocess from flash into
Can the address written to ram be changed?

> > Does this mean that after one process crashes, other processes are not affected.
> 
> Yes. Tock is designed to provide process isolation (to the extent possible). We would generally consider it a bug if one process's behavior, including crashing, were to affect another process.

1. When the crashed app was stopped, was the ram area he used released? Is there a way to view the usage of ram？
2. Can the address written to ram be changed?
```
Loaded process[0] from flash=[0x08040000:0x08044000] into sram=[0x20002000:0x20004000] = Some("cxx_hello")
```
thanks

takoyakiyo, 2020-05-20:
> > Does this mean that after one process crashes, other processes are not affected.
> 
> Yes. Tock is designed to provide process isolation (to the extent possible). We would generally consider it a bug if one process's behavior, including crashing, were to affect another process.

1. When the crashed app was stopped, was the ram area he used released? Is there a way to view the usage of ram？
2. Can the address written to ram be changed?
```
Loaded process[0] from flash=[0x08040000:0x08044000] into sram=[0x20002000:0x20004000] = Some("cxx_hello")
```
thanks

bradjc, 2020-05-20:
1. No, we currently don't have any memory free and re-allocate code.
2. The board's main.rs is what specifies the ram buffer to use for apps, so passing a different buffer would use different addresses. Also, you can write a custom `load_processes()` function if you really wanted to use special addresses.

takoyakiyo, 2020-05-20:
> * No, we currently don't have any memory free and re-allocate code.
> * The board's main.rs is what specifies the ram buffer to use for apps, so passing a different buffer would use different addresses. Also, you can write a custom `load_processes()` function if you really wanted to use special addresses.

When the crashed app was stopped, was the ram area he used keep the state at the moment of the crash unchanged? If I use `tockloader uninstall` to uninstall the application, will this ram area be released?

bradjc, 2020-05-20:
When the app has crashed and stopped, yes, its memory is (currently) mostly preserved (I don't think Tock makes any guarantees about this, but there is no code that clears the memory or anything right now). You can look at the implementation of `terminate()` https://github.com/tock/tock/blob/master/kernel/src/process.rs#L1968 for exactly what happens. The grant region pointers are reset to free up those resources in the kernel, but that isn't in the apps normal working memory section.

All bets are off with something like `tockloader uninstall`. In practice running that causes multiple resets, at which point anything in RAM is lost. But even if that operation did not reset the MCU there is no attempt whatsoever to preserve any memory state.

wu6shen, 2020-05-21:
> When the app has crashed and stopped, yes, its memory is (currently) mostly preserved (I don't think Tock makes any guarantees about this, but there is no code that clears the memory or anything [...]

[image]
I am sorry that I am a raw hand, did you mean that the grant area will be released for reuse, and the stack, data, heap will remain the same as before?
Also, I found that by using `tockloader` operation, the system will stop and can only be restarted after press reset. Is there a way to dynamically load or delete applications when the system and other applications are running?

takoyakiyo, 2020-05-21:
> When the app has crashed and stopped, yes, its memory is (currently) mostly preserved (I don't think Tock makes any guarantees about this, but there is no code that clears the memory or anything [...]

[image]
I am sorry that I am a raw hand, did you mean that the grant area will be released for reuse, and the stack, data, heap will remain the same as before?
Also, I found that by using `tockloader` operation, the system will stop and can only be restarted after press reset. Is there a way to dynamically load or delete applications when the system and other applications are running?

bradjc, 2020-05-21:
> Do you mean that the grant area will be cleared for reuse

Almost. The actual grants themselves won't be changed (I don't think). What will change are the pointers to where those grants were.

> and the stack, data, heap will remain the same as before?

This is correct.

> Also, I found that by using `tockloader` operation, the system will stop and can only be restarted after press reset. Is there a way to dynamically load or delete applications when the system and other applications are running? For example, if there are 3 Apps arranged in order(`a`, `b`, `c`), can I uninstall b and then load a new App `d` to the position(both ram and flash) of `b` without affecting the system and `a` and `c`. I tried to do this but failed.

Short answer: no.

This would require the kernel (or a trusted application) to have the capability to save apps into flash, which currently doesn't exist. (Right now we rely on the bootloader or a JTAG tool to do this loading, which is why a reset is required.) Mechanically, loading a TBF binary to the board and saving it to flash, and then calling `process.create()` shouldn't be _too_ difficult, as we have all of the individual pieces (UART, flash write, process creation). There has also been interest in doing this periodically. I think the main reason it hasn't actually been created is that doing it "right" involves some subtle details, for example:
- Making sure the app binary is loaded to a correctly aligned address for the MPU
- Verifying that the mechanism doesn't violate Tock's threat model
- Restricting permissions so that app loading is done safely
- Ensuring the application has enough memory

takoyakiyo, 2020-05-21:
> Almost. The actual grants themselves won't be changed (I don't think). What will change are the pointers to where those grants were.
>
>
> This is correct.
>
>
> Short answer: no.
>
> This would require the [...]

After the process is stopped, the kernel will no longer access the process's memory, why do I still need to reset the `grant pointers`？

bradjc, 2020-05-21:
> After the process is stopped, the kernel will no longer access the process's memory, why do I still need to reset the `grant pointers`？

That is used as a signal to capsules that the process (and its grant regions) are no longer valid. It also ensures that if the process is restarted that new grant regions are allocated for the process. I think there could be other ways to implement this, but resetting the grant pointers is a straightforward way to achieve both of these goals.

takoyakiyo, 2020-05-26:
> Hmm, maybe this is something tockloader should help with? A `tockloader consolidate-binary` (or something) command? I think that `tockloader consolidate-binary --arch cortex-m4 kernel.elf app1.tab [...]

Now I try to use `tockloader` to install multiple Apps, I used the following command.
```
tockloader install --openocd --openocd-board openocd.cfg --arch cortex-m4 --board nucleo_f446re --app-address 0x08040000 --page-size 1024 c_hello.tab
tockloader install --openocd --openocd-board openocd.cfg --arch cortex-m4 --board nucleo_f446re --app-address 0x08040000 --page-size 1024 cxx_hello.tab
```
But I can't install the two apps successfully. When I added `--bundle-apps` to the command like this
```
tockloader install --openocd --openocd-board openocd.cfg --arch cortex-m4 --board nucleo_f446re --app-address 0x08040000 --page-size 1024 --bundle-apps c_hello.tab
tockloader install --openocd --openocd-board openocd.cfg --arch cortex-m4 --board nucleo_f446re --app-address 0x08040000 --page-size 1024 --bundle-apps cxx_hello.tab
```
The installation was successful, I noticed that `--bundle-apps` forces tockloader to write all apps as a concatentated bundle using only a single flash command. This will require that anytime any app changes in any way (e.g. its header changes or the app is updated or a new app is installed.) all apps are re-written. By using `--bundle-apps`, the apps need to reinstall when I install a new App, but I want to use a method to make the app load independently. How can I modify the command? Thanks.

bradjc, 2020-05-26:
Typically this issue stems from the smallest supported erase size being larger than the app being flashed, and possibly alignment issues with erase and flashing apps. It's also a little difficult to know if the issue lies with the chip itself, the openocd configuration, or the openocd implementation.

We have two workarounds for this (so far). One is `--bundle-apps`. The other is forcing a minimum application size here: https://github.com/tock/tockloader/blob/master/tockloader/tockloader.py#L68. `--bundle-apps` is the brute force approach, as you have seen. Setting a minimum app size allows for independent app loading, but can be very wasteful if the required minimum size is quite large. If you have other ideas, let me know.

closed 2020-07-15 by bradjc