tock/tock #1860: Possible RISC-V PMP improvements

alphan, 2020-05-19:
Creating this issue to keep track of things that I noticed while working on #1822: 
- [X] `disable_mpu` should be a no-op (similar to `enable_mpu`) since permissions are not enforced on M-mode accesses (unless the lock bits are set, which we don't).
    Note: #1822 moves its implementation to `PMPConfig` and calls it from `configure_mpu` but it looks like we don't really need it.
- [ ] `rv32i::pmp::PMPConfig` is used as both `MPUConfig` (in `arch/rv32i/src/pmp.rs`) and `MPU` (in `chips/ibex/src/chip.rs`), which is confusing.
- [ ] Do the regions really have to be sorted each time the MPU is configured? It looks like we can sort the list when it is modified if we can keep track of the app memory region instead of fixing its location.

bradjc, 2020-05-29:
Hopefully a fair summary of the discussion so far:

- `disable()` will remain as a utility function for anyone who might want it, but it will be renamed to "clear" or something to imply that it _cannot_ guarantee that it can disable the MPU (certain hardware may lock access to _any_ software in a certain configuration).
- A new function "disable for kernel" or some name will replace the current `disable()` call. On current platforms this will be a no-op.
- Protecting the kernel from itself will be a new trait. The current trait will be clearly labeled as for applications.
- Performance optimizations will be left to specific implementations (subject to minor trait changes as needed). We don't expect too many implementations to ever exist, and if caching can be done internally to the interface, then better to not have to change the interface.

alistair23, 2020-05-29:
Sounds good :)

That is what is being done here: https://github.com/tock/tock/pull/1873

bradjc, 2021-08-06:
Over time I think these issues have been addressed.

closed 2021-08-06 by bradjc