tock/tock #1864: Fix favicon for rustdoc generated docs

ppannuto, 2020-05-19:
Resurrecting the second bullet from #1334 

- [ ] Figure out how to fix favicon / logo more generally (here's a start https://github.com/tock/tock/tree/doc-updates)

JOE1994, 2020-06-03:
The second link (https://github.com/tock/tock/tree/doc-updates) gives a 404 error 🦀

ppannuto, 2020-06-05:
Hm.., looks like that branch has been deleted from github and isn't on my laptop :(

However, I remember that pointer to some ugly hack, and it looks like the tooling may have improved that an ugly hack is no longer necessary:
https://doc.rust-lang.org/rustdoc/the-doc-attribute.html#html_favicon_url

While we're there, it'd be nice to update the `html_logo_url` as well.

closed 2021-10-15 by ppannuto