tock/tock #1876: Run an app on RISC-V using qemu

alexandruradovici, 2020-05-24:
I am teaching this week a class on TockOS and I would like the students to be able to run a minimal app. The only possible simulation way that I found is to use risc-v.

I am able to run the kernel fine, but when running the c_hello application I get EBUSY for the console command syscall.

```
qemu-system-riscv32 -M sifive_e,revb=true -kernel /Users/alexandru/programe/tock/tock/target/riscv32imac-unknown-none-elf/release/hifive1.elf -device loader,file=../../../libtock-c/examples/c_hello/build/rv32imac/rv32imac.tbf,addr=0x20040000 -nographic
HiFive1 initialization complete.
Entering main loop.
[0] function_call @0x20040068(0x20040040, 0x80002400, 0x1b50, 0x80003000)
[0] memop(0, 0x8000334c) = 0x0 = SUCCESS
[0] memop(10, 0x80002800) = 0x0 = SUCCESS
[0] memop(11, 0x8000334c) = 0x0 = SUCCESS
[0] allow(0x1, 1, @0x0, 0xf) = 0x0 = SUCCESS
[0] remove_pending_callbacks[0x1:1] = 0 callback(s) removed
[0] subscribe(0x1, 1, @0x200400c0, 0x0) = 0x0 = SUCCESS
[0] cmd(0x1, 1, 0xf, 0x0) = 0xfffffffe = EBUSY
[0] yield
```

I had to decrease the STASK_SIZE to 512 and to increase the .app_memory space:

```rust
// RAM to be shared by all application processes.
#[link_section = ".app_memory"]
static mut APP_MEMORY: [u8; 7 * 1024] = [0; 7 * 1024];
```

I am using the manually built qemu from `make emulation-check`.

Any feedback would be very useful.

alistair23, 2020-05-25:
I'm not sure what is wrong from the top of my head. This works as it's used for testing libtock-rs: https://github.com/tock/libtock-rs/commit/b4889dfbbd738c6f945d9744cab3fcf0dc1a36b0

alexandruradovici, 2020-05-26:
I think I found the problem, global variables are not relocated. 

```c
char hello[] = "Hello World!\r\n";

static void nop(
  int a __attribute__((unused)),
  int b __attribute__((unused)),
  int c __attribute__((unused)),
  void* d __attribute__((unused))) {}

int main(void) {
  putnstr_async(hello, sizeof(hello), nop, NULL);
  return 0;
}
```

In the code above ``hello`` has the address 0. elf2tab seems to be the correct one:

```
ELF2TAB_ARGS += --fixed-address-ram 0x80002400 --fixed-address-flash 0x20040040
```

Does this work on a real hardware?

alistair23, 2020-05-26:
Aw, that looks like libtock-c. libtock-c apps do work, but you will need to manually tweak the linker file to lineup the sections. It is a huge pain, I would recommend not using libtock-c

alexandruradovici, 2020-05-26:
Yes, it is libtock-c. I have to use it as I need some C examples first. Using local variables will do.

I looked at the libtock-rs linker file, I was wondering why this does not have the issue? No use of global variables maybe?

alistair23, 2020-05-26:
It's not related to local varialbes. You need to make sure the addresses in the headers (which are used in `_c_start()`) line up with the sections in the linker file.

closed 2020-07-15 by bradjc