tock/tock #1877: Make clean is missing the target dir

dc740, 2020-05-26:
I was having issues with files not being deleted and wrong code being flashed into my board after pulling the latest changes in master, and discovered and issue in the `clean` task.

https://github.com/tock/tock/blob/b9804f35c4caa1d70feee625cd77e9d725b000ae/boards/Makefile.common#L183

Since the target repository was changed globally, it should also be specified when calling cargo clean, or it will keep cleaning the old target repository.

I'm creating the bug to double check this is correct, and I'll create a new PR with the change after confirming it.

The code should be
```
.PHONY: clean
clean::
	$(Q)$(CARGO) clean $(VERBOSE)  --target-dir $(TOCK_ROOT_DIRECTORY)target
```

ppannuto, 2020-05-26:
I don't think this is necessary with cargo workspaces? I just ran some simple tests:

1. Clean, nothing built repo
2. `make allboards`
3. du -hc target/` -> 223M
4. `make clean` (target directory ultimately deleted)
5. `cd boards/hail`
6. `make`
7. `du -hc ../../target/` -> 51M
8. `make clean`
9. (target deleted again)

Is there anything not being cleaned/deleted that you expect to be removed?

dc740, 2020-05-26:
I'm not able to reproduce the problem that ended up in this ticket.

I was having a weird issue where code from a module I removed was still compiled in the binary, to the point I was actually getting interrupts from it. It went away after cleaning like this, but I can't confirm anything. I'll close it and re-open if I ever get this issue again. Thanks.

closed 2020-05-26 by dc740