tock/tock #1890: Add a file system for Tock OS

chongyang0515, 2020-05-29:
I want to read file from the SD card use Tock OS. so i hope add the fatfs component. I have found several proper fatfs:[fat-fs](https://gitlab.com/susurrus/fat-rs) and [rust-fatfs](https://github.com/rafalh/rust-fatfs), Can I add them to the Tock OS?

brghena, 2020-05-29:
Last time I remember putting a filesystem on an SD card with Tock, we did so in an application. Using the [C userland](https://github.com/tock/libtock-c), we wrote an app that used the very popular [FatFS implementation](http://elm-chan.org/fsw/ff/00index_e.html).

Here was the app: https://github.com/lab11/signpost-software/blob/master/signpost/apps/storage_master/test/fat_test/main.c There was also the necessary shim layer that was written here: https://github.com/lab11/signpost-software/blob/f13b32e0661849024b4817ceb08f47fe980277ff/signpost/apps/libsignpost-tock/mmc_io.c As a warning, these files were written like 4 years ago and I don't promise they will just work as-is.

This was a pretty straightforward solution and worked pretty well because only the application needed to interact with the filesystem (not the OS kernel) anyways.

chongyang0515, 2020-05-30:
I've seen the projects you recommended. It is used in the app environment.  I want to know whether the tock kernel plans to support SD cards or FATFS. I want to use the tock os to read files from SD cards. Can I see the roadmap of the tock if possible?

lschuermann, 2020-05-30:
I think this is rather closely linked to some ongoing discussions around nonvolatile storage organization and persistent application storage in general. AFAIK this is something to discuss and the project hasn't settled on a single approach here.

Whether a general purpose in-kernel FS implementation makes sense is one thing, how to expose it to userspace _and_ still create a safe and mutually distrustful environment for apps is another. I know of #1692 and [this ML thread](https://groups.google.com/forum/#!topic/tock-dev/aduN7fHWXdI) but there could be further relevant discussions here.

bradjc, 2020-06-02:
#1692 is one place for relevant discussion, and that is on the OpenTitan-specific roadmap (https://github.com/tock/tock/blob/master/doc/wg/opentitan/roadmap.md).

I image a file system could entirely be implemented as a capsule, and we would be eager to add that to Tock. But I don't know of any ongoing efforts to port a filesystem to a Rust capsule.

closed 2023-08-18 by bradjc