tock/tock #1912: How to use sdcard capsules in stm32fxx board

chongyang0515, 2020-06-06:
I want to add the sdcard read/write function to the stm32fxx development board. I find that the **sdcard.rs** file is included in the capsules, but there is no call related to the sdcard.rs in the **board** folder. My development board is nucleo_f446re. is it possible to implement the read and write on sdcard ?   What steps do I need to take?

brghena, 2020-06-06:
You'll need to create the capsule in your own board `main.rs` file. Generally, you need a SPI device that connects to the SD Card and an Alarm (for timing). You provide references to them to the SD Card capsule, and then you can call functions on the SD Card capsule. It also optionally takes a GPIO pin that detects if the SD Card is installed or not (but it works fine assuming that the card is there without one).

A specific example for initializing the capsule is documented here: https://github.com/tock/tock/blob/master/capsules/src/sdcard.rs#L5 Particularly if you want to use the SD card from the kernel you'll want lines 9-26 (28-31 set up a syscall driver to access the SD card from userspace applications). You'll need to modify various parts of that for your system particularly the detect_pin if you have one.

Here is an example from the last board to use the SD Card. https://github.com/lab11/signpost-software/blob/master/signpost/kernel/boards/storage_master/src/main.rs#L255 My warning is that  this board is about 3 years old and not necessarily all valid Tock code anymore.

To interact with the SD Card, you'll want to write an SDCardClient, which will take in a reference to the SD Card capsule (like the syscall driver from lines 28-31 does) and implements the callback functions from it. Then you can have it call initialize() on the SD card capsule, wait for a callback of the `init_done()` function, and then begin doing filesystem stuff.

That was a lot of stuff. So feel free to ask questions!

takoyakiyo, 2020-06-08:
> You'll need to create the capsule in your own board `main.rs` file. Generally, you need a SPI device that connects to the SD Card and an Alarm (for timing). You provide references to them to the SD [...]

I tried to add a capsule in stm32f446re board by imitating the example, but I got an error like this `the trait `kernel::hil::spi::SpiMaster` is not implemented for `&'static stm32f4xx::usart::Usart<'static>`, I want to achieve this transplant by myself, but it’s a little difficult for me as a novice, Is there any other simple and quick way to make me use the sd card? Or I can only implement the missing trait by myself.

brghena, 2020-06-08:
While the SAM4L has a USART peripheral that acts as a SPI or a UART, the STM32F4xx series actually has a separate SPI peripheral. You should be able to use https://github.com/tock/tock/blob/master/chips/stm32f4xx/src/spi.rs which does implement SpiMaster

alexandruradovici, 2020-06-17:
Take a look at https://github.com/tock/tock/blob/master/boards/stm32f3discovery/src/main.rs, the l3gd20 sensor and its component, https://github.com/tock/tock/blob/master/boards/components/src/l3gd20.rs, use SPI (with an stm32f43xx). It is similar for the stm32f4.

hudson-ayers, 2020-10-27:
Closing due to inactivity, please feel free to reopen if you have any more questions!

closed 2020-10-27 by hudson-ayers