tock/tock #1920: tools: Check that system has correct markdown-toc

ppannuto, 2020-06-08:
This is a nasty gotcha to get caught up on. There is both a [python markdown-toc](https://pypi.org/project/markdown-toc/) and a [node markdown-toc](https://www.npmjs.com/package/markdown-toc), which install CLI commands _with different behavior_.

If a user happened to already have the Python one installed, then our tooling will do the wrong thing. We need to validate in our infra that the `markdown-toc` we are calling is going to do what we expect.

A few fixes:
 - Easiest: Install markdown-toc locally in our `tools/` folder and invoke it instead of relying on system.
 - Better? Find a rust or package that does this. I think this is the only node dependency in Tock, it'd probably be better to just be rid of it.

bradjc, 2020-06-09:
I was thinking recently that what would be really awesome for things like this and the chips HIL support table (and whatever else) is to remove the expectation for developers to worry about it, and instead just have a job run once a day that checks if anything changed, and if so, automatically add a commit. I don't know how viable that is, however.

hudson-ayers, 2021-01-22:
I think that would be pretty easy with the recent improvements to github actions.

Scheduling a periodic action is very easy: https://docs.github.com/en/actions/reference/events-that-trigger-workflows#scheduled-events 

and adding commits should also ve easy: https://lannonbr.com/blog/2019-12-09-git-commit-in-actions 

If we are concerned about automated commits to master, we could have a bot submit PRs periodically instead: https://github.com/peter-evans/create-pull-request

bradjc, 2021-02-02:
I implemented a PR which switches to markdown-toc (implemented in Rust). That work is here: https://github.com/tock/tock/tree/toc-rust

However, as of today it doesn't work very well, and fails to generate the correct links in at least two different cases. I have submitted issues to the project's github.

bradjc, 2021-04-23:
No progress on this https://github.com/pbzweihander/markdown-toc/issues

bradjc, 2023-08-07:
I would rather move the documentation to the book and use mdbook's toc generator at this point.

lschuermann, 2023-08-09:
My 2ct:
- Keeping the technical documentation in the kernel repo is very nice, as it allows us to reference code files via relative symlinks, and update documentation in tandem with code in the same PR.
- Having the docs as dead markdown files in the kernel repo is suboptimal. The book would be a great place to _view_ this documentation.

Would there be an option to keep the source in the kernel repo, but render this as part of the book?

bradjc, 2023-08-09:
> My 2ct:
>
>     * Keeping the technical documentation in the kernel repo is very nice, as it allows us to reference code files via relative symlinks, and update documentation in tandem with code in the [...]

We could try that.

I still think it's time to decouple docs from implementation, but perhaps it could be a slow transition.

bradjc, 2023-08-18:
It would be nice to not have to run CI on rust code for documentation PRs.

ppannuto, 2023-08-18:
I believe that we could use logic similar to what already runs in our 'label-er' jobs that detect when changes only touch `doc/` and skip and/or return-success-without-testing the code-oriented CI jobs

bradjc, 2024-01-09:
https://github.com/pbzweihander/markdown-toc is getting closer.