tock/tock #1921: [CI] Missing board in "CI-Job: Syntax"?

gendx, 2020-06-09:
When running `make ci-nosetup` (after `make clean` to be sure), the output for "CI-Job: Syntax" seems quite short. The only mentioned board is `nrf52dk`. Is it working as intended?

```
********************************************************************************
* CI-Job: Syntax                                                               *
********************************************************************************

make[2]: Entering directory '/tock'
cargo check
   Compiling nrf52dk v0.1.0 (/tock/boards/nordic/nrf52dk)
    Finished dev [optimized + debuginfo] target(s) in 0.31s
make[2]: Leaving directory '/tock'
```

hudson-ayers, 2020-06-09:
I think that is just the result of earlier stages of `make ci-nosetup` having compiled everything, so `cargo check` doesn't have to do anything. If you run `make clean` then `make ci-job-syntax` You can see that it builds every board crate:
```
(.venv) hudson: ~/tock (master) $ make ci-job-syntax

********************************************************************************
* CI-Job: Syntax                                                               *
********************************************************************************

make[1]: Entering directory '/home/hudson/tock'
cargo check
    Checking tock-registers v0.5.0 (/home/hudson/tock/libraries/tock-register-interface)
    Checking tock-cells v0.1.0 (/home/hudson/tock/libraries/tock-cells)
    Checking tock-rt0 v0.1.0 (/home/hudson/tock/libraries/tock-rt0)
    Checking enum_primitive v0.1.0 (/home/hudson/tock/libraries/enum_primitive)
   Compiling launchxl v0.1.0 (/home/hudson/tock/boards/launchxl)
   Compiling nrf52dk v0.1.0 (/home/hudson/tock/boards/nordic/nrf52dk)
   Compiling arty_e21 v0.1.0 (/home/hudson/tock/boards/arty_e21)
   Compiling hail v0.1.0 (/home/hudson/tock/boards/hail)
   Compiling acd52832 v0.1.0 (/home/hudson/tock/boards/acd52832)
   Compiling nucleo_f429zi v0.1.0 (/home/hudson/tock/boards/nucleo_f429zi)
   Compiling imix v0.1.0 (/home/hudson/tock/boards/imix)
   Compiling nrf52840dk v0.1.0 (/home/hudson/tock/boards/nordic/nrf52840dk)
   Compiling hifive1 v0.1.0 (/home/hudson/tock/boards/hifive1)
   Compiling redboard_artemis_nano v0.1.0 (/home/hudson/tock/boards/redboard_artemis_nano)
   Compiling nucleo_f446re v0.1.0 (/home/hudson/tock/boards/nucleo_f446re)
   Compiling stm32f3discovery v0.1.0 (/home/hudson/tock/boards/stm32f3discovery)
   Compiling nrf52840_dongle v0.1.0 (/home/hudson/tock/boards/nordic/nrf52840_dongle)
   Compiling opentitan v0.1.0 (/home/hudson/tock/boards/opentitan)
    Checking kernel v0.1.0 (/home/hudson/tock/kernel)
    Checking riscv-csr v0.1.0 (/home/hudson/tock/libraries/riscv-csr)
    Checking cortexm v0.1.0 (/home/hudson/tock/arch/cortex-m)
    Checking capsules v0.1.0 (/home/hudson/tock/capsules)
    Checking nrf5x v0.1.0 (/home/hudson/tock/chips/nrf5x)
    Checking rv32i v0.1.0 (/home/hudson/tock/arch/rv32i)
    Checking cortexm4 v0.1.0 (/home/hudson/tock/arch/cortex-m4)
    Checking cortexm0 v0.1.0 (/home/hudson/tock/arch/cortex-m0)
    Checking cortexm3 v0.1.0 (/home/hudson/tock/arch/cortex-m3)
    Checking sifive v0.1.0 (/home/hudson/tock/chips/sifive)
    Checking lowrisc v0.1.0 (/home/hudson/tock/chips/lowrisc)
    Checking nrf52 v0.1.0 (/home/hudson/tock/chips/nrf52)
    Checking stm32f4xx v0.1.0 (/home/hudson/tock/chips/stm32f4xx)
    Checking sam4l v0.1.0 (/home/hudson/tock/chips/sam4l)
    Checking cc26x2 v0.1.0 (/home/hudson/tock/chips/cc26x2)
    Checking stm32f303xc v0.1.0 (/home/hudson/tock/chips/stm32f303xc)
    Checking apollo3 v0.1.0 (/home/hudson/tock/chips/apollo3)
    Checking e310x v0.1.0 (/home/hudson/tock/chips/e310x)
    Checking arty_e21_chip v0.1.0 (/home/hudson/tock/chips/arty_e21_chip)
    Checking ibex v0.1.0 (/home/hudson/tock/chips/ibex)
    Checking components v0.1.0 (/home/hudson/tock/boards/components)
    Checking nrf52840 v0.1.0 (/home/hudson/tock/chips/nrf52840)
    Checking nrf52832 v0.1.0 (/home/hudson/tock/chips/nrf52832)
    Checking nrf52dk_base v0.1.0 (/home/hudson/tock/boards/nordic/nrf52dk_base)
    Checking stm32f446re v0.1.0 (/home/hudson/tock/chips/stm32f446re)
    Checking stm32f429zi v0.1.0 (/home/hudson/tock/chips/stm32f429zi)
    Finished dev [optimized + debuginfo] target(s) in 11.87s
make[1]: Leaving directory '/home/hudson/tock'
```

gendx, 2020-06-09:
> I think that is just the result of earlier stages of `make ci-nosetup` having compiled everything, so `cargo check` doesn't have to do anything. If you run `make clean` then `make ci-job-syntax` You can see that it builds every board crate:

I checked again by running `make ci-job-syntax` twice in a raw and this explanation is consistent with what I observed. Interesting that Cargo chooses to show the `nrf52dk` rule as a default when there is nothing to do.

closed 2020-06-09 by gendx